luxas/kubernetes-on-arm #1: Hypriot as an alternative to Arch?

nsteinmetz, 2015-10-08:
Hi,

I think you may know the project but [Hypriot](http://blog.hypriot.com/post/get-your-all-in-one-docker-playground-now-hypriotos-reloaded/) provides docker (+compose + swarm + machine) based on Debian Wheezy/Jessie.

Don't know if it could have some interest for your project.

luxas, 2015-10-08:
Hi @nsteinmetz, thanks for reviewing my project!

Yeah, Hypriot has the best choice if one prefers Debian.
This project has two parts, the images (core functions) and the sdcard builder.
In the `sdcard/` folder, there are three subfolders
 - `os`: Here one should add the support for HypriotOS
 - `boot`: There the SD Card partitions is set up
 - `rootfs`: Some files that are copied to the card. Does HypriotOS have systemd?

In fact, I don't know how to make this sd card writing really scalable for all types of OSes and machines.
This is like the "matrix of hell". ARM Boards x OSes supported = number of setup scripts
I would really appreciate thoughts on that point.

E.g. how should the files be downloaded, how scripts could be shared and maintained etc.

But we could come up with something together for sure

nsteinmetz, 2015-10-08:
Hypriot has systemd as it's based on Debian 8

Regarding the matrix of hell, good question, no idea on this as I have no particular experiment on such a topic.

luxas, 2015-10-19:
It may be done in future releases, or I may make a package of this to install on HypriotOS.

MathiasRenner, 2015-12-08:
We (Hypriot) just got in touch with armbian: http://www.armbian.com/download/
Igor, the project's main dev, made it happen to support several ARM boards, now its 19 boards. Together, might be able to extend support of HypriotOS for several boards, but please do not ask for ETA :-)

nsteinmetz, 2015-12-08:
This is so huge that I don't ask for ETA (yet ;-) )
* Armbian is really a nice and effective job on Cubietruck ; I used it for a year at least
* HypriotOS and Kubernetes on arm are nice projects too.

So mixing all together is just awesome :)

And as already said, happy to contribute to make it happen :)

luxas, 2015-12-26:
Whohoo! Finally, `v0.6.2` is released.
`v0.6.0` and `v0.6.2` is tough ones.

I will test Hypriot once more, but I think we could close this.
And I'm really glad to do so, 'cause it means the Hypriot world is ready for Kubernetes OOTB!

I hope this is a good christmas present to you @nsteinmetz :smile:
Test it out when you're ready

nsteinmetz, 2015-12-26:
Awesome @luxas !!!!

My 2 RPI1 are used to test hypriotos cluster-lab but my 2 cubietrucks will be used for 0.6.2 release :)

Awesome to see this two project sharing some of their directions and if the port with armbian works too, arm platform will be definitley awesome for cloud computing and self hosting !

luxas, 2015-12-27:
Don't forget the `.deb` file deployment for your `cluster-lab` Pis :)

closed 2015-12-27 by luxas

nsteinmetz, 2015-12-27:
It was the question I had in mind after answering this - can we start from a cluster-lab to make it k8s aware ? Seems it is then ?!

How do the both project mix together ? Having a docker-swarm cluster in which runs k8s in which runs container is maybe too much inception ? Or is it the way it's implemented ? (we already had docker container to run k8s cluster)

luxas, 2015-12-27:
I think it's best to just run plain Kubernetes first, and see that/how it works. 
At least, that's the only thing that's supported now.
Running Kubernetes under Swarm should technically be possible, but haven't tried it.
See this repo: https://github.com/docker/swarm-frontends/tree/master/kubernetes

Here is some inspiration for ARM: https://gist.github.com/luxas/3a6b0598c3ee2fb84e63
I promise, it won't work :) Just converted the script in swarm-frontends
But, maybe a new issue could be created for this?

luxas, 2015-12-27:
@MathiasRenner Check out the gist. That could maybe be something we could work on...
I was wondering... Is flannel even required when using `docker network overlay`? I don't know