luxas/kubernetes-on-arm #3: New features and usage

luxas, 2015-10-14:
@nsteinmetz
Have you used the project?
Any suggestions?

I've added some new features in this release, v0.5.5
Something more that should be added?
Next thing I'm going to compile is Kube-UI

Anyone else that has tested this?

nsteinmetz, 2015-10-15:
Hi

Not yet ; quite busy right now ; and I just received my latest Rpi for my cluster I ordered one month ago...

Next week-end may be too optimistic but will try from next week. I'll keep you posted.

luxas, 2015-10-15:
@nsteinmetz OK

@piotrhosa What do you think?
Suggestions/improvements/ideas?

luxas, 2015-10-22:
@azimut Do you have suggestions? Are you testing the project out?

azimut, 2015-10-28:
@luxas I am still waiting for other raspberry pi (and some free time) to start testing this. Glad to hear that more people had the same idea.

Though my setup will use Funtoo as the base OS and OpenRC as the init system. I will try to help here too.

Some ideas:
- Have you tried other type of backend for flannel? Not sure if "alloc" will work with kubernetes.
- If you use >=1.1.0 might be you could start testing https://github.com/kubernetes/kubernetes/issues/3760
- I have good experiences with https://github.com/kubernetes/contrib/tree/master/ansible , might be you could fork it or use their logic for the service certificates (I will need to do it at some point too)
- Not k8s related, but have you checked https://github.com/hashicorp/nomad/issues/239 ?

luxas, 2015-10-28:
@azimut 
 - No I haven't tried other backends for flannel, vxlan may work with kubernetes. Do you know pros/cons for the backends?
 - Is Kubernetes 1.1.0 even released? Yes, there are a v1.1.0 tag, but is that stable enough? It might be worth a try anyway.
 - I'll take a look at the ansible playbooks, but it seems like they run the binaries directly on the host, which this project doesn't support right now. But feel free to experiment and share your results.
 - No, I hadn't heard of nomad. Looks pretty good. In which way does it differ from k8s?

You could fork this project, and maybe write some Funtoo and OpenRC scripts. 
Open an issue if you think the documentation should be better at any specific place. (Yeah, it may be improved, believe it or not :smile:)

luxas, 2015-10-28:
Oh, forgot to say that if you haven't noticed, here are the latest prebuilt docker images and plain binaries: https://github.com/luxas/kubernetes-on-arm/releases/tag/v0.5.5

luxas, 2015-11-01:
@Millyna Are you using the project?
Do you have some thoughts?
I'm glad you are watching this project.
Feel free to open issues etc.

luxas, 2015-12-30:
Closing in favor for: #40

closed 2015-12-30 by luxas