luxas/kubernetes-on-arm #4: write.sh only manage /dev/sdX but not /dev/mmcblk0

nsteinmetz, 2015-10-20:
Hi,

My sdcard is seen as /dev/mmcblk0 when using my internal sdcard reader. so write.sh does not work in this context as partitions should be named /dev/mmcblk0p1, /dev/mmcblk0p2, etc.

Output is in French but idea is that /dev/mmcblk01 and /dev/mmcblk02 does not exist as not correclty named. So I'll use my external sdcard reader to have /dev/sdX to go further ;-)

Btw, you need first to make write.sh executable (via chmod +x for example)

```
sudo sdcard/write.sh /dev/mmcblk0 rpi archlinux kube-archlinux
You are going to lose all your data on /dev/mmcblk0. Continue? [Y/n]y
OK. Continuing...
Now /dev/mmcblk0 is going to be partitioned

Bienvenue dans fdisk (util-linux 2.27).
Les modifications resteront en mémoire jusqu'à écriture.
Soyez prudent avant d'utiliser la commande d'écriture.

Commande (m pour l'aide) : Création d'une nouvelle étiquette pour disque de type DOS avec identifiant de disque 0x9114f8ad.

Commande (m pour l'aide) : Disque /dev/mmcblk0 : 3,7 GiB, 3904897024 octets, 7626752 secteurs
Unités : secteur de 1 × 512 = 512 octets
Taille de secteur (logique / physique) : 512 octets / 512 octets
taille d'E/S (minimale / optimale) : 512 octets / 512 octets
Type d'étiquette de disque : dos
Identifiant de disque : 0x9114f8ad

Commande (m pour l'aide) : Type de partition
   p   primaire (0 primaire, 0 étendue, 4 libre)
   e   étendue (conteneur pour partitions logiques)
Sélectionnez (p par défaut) : Numéro de partition (1-4, 1 par défaut) : Premier secteur (2048-7626751, 2048 par défaut) : Dernier secteur, +secteurs ou +taille{K,M,G,T,P} (2048-7626751, 7626751 par défaut) : 
Une nouvelle partition 1 de type « Linux » et de taille 100 MiB a été créée.

Commande (m pour l'aide) : Partition 1 sélectionnée
Type de partition (taper L pour afficher tous les types) : Type de partition « Linux » modifié en « W95 FAT32 (LBA) ».

Commande (m pour l'aide) : Type de partition
   p   primaire (1 primaire, 0 étendue, 3 libre)
   e   étendue (conteneur pour partitions logiques)
Sélectionnez (p par défaut) : Numéro de partition (2-4, 2 par défaut) : Premier secteur (206848-7626751, 206848 par défaut) : Dernier secteur, +secteurs ou +taille{K,M,G,T,P} (206848-7626751, 7626751 par défaut) : 
Une nouvelle partition 2 de type « Linux » et de taille 3,6 GiB a été créée.

Commande (m pour l'aide) : La table de partitions a été altérée.
Appel d'ioctl() pour relire la table de partitions.
Synchronisation des disques.

which: no mkfs.vfat in (/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/sbin:/usr/sbin)
résolution des dépendances...
recherche des conflits entre paquets...

Paquets (1) dosfstools-3.0.28-1

Taille totale du téléchargement :  0,07 MiB
Taille totale installée :         0,23 MiB

:: Procéder à l’installation ? [O/n] 
:: Récupération des paquets...
 dosfstools-3.0.28-1...    75,9 KiB  4,36M/s 00:00 [######################] 100%
(1/1) vérification des clés dans le trousseau      [######################] 100%
(1/1) vérification de l’intégrité des paquets      [######################] 100%
(1/1) chargement des fichiers des paquets          [######################] 100%
(1/1) analyse des conflits entre fichiers          [######################] 100%
(1/1) vérification de l’espace disque disponible   [######################] 100%
(1/1) installation de dosfstools                   [######################] 100%
mkfs.fat 3.0.28 (2015-05-16)
/dev/mmcblk01: No such file or directory
mount: le périphérique spécial /dev/mmcblk01 n'existe pas
mke2fs 1.42.13 (17-May-2015)
Le fichier /dev/mmcblk02 n'existe pas et aucune taille n'a été spécifiée.
mount: le périphérique spécial /dev/mmcblk02 n'existe pas
nsteinmetz@swiip:~/Téléchargements/kubernetes-on-arm$ 
```

luxas, 2015-10-20:
Yeah, when you say it, that's how it is.
The script takes the input `/dev/sda` and appends `1` and `2` for the partitions.
Thanks for observing this.
Which way is best to do it? Ask the user and provide options `PART1=p1 PART2=p2 sudo sdcard/...` for automating the process. What do you think?

But, how couldn't write.sh be executable by default?
I have to investigate it.

nsteinmetz, 2015-10-20:
I was thinking about write.sh being able to manage something like:

In pseudo code:

```
if $1 starts with "/dev/sdX": then :
    part1= /dev/sdX1
    part2= /dev/sdX2
elif $1 starts with "/dev/mmcblkX"; then :
    part1= /dev/mmcblkXp1
    part2= /dev/mmcblkXp2
else:
    exit()

... do some stuff with part1 and part2.
```
I don't know if there could be something else than these 2 cases.

luxas, 2015-10-20:
It's fully possible. 
One could set it like this to handle every case (I don't either know if there are more cases than this):
```
# Special case
if $1 starts with "/dev/mmcblkX"; then :
    part1= ${1}p1
    part2= ${1}p2
else:
    # General case, /dev/sda, (/dev/hda?)
    part1= ${1}1
    part2= ${1}2

# Question the user about continuing
if $continue is some of (n|N); then
   exit
else: 
   # do stuff
```

luxas, 2015-10-22:
Oh, I didn't know that commits could automatically close issues.
Feel free to comment more if it isn't working in future releases.

commit a3b8821b (2015-10-22): Fixes #4, added support for internal sd card readers and made Yy default answer on the overwrite question
closed 2015-10-22 by luxas