luxas/kubernetes-on-arm #5: Write.sh output - default choice & tar mentions "gnoring unknown extended header keyword"

nsteinmetz, 2015-10-20:
Hi,

Two comments on the write.sh output:
* If Y is upper case, then it should be the default choice. If you don"t type y or Y, program exists instead of going further by default.
```
sudo sdcard/write.sh /dev/sdc rpi archlinux kube-archlinux
You are going to lose all your data on /dev/sdc. Continue? [Y/n]
Quitting...
sudo sdcard/write.sh /dev/sdc rpi archlinux kube-archlinux
You are going to lose all your data on /dev/sdc. Continue? [Y/n]y
OK. Continuing...
```

* For the 2nd point, are the tar mentions about ignoring an extended header an issue or not ?

```
sudo sdcard/write.sh /dev/sdc rpi archlinux kube-archlinux
You are going to lose all your data on /dev/sdc. Continue? [Y/n]y
OK. Continuing...
Now /dev/sdc is going to be partitioned

Bienvenue dans fdisk (util-linux 2.27).
Les modifications resteront en mémoire jusqu'à écriture.
Soyez prudent avant d'utiliser la commande d'écriture.

Commande (m pour l'aide) : Création d'une nouvelle étiquette pour disque de type DOS avec identifiant de disque 0xbc2937bf.

Commande (m pour l'aide) : Disque /dev/sdc : 3,7 GiB, 3904897024 octets, 7626752 secteurs
Unités : secteur de 1 × 512 = 512 octets
Taille de secteur (logique / physique) : 512 octets / 512 octets
taille d'E/S (minimale / optimale) : 512 octets / 512 octets
Type d'étiquette de disque : dos
Identifiant de disque : 0xbc2937bf

Commande (m pour l'aide) : Type de partition
   p   primaire (0 primaire, 0 étendue, 4 libre)
   e   étendue (conteneur pour partitions logiques)
Sélectionnez (p par défaut) : Numéro de partition (1-4, 1 par défaut) : Premier secteur (2048-7626751, 2048 par défaut) : Dernier secteur, +secteurs ou +taille{K,M,G,T,P} (2048-7626751, 7626751 par défaut) : 
Une nouvelle partition 1 de type « Linux » et de taille 100 MiB a été créée.

Commande (m pour l'aide) : Partition 1 sélectionnée
Type de partition (taper L pour afficher tous les types) : Type de partition « Linux » modifié en « W95 FAT32 (LBA) ».

Commande (m pour l'aide) : Type de partition
   p   primaire (1 primaire, 0 étendue, 3 libre)
   e   étendue (conteneur pour partitions logiques)
Sélectionnez (p par défaut) : Numéro de partition (2-4, 2 par défaut) : Premier secteur (206848-7626751, 206848 par défaut) : Dernier secteur, +secteurs ou +taille{K,M,G,T,P} (206848-7626751, 7626751 par défaut) : 
Une nouvelle partition 2 de type « Linux » et de taille 3,6 GiB a été créée.

Commande (m pour l'aide) : La table de partitions a été altérée.
Appel d'ioctl() pour relire la table de partitions.
Synchronisation des disques.

mkfs.fat 3.0.28 (2015-05-16)
mke2fs 1.42.13 (17-May-2015)
En train de créer un système de fichiers avec 927488 4k blocs et 232000 i-noeuds.
UUID de système de fichiers=db5cddad-51f9-4a26-9000-40e04c548260
Superblocs de secours stockés sur les blocs : 
	32768, 98304, 163840, 229376, 294912, 819200, 884736

Allocation des tables de groupe : complété                        
Écriture des tables d'i-noeuds : complété                        
Création du journal (16384 blocs) : complété
Écriture des superblocs et de l'information de comptabilité du système de
fichiers : complété

Partitions mounted
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'LIBARCHIVE.xattr.security.capability'
tar: Ignoring unknown extended header keyword 'LIBARCHIVE.xattr.security.capability'
tar: Ignoring unknown extended header keyword 'LIBARCHIVE.xattr.security.capability'
tar: Ignoring unknown extended header keyword 'LIBARCHIVE.xattr.security.capability'
tar: Ignoring unknown extended header keyword 'LIBARCHIVE.xattr.security.capability'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
tar: Ignoring unknown extended header keyword 'SCHILY.fflags'
OS written to SD Card
nsteinmetz@swiip:~/Téléchargements/kubernetes-on-arm$ 
```
Thanks,
Nicolas

luxas, 2015-10-20:
Yeah, I know about the tar ignore messages, they occur every time. It is the Arch Linux ARM tar package that has some unknown things, but it's working fine anyway.
Maybe I should silence them? It's best for the UX.

It's done by appending the untar with `>/dev/null`. That should hide the warnings.

nsteinmetz, 2015-10-20:
Yes maybe it should be better to silence them (or maybe redirect them to a log and not to stdout ?). Depends on if you expect them to be useful for some debug or not.

luxas, 2015-10-20:
I may redirect them to `/tmp/kubernetes-on-arm.log`, so they are possible to debug, but won't be there forever (I think `/tmp` is flushed at reboot)

luxas, 2015-10-25:
Now it should be fixed with the warnings. 
`stderr` and `stdout` is redirected and appended to `/tmp/kubernetes-on-arm.log`, which is flushed at reboot.

luxas, 2015-10-25:
This is fixed now in latest version.

closed 2015-10-25 by luxas