luxas/kubernetes-on-arm #11: [dev] kube-config install raise some errors 2/2

nsteinmetz, 2015-10-25:
With latest dev as of today:
* https://github.com/luxas/kubernetes-on-arm/blob/dev/sdcard/rootfs/kube-archlinux/usr/bin/kube-config.sh#L192 will raise:

```
Do you want to download Kubernetes for ARM docker images? So you won't have to build them yourself. Y is default. [Y/n] 
rm: cannot remove '/tmp/dlk8s': No such file or directory
```

luxas, 2015-10-25:
I'm updating the whole process. Got this myself some minutes ago. I have a patch soon.

nsteinmetz, 2015-10-25:
fixed with c63bd77989a51025a71d18b11a38c4d1ce6ff87b

Let me know once I can test it :)

closed 2015-10-25 by nsteinmetz

luxas, 2015-10-26:
Yeah, sorry. 
I was interrupted when I should have uploaded the second of the two commits.
At least it's there now.
But that's how `dev` may be :smile: