luxas/kubernetes-on-arm #12: [dev] kube-config - symlink for /etc/kubernetes/binaries is broken

nsteinmetz, 2015-10-25:
Symlink is false:
```
[root@thanos kubernetesonarm]# ls -al /etc/kubernetes/binaries 
lrwxrwxrwx 1 root root 57 Oct 25 19:40 /etc/kubernetes/binaries -> /etc/kubernetes/source/images/kubernetesonarm/_bin/latest
```

However, this directory does not contain a _bin directory:

```
[root@thanos kubernetesonarm]# pwd
/etc/kubernetes/source/images/kubernetesonarm
[root@thanos kubernetesonarm]# ls
build  etcd  exechealthz  flannel  hyperkube  kube-ui  kube2sky  pause	registry  skydns
```

luxas, 2015-10-25:
This is because I forgot a `mkdir` statement. Will soon be fixed. It is right, in _bin/latest the binaries will be stored.

nsteinmetz, 2015-10-25:
Fixed with c63bd77989a51025a71d18b11a38c4d1ce6ff87b

closed 2015-10-25 by nsteinmetz