luxas/kubernetes-on-arm #13: Test Results - V 0.5.8 - Cubietruck

nsteinmetz, 2015-10-30:
Not only to open issues but also track what works well :)

Will update it over the week-end.

On Cubietruck:
* Write to sdcard: OK
* kube-config install process: OK
* kube-config enable-master: OK

I have as output:

```
[root@thanos ~]# docker ps
CONTAINER ID        IMAGE                       COMMAND                CREATED             STATUS              PORTS               NAMES
f7f6d35699b1        kubernetesonarm/hyperkube   "/hyperkube schedule   2 minutes ago       Up 2 minutes                            k8s_scheduler.e5efe276_k8s-master-192.168.8.100_kube-system_447c171dfac8ae64dc585f8e9cbfa7e6_da0c9c8a            
fff8c1807c96        kubernetesonarm/hyperkube   "/hyperkube apiserve   2 minutes ago       Up 2 minutes                            k8s_apiserver.c358020f_k8s-master-192.168.8.100_kube-system_447c171dfac8ae64dc585f8e9cbfa7e6_dca5fe9f            
a62c47d12e76        kubernetesonarm/hyperkube   "/hyperkube controll   2 minutes ago       Up 2 minutes                            k8s_controller-manager.7047e990_k8s-master-192.168.8.100_kube-system_447c171dfac8ae64dc585f8e9cbfa7e6_441ccae3   
4a4ad3f6c696        kubernetesonarm/pause       "/pause"               2 minutes ago       Up 2 minutes                            k8s_POD.7ad6c339_k8s-master-192.168.8.100_kube-system_447c171dfac8ae64dc585f8e9cbfa7e6_966ab67b                  
4a78dccf9faf        kubernetesonarm/hyperkube   "/hyperkube kubelet    2 minutes ago       Up 2 minutes                            k8s-master                                                                                                       
d244bc1afb7e        kubernetesonarm/hyperkube   "/hyperkube proxy --   2 minutes ago       Up 2 minutes                            k8s-worker-proxy
```
Question : is it normal, I have twice each images, one in the latest stage and one in 0.5.5 ?

```                                                                                                 
[root@thanos ~]# docker images
REPOSITORY                    TAG                 IMAGE ID            CREATED             VIRTUAL SIZE
kubernetesonarm/skydns        0.5.5               0a73ba8222b8        2 weeks ago         9.953 MB
kubernetesonarm/skydns        latest              0a73ba8222b8        2 weeks ago         9.953 MB
kubernetesonarm/flannel       latest              323b7a93d03b        2 weeks ago         90.25 MB
kubernetesonarm/flannel       0.5.5               323b7a93d03b        2 weeks ago         90.25 MB
kubernetesonarm/registry      0.5.5               9b888b2f2d5a        2 weeks ago         109 MB
kubernetesonarm/registry      latest              9b888b2f2d5a        2 weeks ago         109 MB
kubernetesonarm/exechealthz   0.5.5               8d67dee89c29        2 weeks ago         11.73 MB
kubernetesonarm/exechealthz   latest              8d67dee89c29        2 weeks ago         11.73 MB
kubernetesonarm/kube2sky      0.5.5               d7e9a144c99d        2 weeks ago         13.1 MB
kubernetesonarm/kube2sky      latest              d7e9a144c99d        2 weeks ago         13.1 MB
kubernetesonarm/pause         latest              22e22d3123c0        2 weeks ago         227 kB
kubernetesonarm/pause         0.5.5               22e22d3123c0        2 weeks ago         227 kB
kubernetesonarm/hyperkube     0.5.5               fd5c7305c597        2 weeks ago         125.5 MB
kubernetesonarm/hyperkube     latest              fd5c7305c597        2 weeks ago         125.5 MB
kubernetesonarm/etcd          latest              cc678c339590        2 weeks ago         18.2 MB
kubernetesonarm/etcd          0.5.5               cc678c339590        2 weeks ago         18.2 MB
```

I'll need to move current content of my cubietruck to a RPi1 so that I can use the former as a worker ndoe.

So far so good :)

luxas, 2015-10-31:
Yes, it's normal to have two of each image, it included both when i packaged the images (automatically). 
Which is better in the future, to package just "latest" or "v0.x.x" or both.
Pros with latest: doesn't give suspicious signals to the user 
Cons with latest: if newer images are loaded, I think all "old" images will turn into "\<none\>". But this may be fixed in the download script

Cons with version number: isn't related at all to the version of the binary in the image (e.g. k8s v1.0.6 => kubernetes-on-arm v0.5.5, v0.5.6, v0.5.8)

What do you think?

nsteinmetz, 2015-10-31:
I would think that explicit version may be better than only "latest" ; as latest may not mean the same thing at a given time. Especially during a development phase ;-)

nsteinmetz, 2015-11-01:
So, on my 2nd cubietruck, seems everything worked well too !

```
[root@or-azur ~]# kube-config enable-worker
Disabling k8s if it is running
What is the Master IP? It isn't specified or reachable at the moment. 192.168.8.100
Checks so all images are present
Downloading Kubernetes docker images from Github
Loading them to docker
Finished loading them to docker
Transferring images to system-docker, if necessary
Copying kubernetesonarm/flannel to system-docker
Created symlink from /etc/systemd/system/multi-user.target.wants/flannel.service to /usr/lib/systemd/system/flannel.service.
Starting the worker containers
Created symlink from /etc/systemd/system/multi-user.target.wants/k8s-worker.service to /usr/lib/systemd/system/k8s-worker.service.
Worker Kubernetes services enabled
```
As docker processes:
```
[root@or-azur ~]# docker ps
CONTAINER ID        IMAGE                       COMMAND                CREATED             STATUS              PORTS               NAMES
4ba43ac2f225        kubernetesonarm/hyperkube   "/hyperkube kubelet    23 minutes ago      Up 23 minutes                           k8s-worker          
6c9bd057a4d9        kubernetesonarm/hyperkube   "/hyperkube proxy --   23 minutes ago      Up 23 minutes                           k8s-worker-proxy    
```
All images are there:
```
[root@or-azur ~]# docker  images
REPOSITORY                    TAG                 IMAGE ID            CREATED             VIRTUAL SIZE
kubernetesonarm/skydns        latest              0a73ba8222b8        2 weeks ago         9.953 MB
kubernetesonarm/skydns        0.5.5               0a73ba8222b8        2 weeks ago         9.953 MB
kubernetesonarm/flannel       0.5.5               323b7a93d03b        2 weeks ago         90.25 MB
kubernetesonarm/flannel       latest              323b7a93d03b        2 weeks ago         90.25 MB
kubernetesonarm/registry      0.5.5               9b888b2f2d5a        2 weeks ago         109 MB
kubernetesonarm/registry      latest              9b888b2f2d5a        2 weeks ago         109 MB
kubernetesonarm/exechealthz   0.5.5               8d67dee89c29        2 weeks ago         11.73 MB
kubernetesonarm/exechealthz   latest              8d67dee89c29        2 weeks ago         11.73 MB
kubernetesonarm/kube2sky      0.5.5               d7e9a144c99d        2 weeks ago         13.1 MB
kubernetesonarm/kube2sky      latest              d7e9a144c99d        2 weeks ago         13.1 MB
kubernetesonarm/pause         0.5.5               22e22d3123c0        2 weeks ago         227 kB
kubernetesonarm/pause         latest              22e22d3123c0        2 weeks ago         227 kB
kubernetesonarm/hyperkube     0.5.5               fd5c7305c597        2 weeks ago         125.5 MB
kubernetesonarm/hyperkube     latest              fd5c7305c597        2 weeks ago         125.5 MB
kubernetesonarm/etcd          0.5.5               cc678c339590        2 weeks ago         18.2 MB
kubernetesonarm/etcd          latest              cc678c339590        2 weeks ago         18.2 MB
```
Seems I can now play with kubernetes :)

nsteinmetz, 2015-11-01:
Seems everything is fine:

```
[root@thanos ~]# kubectl get pods
NAME             READY     STATUS    RESTARTS   AGE
my-nginx-g8hfp   1/1       Running   0          4m
my-nginx-r64pq   1/1       Running   0          4m
my-nginx-s4dqc   1/1       Running   0          4m
```

```
[root@thanos ~]# kubectl get nodes
NAME            LABELS                                 STATUS
192.168.8.100   kubernetes.io/hostname=192.168.8.100   Ready
192.168.8.101   kubernetes.io/hostname=192.168.8.101   Ready
```

```
kubectl expose rc/my-nginx --port=80
NAME       LABELS         SELECTOR       IP(S)     PORT(S)
my-nginx   run=my-nginx   run=my-nginx             80/TCP
```

```
[root@thanos ~]# kubectl get svc
NAME         LABELS                                    SELECTOR       IP(S)        PORT(S)
kubernetes   component=apiserver,provider=kubernetes   <none>         10.0.0.1     443/TCP
my-nginx     run=my-nginx                              run=my-nginx   10.0.0.134   80/TCP
```

```
[root@thanos ~]# curl http://10.0.0.134/
<p>WELCOME TO NGINX</p>
```

```
[root@thanos ~]# kube-config enable-addon dns
replicationcontrollers/kube-dns-v8
services/kube-dns
Started addon: dns
```

```
[root@thanos ~]# kubectl --namespace=kube-system get pods,rc,svc
NAME                       READY     STATUS    RESTARTS   AGE
k8s-master-192.168.8.100   3/3       Running   0          1d
kube-dns-v8-wgpej          4/4       Running   0          1m
CONTROLLER    CONTAINER(S)   IMAGE(S)                      SELECTOR           REPLICAS
kube-dns-v8   etcd           kubernetesonarm/etcd          k8s-app=kube-dns   1
              kube2sky       kubernetesonarm/kube2sky                         
              skydns         kubernetesonarm/skydns                           
              healthz        kubernetesonarm/exechealthz                      
NAME       LABELS                                                                           SELECTOR           IP(S)       PORT(S)
kube-dns   k8s-app=kube-dns,kubernetes.io/cluster-service=true,kubernetes.io/name=KubeDNS   k8s-app=kube-dns   10.0.0.10   53/UDP
                                                                                                                           53/TCP
```

Even if you say it later, it may be worth saying you need to add 10.0.0.10 as DNS on your cluster to resolve cluster.local zone.
```
[root@thanos ~]# curl my-nginx.default.svc.cluster.local
<p>WELCOME TO NGINX</p>
```

```
kube-config enable-addon registry
replicationcontrollers/registry
services/registry
Started addon: registry
```

[... 34 lines cut ...]
udp        0      0 0.0.0.0:8285            0.0.0.0:*                           820/flanneld        
udp6       0      0 :::5355                 :::*                                188/systemd-resolve 
udp6       0      0 :::53525                :::*                                962/hyperkube       
Active UNIX domain sockets (only servers)
Proto RefCnt Flags       Type       State         I-Node   PID/Program name     Path
unix  2      [ ACC ]     STREAM     LISTENING     19726    1/init               /var/run/docker.sock
unix  2      [ ACC ]     STREAM     LISTENING     16165    701/docker           /var/run/system-docker.sock
unix  2      [ ACC ]     STREAM     LISTENING     556      1/init               /run/lvm/lvmetad.socket
unix  2      [ ACC ]     STREAM     LISTENING     11569    1/init               /run/dbus/system_bus_socket
unix  2      [ ACC ]     STREAM     LISTENING     15673    306/systemd          /run/user/0/bus
unix  2      [ ACC ]     STREAM     LISTENING     17211    1/init               /var/run/system-docker.sock
unix  2      [ ACC ]     STREAM     LISTENING     19014    876/docker           /var/run/docker.sock
unix  2      [ ACC ]     SEQPACKET  LISTENING     352      1/init               /run/udev/control
unix  2      [ ACC ]     STREAM     LISTENING     366      1/init               /run/systemd/journal/stdout
unix  2      [ ACC ]     STREAM     LISTENING     16567    306/systemd          /run/user/0/systemd/private
unix  2      [ ACC ]     STREAM     LISTENING     11201    1/init               /run/systemd/private
```
and:
```
[root@thanos ~]# kubectl cluster-info
Kubernetes master is running at http://localhost:8080
KubeDNS is running at http://localhost:8080/api/v1/proxy/namespaces/kube-system/services/kube-dns
KubeRegistry is running at http://localhost:8080/api/v1/proxy/namespaces/kube-system/services/registry
```
Seems everything is fine so far :)

nsteinmetz, 2015-11-01:
Now I need to dive deeper in kuberntes world to make my docker images accessible from outside of the pods :)

closed 2015-11-01 by nsteinmetz

luxas, 2015-11-02:
Glad it worked! :)

What do you mean with this?
> Even if you say it later, it may be worth saying you need to add 10.0.0.10 as DNS on your cluster to resolve cluster.local zone.

First, you'll have to make a service of some pods. The service will expose some of the pod's ports (those you specify) and the service will get an ip (10.0.0.156).

I may add the [service-loadbalancer](https://github.com/kubernetes/contrib/tree/master/service-loadbalancer) as an addon.
That would make some services accessible at e.g. http://[node-ip]:[service-port]/[service-name] if I remember correctly.

But already your nginx is accessible from outside, via the master: http://[master-ip]:8080/api/v1/proxy/namespaces/default/services/my-nginx

nsteinmetz, 2015-11-02:
I mean : in your readme,md file, from which I type all the commands above, there is no mention of the 10.0.0.10 as the DNS. It may be worth being mentionned. I found it was mentionned later in the addons section that the DNS was using this IP.

It may be also worth to add that you can add "http://[master-ip]:8080/api/v1/proxy/namespaces/default/services/my-nginx" in the readme.md too.

luxas, 2015-11-02:
Thanks for your opinion! 
I'm adding it to dev :)

nsteinmetz, 2015-12-02:
For the record, I re-run this test suite against 0.6.0 and it works like a charm :)