luxas/kubernetes-on-arm #14: Docker hello world error

tjskrish, 2015-11-06:
Getting this error while trying to run docker hello-world:

`exec format error`
`Error response from daemon: Cannot start container 977d87157e8c34e70c1ae8af399a7dd3a41623b991a907ba969011b6ef6f5e54: [8] System error: exec format error`

Any idea why ?

luxas, 2015-11-06:
Hello @tjskrish and thanks for using this project!

I struggled many times with this in the beginning of the project,
Docker is said run on everywhere, but that is not the case.
Docker "officially" supports just `amd64` machines.
So even if docker is built on a lot of platforms, nearly every *docker image* is `amd64`
That makes nearly whole Docker Hub unavailable for ones that run things on ARM, which is *32-bit*

In an ideal world, docker images could run on every machine regardless of platform.

Think of it like this: 
- One 64-bit user makes a nginx image
- In the `Dockerfile`, a 64-bit nginx binary is downloaded from nginx website, to `/usr/bin/nginx`
- The user builds the docker image.

If I then downloaded this image and tried to run it, I wouldn't work, because of that **64-bit** binary
So, therefore, docker exits before it even starts the image.

IMO, this error message could be much more user-friendly

Hope this answers your question! :smile:

tjskrish, 2015-11-06:
Ahh!! I guessed it right! Not even a 32 bit java image will work on it in that case right ?

luxas, 2015-11-06:
I don't think so. It has to be built on ARM.

So, for this project, I've built every required image by hand.
Often things has to be compiled from source.

But if you search for e.g. `armhf` on Docker Hub, you'll find useful stuff anyway.
Some cool images for ARM:
 - Hypriot: https://hub.docker.com/u/hypriot
 - Armbuild: https://hub.docker.com/u/armbuild
 - and myself: https://hub.docker.com/u/luxas

Maybe you'll find an java image there.
If you just want so start hacking:
```
docker run -it luxas/raspbian /bin/bash
```

tjskrish, 2015-11-06:
Alright. Thanks for the quick reply. Really impressed! Il close this issue for now.

closed 2015-11-06 by tjskrish

luxas, 2015-11-06:
BTW, Did you get Kubernetes working on your Pi?

tjskrish, 2015-11-06:
Yup, I did! Your repo is really helpful!

luxas, 2015-11-06:
Thanks! And I will make it better :smile:
New versions of this software is released really fast.

What do you think, should I make a entry in `docs/` about this?
After all, I've written it down already.

tjskrish, 2015-11-06:
Well, you should do it if you have the time! Iam pretty sure more people like me will find it useful. N btw, i was not able to get the go container running with the image you provided. I would also suggest adding that docker wouldn't run on 32 bit arm in the known issues.

luxas, 2015-11-06:
Why didn't `luxas/go` start? It has to be pulled at least

tjskrish, 2015-11-06:
Its getting pulled, but not starting. I will paste the error that I am getting after a while.