luxas/kubernetes-on-arm #17: You make me smarter when you leave stuff as an exercise to the user

faddat, 2015-11-14:
Learning to make a .deb package now

Thanks!

luxas, 2015-11-14:
Great! Just a small note, you have to source `dynamic-rootfs.sh` and run the `rootfs` function for all files to be copied

Something like this...
```bash
cd kubernetes-on-arm
export PROJROOT=$(pwd)
export ROOT=<dir> (/tmp/somedir)
mkdir $ROOT
cp -r $PROJROOT/sdcard/rootfs/kube-archlinux/* $ROOT
source $ROOT/dynamic-rootfs.sh
rootfs
rm $ROOT/dynamic-rootfs.sh
# make deb package
```

Say how you did it when you succeed :smile:

faddat, 2015-11-19:
@luxas this is really a bulwark, isn't it?  It'll take me a little bit of time to grasp your dependency system, but FWIW, I can *use* it now.  The thing that comes to my mind first acutally isnt .debs anymore-- it's "how can we get this merged with the k8s mainline?"

Was there one particular change that you made to get things running on ARM, or....?

Something tells me it wasn't *just* one.

luxas, 2015-11-19:
@faddat My long-term goal is getting this in mainline k8s, and I am working on it via this project.
The thing is, we can't build it on amd64 right now, but this will probably be fixed with `go1.5.2`. https://github.com/golang/go/issues/13024

And I don't think the k8s guys want ARM as a "official" platform when even docker isn't supporting ARM. 
Sometimes, things also fails (more or less strangely) on ARM. https://github.com/kubernetes/kubernetes/issues/17060

But, soon https://github.com/kubernetes/kubernetes/pull/13901 maybe is merged and then k8s will have a own `k8s-in-docker` deployment strategy. 
However that one doesn't have `systemd` support at all, and one can't choose which ARM images to spin up.

You mentioned how I managed to run k8s on ARM:
 - Make a `golang` image for ARM.
 - Download all source code from github: `etcd`, `flannel`, `kubernetes`, `contrib`, `registry`, `skydns` and compile them
 - Make all docker images, however if one prefers a binary installation, prebuilt binaries are downloadable from GitHub
 - The repo also builds `skydns`, `kube2sky`, `exechealthz`, `registry` images for addons.

Here are the parts of the project: 
 - core k8s (binaries, images, compilation)
 - addons (they also have to be built)
 - the sdcard build process (ARM boards often use a SD Card for root storage, in fact this may be whatever drive)
 - the `systemd` integration with `k8s-in-docker` => kube-archlinux rootfs

You mentioned that it takes some time to understand the sd card build process, I may partially agree, bash isn't a very "good" programming language in terms of modularity and scalability... But if you have suggestions to the process, notify me
But it have to be so modular as possible when new boards and OSes are added

Another thing to consider is the OS. Arch Linux ARM is good indeed, but it would be great if it didn't matter which OS kubernetes ran on. This may be simplified when I maybe will include a statically linked docker daemon.

TL;DR
It isn't just one thing to consider, nearly everything has to be customized for ARM
And it is a goal to merge in things to mainline k8s, but not the highest priority.

Did you ever make the package?

faddat, 2015-11-19:
Sorry, haven't read right now, have just looked at the end, but:

No-- I'm still picking apart what you've got going on with the containers, more detail tomorrow....

faddat, 2015-11-21:
Trying this: 
https://github.com/jordansissel/fpm

faddat, 2015-11-21:
I think I did very poorly reading the middle section of the docs...

luxas, 2015-12-27:
@faddat I've made a `.deb` package now.
See the source code: https://github.com/luxas/kubernetes-on-arm/blob/master/images/kubernetesonarm/make-deb/makedeb.sh

Will you test it on Scaleway? It runs at least on plain Raspbian, so it should work on Scaleway too

luxas, 2016-01-05:
Closing as we have a working `.deb` package now

closed 2016-01-05 by luxas