luxas/kubernetes-on-arm #18: error building

streamnsight, 2015-11-14:
I pull the repo (master branch = 0.5.8?) , and followed the instructions. 
wrote the SD card from a ubuntu distro.

I'm getting an error building hypercube

```
Building: kubernetesonarm/build
Sending build context to Docker daemon 9.728 kB
Sending build context to Docker daemon
Step 0 : FROM luxas/go
 ---> 65e1d60afb17
Step 1 : COPY inbuild.sh version.sh /
 ---> 3e841df0eb1e
Removing intermediate container 9ee57d74055c
Step 2 : RUN ./inbuild.sh
 ---> Running in 03640d1cfa71
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   645  100   645    0     0    112      0  0:00:05  0:00:05 --:--:--   199
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   368  100   368    0     0     64      0  0:00:05  0:00:05 --:--:--    87
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   500  100   500    0     0     86      0  0:00:05  0:00:05 --:--:--   117
patching file etcdserver/raft.go
patching file etcdserver/server.go
Hunk #1 succeeded at 114 (offset 1 line).
patching file store/watcher_hub.go
fatal: Not a git repository (or any of the parent directories): .git
Building flanneld...
+++ [1114 01:52:01] Building go targets for linux/arm:
    cmd/hyperkube
    cmd/kubectl
# k8s.io/kubernetes/cmd/hyperkube
/goroot/pkg/tool/linux_arm/5l: running gcc failed: Cannot allocate memory
!!! Error in /build/kubernetes/hack/lib/golang.sh:383
  'go install "${goflags[@]:+${goflags[@]}}" -ldflags "${version_ldflags}" "${nonstatics[@]:+${nonstatics[@]}}"' exited with status 2
Call stack:
  1: /build/kubernetes/hack/lib/golang.sh:383 kube::golang::build_binaries_for_platform(...)
  2: /build/kubernetes/hack/lib/golang.sh:521 kube::golang::build_binaries(...)
  3: ./hack/build-go.sh:26 main(...)
Exiting with status 1
!!! Error in /build/kubernetes/hack/lib/golang.sh:439
  '( kube::golang::setup_env; local version_ldflags; version_ldflags=$(kube::version::ldflags); local host_platform; host_platform=$(kube::golang::host_platform); local goflags; eval "goflags=(${KUBE_GOFLAGS:-})"; local use_go_build; local -a targets=(); local arg; for arg in "$@";
do
    if [[ "${arg}" == "--use_go_build" ]]; then
        use_go_build=true;
    else
        if [[ "${arg}" == -* ]]; then
            goflags+=("${arg}");
        else
            targets+=("${arg}");
        fi;
    fi;
done; if [[ ${#targets[@]} -eq 0 ]]; then
    targets=("${KUBE_ALL_TARGETS[@]}");
fi; local -a platforms=("${KUBE_BUILD_PLATFORMS[@]:+${KUBE_BUILD_PLATFORMS[@]}}"); if [[ ${#platforms[@]} -eq 0 ]]; then
    platforms=("${host_platform}");
fi; local binaries; binaries=($(kube::golang::binaries_from_targets "${targets[@]}")); local parallel=false; if [[ ${#platforms[@]} -gt 1 ]]; then
    local gigs; gigs=$(kube::golang::get_physmem); if [[ ${gigs} -gt ${KUBE_PARALLEL_BUILD_MEMORY} ]]; then
        kube::log::status "Multiple platforms requested and available ${gigs}G > threshold ${KUBE_PARALLEL_BUILD_MEMORY}G, building platforms in parallel"; parallel=true;
    else
        kube::log::status "Multiple platforms requested, but available ${gigs}G < threshold ${KUBE_PARALLEL_BUILD_MEMORY}G, building platforms in serial"; parallel=false;
    fi;
fi; if [[ "${parallel}" == "true" ]]; then
    kube::log::status "Building go targets for ${platforms[@]} in parallel (output will appear in a burst when complete):" "${targets[@]}"; local platform; for platform in "${platforms[@]}";
    do
        ( kube::golang::set_platform_envs "${platform}"; kube::log::status "${platform}: go build started"; kube::golang::build_binaries_for_platform ${platform} ${use_go_build:-}; kube::log::status "${platform}: go build finished" ) &> "/tmp//${platform//\//_}.build" &
    done; local fails=0; for job in $(jobs -p);
    do
        wait ${job} || let "fails+=1";
    done; for platform in "${platforms[@]}";
    do
        cat "/tmp//${platform//\//_}.build";
    done; exit ${fails};
else
    for platform in "${platforms[@]}";
    do
        kube::log::status "Building go targets for ${platform}:" "${targets[@]}"; kube::golang::set_platform_envs "${platform}"; kube::golang::build_binaries_for_platform ${platform} ${use_go_build:-};
    done;
fi )' exited with status 1
Call stack:
  1: /build/kubernetes/hack/lib/golang.sh:439 kube::golang::build_binaries(...)
  2: ./hack/build-go.sh:26 main(...)
Exiting with status 1
cp: cannot stat '_output/local/bin/linux/arm/*': No such file or directory
+ go build --ldflags '-extldflags "-static" -s' pause.go
+ go get github.com/pwaller/goupx
+ goupx pause
2015/11/14 01:54:26 {Class:ELFCLASS32 Data:ELFDATA2LSB Version:EV_CURRENT OSABI:ELFOSABI_NONE ABIVersion:0 ByteOrder:LittleEndian Type:ET_EXEC Machine:EM_ARM Entry:284028}
2015/11/14 01:54:26 File fixed!
                       Ultimate Packer for eXecutables
                          Copyright (C) 1996 - 2013
UPX 3.91        Markus Oberhumer, Laszlo Molnar & John Reiser   Sep 30th 2013

        File size         Ratio      Format      Name
   --------------------   ------   -----------   -----------
    632584 ->    227064   35.89%   linux/armel   pause

Packed 1 file.
```

then when buildign the hyperkube image it says the exe is missing.

i haven't do much go beside reading through the k8s code to understand stuff, so I'm not sure what this all means.

any idea? 

Thanks for help.

luxas, 2015-11-14:
It seems like you haven't made the swap file when you installed.

How to do it manually:
```bash
# Make 1GB swap
dd if=/dev/zero of=/swapfile bs=1M count=1024

# Enable it with right permissions
mkswap /swapfile
chmod 600 /swapfile
swapon /swapfile

# And recreate it on every boot
cat >> /etc/fstab <<EOF
/swapfile  none  swap  defaults  0  0
EOF
```
You probably have to reboot after that.

streamnsight, 2015-11-14:
You're the man! I guess I had not followed the instruction carefully enough... 
Thanks

closed 2015-11-14 by streamnsight