luxas/kubernetes-on-arm #19: enable-master breaks docker

stevesloka, 2015-11-20:
Hey I'm trying to get just the master node up and running. I did `kube-config install` then `kube-config enable-master` and rebooted. Before enabling the master docker works fine, after it's broken. When I do a status I get the following, have you run into this?

```
[root@k8s-master ~]# systemctl start docker.service
Job for docker.service failed because a configured resource limit was exceeded. See "systemctl status docker.service" and "journalctl -xe" for details.
[root@k8s-master ~]# systemctl status docker.service
* docker.service - Docker Application Container Engine
   Loaded: loaded (/usr/lib/systemd/system/docker.service; enabled; vendor preset: disabled)
  Drop-In: /usr/lib/systemd/system/docker.service.d
           `-docker-flannel.conf
   Active: failed (Result: start-limit)
     Docs: https://docs.docker.com

Nov 20 15:36:27 k8s-master systemd[1]: Failed to start Docker Application Container Engine.
Nov 20 15:36:27 k8s-master systemd[1]: docker.service: Failed with result 'start-limit'.
Nov 20 15:37:52 k8s-master systemd[1]: docker.service: Failed to load environment files: No such file or directory
Nov 20 15:37:52 k8s-master systemd[1]: docker.service: Failed to run 'start-pre' task: No such file or directory
Nov 20 15:37:52 k8s-master systemd[1]: Failed to start Docker Application Container Engine.
Nov 20 15:37:52 k8s-master systemd[1]: docker.service: Failed with result 'resources'.
Nov 20 15:37:55 k8s-master systemd[1]: docker.service: Failed to load environment files: No such file or directory
Nov 20 15:37:55 k8s-master systemd[1]: docker.service: Failed to run 'start-pre' task: No such file or directory
Nov 20 15:37:55 k8s-master systemd[1]: Failed to start Docker Application Container Engine.
Nov 20 15:37:55 k8s-master systemd[1]: docker.service: Failed with result 'resources'.
```

luxas, 2015-11-20:
Hello @stevesloka!
Glad to see you are testing this out.
Yeah, this is a known issue, that reboots doesn't work properly.
I've fixed it in bd95c0e4871a1c7f9542631042482417056eeeee

The only thing you will have to do:
```
nano /etc/kubernetes/dynamic-dropins/docker-flannel.conf
# Add these two lines
Restart=on-failure
RestartSec=2
```
The issue happens because when docker starts, it want a file that flannel provides.
However, flannel requires some secs to start up, so docker fails too many times, before flannel is ready
This patch makes docker wait longer for flannel

This code is already merged in `dev`, which will be released quite soon.

Generally speaking, you may always revert `enable-master` with `kube-config disable-node`

stevesloka, 2015-11-20:
Thanks for the heads up. I copied in those two lines, but still no go. Docker is still not starting.

```
[root@k8s-master ~]# systemctl status docker.service
* docker.service - Docker Application Container Engine
   Loaded: loaded (/usr/lib/systemd/system/docker.service; enabled; vendor preset: disabled)
  Drop-In: /usr/lib/systemd/system/docker.service.d
           `-docker-flannel.conf
   Active: activating (auto-restart) (Result: resources) since Fri 2015-11-20 20:00:36 EET; 1s ago
     Docs: https://docs.docker.com

Nov 20 20:00:36 k8s-master systemd[1]: docker.service: Failed to run 'start-pre' task: No such file or directory
Nov 20 20:00:36 k8s-master systemd[1]: Failed to start Docker Application Container Engine.
Nov 20 20:00:36 k8s-master systemd[1]: docker.service: Unit entered failed state.
Nov 20 20:00:36 k8s-master systemd[1]: docker.service: Failed with result 'resources'.
```

luxas, 2015-11-20:
Which docker version are you using?
Can you output `kube-config info`?

stevesloka, 2015-11-20:
```
[root@k8s-master ~]# docker ps
CONTAINER ID        IMAGE               COMMAND             CREATED             STATUS              PORTS               NAMES
[root@k8s-master ~]# docker info
Containers: 1
Images: 27
Server Version: 1.9.0
Storage Driver: overlay
 Backing Filesystem: extfs
Execution Driver: native-0.2
Logging Driver: json-file
Kernel Version: 4.1.13-1-ARCH
Operating System: Arch Linux ARM (containerized)
CPUs: 4
Total Memory: 923.1 MiB
Name: k8s-master
ID: GVUL:M3IB:D7QC:AAGZ:4F3O:LM3H:6O5L:R6OY:WQYP:JEHC:Z3YS:CZF7
```

```
[root@k8s-master ~]# kube-config info
Architecture: armv7l
Kernel: Linux 4.1.13
CPU: 4 cores x 900 MHz

Used RAM Memory: 41 MiB
RAM Memory: 923 MiB

Used disk space: 1.5GB (1557644 KB)
Free disk space: 53GB (55302780 KB)

SD Card was built: 19-11-2015 22:13

kubernetes-on-arm: 
Latest commit: ccea1ef
Version: 0.5.8

systemd version: 227
docker version: 
operation not permitted
Error response from daemon: Cannot start container fafa71e602e21b66fc162b4caa0855731cb169820115920b2cf0d4cc0590b811: [8] System error: operation not permitted
kubernetes version: 
```

luxas, 2015-11-20:
Oh, you've run on the same problem as I. I ran into it a few minutes ago
I have no idea why this is happening. 
At least, it's because of `docker 1.9.0`, which was updated for pacman a few days ago.
The bad thing with pacman is that you can't downgrade easily

My issue: #18125

stevesloka, 2015-11-20:
Can I define the version to install for now? If I have to blow away my sd cards I'm cool with that, it's not too too painful.

luxas, 2015-11-20:
The problem is that one can't specify to install when using pacman. So, not either a sd card refresh would do it

stevesloka, 2015-11-23:
Ok gotcha, no worries. Thanks for the help!

closed 2015-11-23 by stevesloka

luxas, 2015-11-26:
@stevesloka See this: https://github.com/docker/docker/issues/18125#issuecomment-159736200
It solved the problem for me.

Mangirdaz, 2016-07-17:
Looks like with docker 1.10 same issue:
[root@k8s-master ~]# docker version
Client:
 Version:      1.10.0
 API version:  1.22
 Go version:   go1.4.3
 Git commit:   590d5108-unsupported
 Built:        Fri Feb 12 16:35:07 2016
 OS/Arch:      linux/arm
An error occurred trying to connect: Get http:///var/run/docker.sock/v1.22/version: read unix /var/run/docker.sock: connection reset by peer

Any suggestions before I start debugging? Rollback for docker?

luxas, 2016-07-18:
Which version of Kubernetes on ARM are you running @Mangirdaz?
If it's 0.7, please open a new issue.