luxas/kubernetes-on-arm #20: Get logs from node

stevesloka, 2015-11-23:
Just curious if you are able to get logs (i.e. `kubectl logs -f <pod>`). I'm getting connection refused. Just an fyi, I am working with hypriot now, and copying out your config / images. 

```
Error from server: Get https://k8s-node03:10250/containerLogs/default/my-nginx-5dgq1/my-nginx?follow=true: dial tcp 192.168.2.13:10250: connection refused
```

stevesloka, 2015-11-23:
So setting the ip and not `127.0.0.1` seems to help in the kubelet (k8s-worker).

luxas, 2015-11-23:
What do you mean by using hypriot now?
In the future one may support HypriotOS out-of-the-box

Now I am trying to statically compile docker for **both armv6 and armv7**
If I succeed with that, we won't depend on Arch Linux ARM packages anymore, and it would be easier to use it on other platforms.

But, to this question:
I assume you have to make a `kubconfig` file which points to the master.
Or just use: 
```
kubectl -s https://{MASTER_IP}:8080 get logs ...
```

stevesloka, 2015-11-23:
So I installed Hypriot on all my pi's, then used your images / setup to rebuild manually. I think k8s talks direct to each node to get logs and I couldn't lookup my nodes by name. I was thinking instead of 127.0.0.1 we might need the ip of the node (https://github.com/luxas/kubernetes-on-arm/blob/ccea1ef605d29637e96f9d8ce8265e8d6522bb46/sdcard/rootfs/kube-archlinux/usr/lib/systemd/system/k8s-worker.service#L9). No worries, I think I'm working now. Thanks!

closed 2015-11-23 by stevesloka