luxas/kubernetes-on-arm #22: kubectl: command not found

gyoho, 2015-11-27:
I successfully started the master node.
```
[root@pi1 ~]# docker ps
CONTAINER ID        IMAGE                       COMMAND                CREATED              STATUS              PORTS               NAMES
fdc7e6e5722a        kubernetesonarm/hyperkube   "/hyperkube schedule   21 seconds ago       Up 20 seconds                           k8s_scheduler.e5efe276_k8s-master-192.168.10.101_kube-system_447c171dfac8ae64dc585f8e9cbfa7e6_a5a8ae29
93ebc0b6ceac        kubernetesonarm/hyperkube   "/hyperkube apiserve   22 seconds ago       Up 20 seconds                           k8s_apiserver.c358020f_k8s-master-192.168.10.101_kube-system_447c171dfac8ae64dc585f8e9cbfa7e6_df2822eb
e57bf6126cb0        kubernetesonarm/hyperkube   "/hyperkube controll   22 seconds ago       Up 21 seconds                           k8s_controller-manager.7047e990_k8s-master-192.168.10.101_kube-system_447c171dfac8ae64dc585f8e9cbfa7e6_e12a78d6
a6ba8833893a        kubernetesonarm/pause       "/pause"               32 seconds ago       Up 31 seconds                           k8s_POD.7ad6c339_k8s-master-192.168.10.101_kube-system_447c171dfac8ae64dc585f8e9cbfa7e6_4b992c87
d0eb1acfa2cd        kubernetesonarm/hyperkube   "/hyperkube kubelet    About a minute ago   Up About a minute                       k8s-master
6bb0ad12d67d        kubernetesonarm/hyperkube   "/hyperkube proxy --   About a minute ago   Up About a minute                       k8s-worker-proxy
```
But, when I try to use Kubernetes it got this error...
```
[root@pi1 ~]# kubectl run my-nginx --image=luxas/nginx-test --replicas=3
-bash: kubectl: command not found
```

Am I missing something?

luxas, 2015-11-28:
Have you run `kube-config build-images` manually and interrupted it?
`kubectl` should be in `/etc/kubernetes/binaries`
Try `ls -la /etc/kubernetes/binaries`

That dir is empty, download manually
```
curl -sSL https://github.com/luxas/kubernetes-on-arm/releases/download/v0.5.5/binaries.tar.gz | tar -xz -C /etc/kubernetes/binaries
```

gyoho, 2015-11-28:
Yes, you're right. After re-executing `kube-config build-images`, I can ran `kubectl` now.
So, do I always execute the `kubectl` command on *the master node*?

By the way, I'm trying to remove some images to release some disk space. But, when I ran `docker rmi`, it threw me this error:
```
[root@pi2 ~]# docker rmi 22335e79f6fd
Error response from daemon: Error while getting parent image: image ID '' is invalid
Error: failed to remove images: [22335e79f6fd]
```
```
[root@pi2 ~]# docker images
armbuild/ubuntu                     latest              22335e79f6fd        11 months ago       163.2 MB
```
I googled that error message, but found no relevant answer...

Do you have any clue?

luxas, 2015-11-28:
It doesn't matter from where you run `kubectl`. 
For example, I am using my `amd64` laptop sometimes for controlling the Pis.
You just download the `kubectl` binary for the right platform and:
```
kubectl -s http://$MASTER_IP:8080 get no
```

I don't know why `docker rmi` doesn't remove your image, but try with the tag also in the end.

gyoho, 2015-11-28:
Hi Lucas,

I installed `kubectl` on my Mac and now I can reach the cluster nicely.
```
frontier:~ gyoho$ kubectl -s http://192.168.10.101:8080 cluster-info
Kubernetes master is running at http://192.168.10.101:8080

frontier:~ gyoho$ kubectl -s http://192.168.10.101:8080 get nodes
NAME             LABELS                                  STATUS
192.168.10.101   kubernetes.io/hostname=192.168.10.101   Ready
192.168.10.102   kubernetes.io/hostname=192.168.10.102   Ready
```

Now, I'm trying the `nginx` tutorial, but stuck with curling the server.
```
frontier:~ gyoho$ kubectl -s http://192.168.10.101:8080 run my-nginx --image=luxas/nginx-test --port=80
CONTROLLER   CONTAINER(S)   IMAGE(S)           SELECTOR       REPLICAS
my-nginx     my-nginx       luxas/nginx-test   run=my-nginx   1

frontier:~ gyoho$ kubectl -s http://192.168.10.101:8080 get pods
NAME             READY     STATUS    RESTARTS   AGE
my-nginx-bwl9i   1/1       Running   0          35m

frontier:~ gyoho$ kubectl -s http://192.168.10.101:8080 expose rc/my-nginx --port=80
NAME       LABELS         SELECTOR       IP(S)     PORT(S)
my-nginx   run=my-nginx   run=my-nginx             80/TCP

kubectl -s http://192.168.10.101:8080 get service
NAME         LABELS                                    SELECTOR       IP(S)        PORT(S)
kubernetes   component=apiserver,provider=kubernetes   <none>         10.0.0.1     443/TCP
my-nginx     run=my-nginx                              run=my-nginx   10.0.0.222   80/TCP

frontier:~ gyoho$ curl 10.0.0.222
curl: (7) Failed to connect to 10.0.0.222 port 80: Network is unreachable
```

It seems the problem is docker didn't expose the port appropriately because `docker ps` on the node shows no port mapping...
```
root@pi2 ~]# docker ps
CONTAINER ID        IMAGE                       COMMAND                CREATED             STATUS              PORTS               NAMES
9d1ba02e6335        luxas/nginx-test            "nginx -g 'daemon of   23 minutes ago      Up 23 minutes                           k8s_my-nginx.847bf08b_my-nginx-bwl9i_default_d5c1ba0d-95fa-11e5-80f5-b827ebf03a05_446c51f7
3b8363a2e369        kubernetesonarm/pause       "/pause"               24 minutes ago      Up 24 minutes                           k8s_POD.c637e3f5_my-nginx-bwl9i_default_d5c1ba0d-95fa-11e5-80f5-b827ebf03a05_3b55be35
694ea03a3ae6        kubernetesonarm/hyperkube   "/hyperkube kubelet    58 minutes ago      Up 58 minutes                           k8s-worker
0da310e668c3        kubernetesonarm/hyperkube   "/hyperkube proxy --   58 minutes ago      Up 58 minutes                           k8s-worker-proxy
```

Do you know how to perform the port mapping docker command: `-p 80:80 nginx` in `kubectl`?

luxas, 2015-11-28:
It's working as it should. 
However, your Mac doesn't belong to the `10.0.0.1/24` subnet, which k8s uses.
That means, only nodes in the cluster may access the services.

But there are ways to solve that:
```
# One way
# Only access the apiserver proxy
kubectl -s http://$MASTER_IP:8080 proxy --port=8080 &
curl -L http://localhost:8080/api/v1/proxy/namespaces/default/services/my-nginx

# The other way (this example is for a linux/amd64 machine, but just switch the images if necessary)
# This is much better. Gives access to all services
docker run --net=host -d --privileged -v /dev/net:/dev/net quay.io/coreos/flannel:0.5.4 /opt/bin/flanneld --etcd-endpoints=http://$MASTER_IP:4001

docker run --net=host -d --privileged gcr.io/google_containers/hyperkube:v1.1.1 /hyperkube proxy --master=http://$MASTER_IP:8080 --v=2

curl $NGINX_IP
```

gyoho, 2015-11-28:
My goal is to provide cloud-based iPython notebook service for users. In that sense, I think the second way is much more flexible.

Very simple questions I have are:
- Should I execute the two docker commands on every node?
- Is there any extra docker images I need to pull to do this? 

I'm not so sure about what you meant by:
*this example is for a linux/amd64 machine, but just switch the images if necessary*

Thanks for your help in advance.

luxas, 2015-11-28:
The second example is for "consuming" machines like my `amd64` laptop, not compute nodes.
I guess your users will have such machines, even though you run k8s on ARM, users may access the services from other platforms.

I meant that the images I used in the example will only run on linux/amd64, if you want other platforms, you will have to find corresponding images/binaries for those platforms. 
Or just run `flannel` and `kube-proxy` outside docker.

Hope it helps.
Remember to link to this project when you are demonstrating your cluster.
That would be nice, since I'm working hard on this :smile:

gyoho, 2015-11-28:
Of course, I will definitely refer to this project in my demo :) I really appreciate all your hard work!

I'm planing to have a node.js server in front-end to handle user authentication and return the IP address of the running iPython-netebook container. The node.js server is running outside of the cluster and probably I'm gonna use my Mac for that.

In that case, all I need to do on my node.js server machine are:
- Install kubectl
- Install docker
- Execute the two docker commands:
```
docker run --net=host -d --privileged -v /dev/net:/dev/net quay.io/coreos/flannel:0.5.4 /opt/bin/flanneld --etcd-endpoints=http://$MASTER_IP:4001
docker run --net=host -d --privileged gcr.io/google_containers/hyperkube:v1.1.1 /hyperkube proxy --master=http://$MASTER_IP:8080 --v=2
```

Just want to double check before digging it down.

luxas, 2015-11-28:
Unfortunately not. Main problem: `docker` doesn't run on Mac OS.
`docker daemon` just runs on Linux.
One option is downloading `go` to your Mac, and building `hyperkube` and `flannel` something like in: https://github.com/luxas/kubernetes-on-arm/blob/dev/images/kubernetesonarm/build/inbuild.sh
Line: 50-56 and 65-132
However, this is a very complicated process and probably not bug-free.

So the other option is running a VM with Ubuntu under your Mac and run the above commands.

gyoho, 2015-11-28:
I installed Docker by following [this page](https://docs.docker.com/mac/step_one/). Looks like the `Docker Toolbox` installs VirtualBox and `docker daemon` is running as a VM.

I started the docker CLI and executed the two commands. Now, I have two containers running on my Mac.
```
frontier:~ gyoho$ docker ps
CONTAINER ID        IMAGE                                       COMMAND                  CREATED             STATUS              PORTS               NAMES
298187382cf6        gcr.io/google_containers/hyperkube:v1.1.1   "/hyperkube proxy --m"   9 seconds ago       Up 9 seconds                            jovial_babbage
60e1cb84ee71        quay.io/coreos/flannel:0.5.4                "/opt/bin/flanneld --"   4 minutes ago       Up 4 minutes                            gloomy_mccarthy
```

However, when I curl the `nginx` server I still get the same error:
```
frontier:~ gyoho$ kubectl -s http://192.168.10.101:8080 get service
NAME         LABELS                                    SELECTOR       IP(S)        PORT(S)
kubernetes   component=apiserver,provider=kubernetes   <none>         10.0.0.1     443/TCP
my-nginx     run=my-nginx                              run=my-nginx   10.0.0.222   80/TCP
test         run=test                                  run=test       10.0.0.135   80/TCP
frontier:~ gyoho$ curl 10.0.0.222
curl: (7) Failed to connect to 10.0.0.222 port 80: Network is unreachable
```

Any thoughts?

luxas, 2015-11-28:
Have you checked the logs for the containers?

gyoho, 2015-11-28:
#### Logs

##### gcr.io/google_containers/hyperkube
```
frontier:~ gyoho$ docker logs 298187382cf6
I1128 21:36:14.708352       1 server.go:202] Running in resource-only container "/kube-proxy"
I1128 21:36:14.713941       1 iptables.go:173] Could not connect to D-Bus system bus: dial unix /var/run/dbus/system_bus_socket: no such file or directory
E1128 21:36:14.714279       1 server.go:355] Not trying iptables proxy: can't get Node "default": Get http://:8080/api/v1/nodes/default: dial tcp :8080: connection refused
I1128 21:36:14.714348       1 server.go:261] Using userspace Proxier.
I1128 21:36:14.714568       1 proxier.go:157] Setting proxy IP to 10.0.2.15 and initializing iptables
I1128 21:36:14.760957       1 server.go:274] Tearing down pure-iptables proxy rules. Errors here are acceptable.
E1128 21:36:14.765782       1 proxier.go:193] Error removing pure-iptables proxy rule: error checking rule: exit status 2: iptables v1.4.21: Couldn't load target `KUBE-SERVICES':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
E1128 21:36:14.766872       1 proxier.go:197] Error removing pure-iptables proxy rule: error checking rule: exit status 2: iptables v1.4.21: Couldn't load target `KUBE-SERVICES':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
E1128 21:36:14.768583       1 event.go:197] Unable to write event: 'Post http://:8080/api/v1/namespaces/default/events: dial tcp :8080: connection refused' (may retry after sleeping)
E1128 21:36:17.582903       1 event.go:197] Unable to write event: 'Post http://:8080/api/v1/namespaces/default/events: dial tcp :8080: connection refused' (may retry after sleeping)
E1128 21:36:27.585108       1 event.go:197] Unable to write event: 'Post http://:8080/api/v1/namespaces/default/events: dial tcp :8080: connection refused' (may retry after sleeping)
E1128 21:36:37.586819       1 event.go:197] Unable to write event: 'Post http://:8080/api/v1/namespaces/default/events: dial tcp :8080: connection refused' (may retry after sleeping)
E1128 21:36:47.588236       1 event.go:197] Unable to write event: 'Post http://:8080/api/v1/namespaces/default/events: dial tcp :8080: connection refused' (may retry after sleeping)
E1128 21:36:57.589047       1 event.go:197] Unable to write event: 'Post http://:8080/api/v1/namespaces/default/events: dial tcp :8080: connection refused' (may retry after sleeping)
E1128 21:37:07.590343       1 event.go:197] Unable to write event: 'Post http://:8080/api/v1/namespaces/default/events: dial tcp :8080: connection refused' (may retry after sleeping)
E1128 21:37:17.591655       1 event.go:197] Unable to write event: 'Post http://:8080/api/v1/namespaces/default/events: dial tcp :8080: connection refused' (may retry after sleeping)
E1128 21:37:27.592551       1 event.go:197] Unable to write event: 'Post http://:8080/api/v1/namespaces/default/events: dial tcp :8080: connection refused' (may retry after sleeping)
E1128 21:37:37.593525       1 event.go:197] Unable to write event: 'Post http://:8080/api/v1/namespaces/default/events: dial tcp :8080: connection refused' (may retry after sleeping)
E1128 21:37:47.594356       1 event.go:197] Unable to write event: 'Post http://:8080/api/v1/namespaces/default/events: dial tcp :8080: connection refused' (may retry after sleeping)
E1128 21:37:57.595344       1 event.go:197] Unable to write event: 'Post http://:8080/api/v1/namespaces/default/events: dial tcp :8080: connection refused' (may retry after sleeping)
E1128 21:37:57.595425       1 event.go:131] Unable to write event '&api.Event{TypeMeta:unversioned.TypeMeta{Kind:"", APIVersion:""}, ObjectMeta:api.ObjectMeta{Name:"default.141afb557e8cd34e", GenerateName:"", Namespace:"default", SelfLink:"", UID:"", ResourceVersion:"", Generation:0, CreationTimestamp:unversioned.Time{Time:time.Time{sec:0, nsec:0, loc:(*time.Location)(nil)}}, DeletionTimestamp:(*unversioned.Time)(nil), DeletionGracePeriodSeconds:(*int64)(nil), Labels:map[string]string(nil), Annotations:map[string]string(nil)}, InvolvedObject:api.ObjectReference{Kind:"Node", Namespace:"", Name:"default", UID:"default", APIVersion:"", ResourceVersion:"", FieldPath:""}, Reason:"Starting", Message:"Starting kube-proxy.", Source:api.EventSource{Component:"kube-proxy", Host:"default"}, FirstTimestamp:unversioned.Time{Time:time.Time{sec:63584343374, nsec:766986062, loc:(*time.Location)(0x2a668c0)}}, LastTimestamp:unversioned.Time{Time:time.Time{sec:63584343374, nsec:766986062, loc:(*time.Location)(0x2a668c0)}}, Count:1}' (retry limit exceeded!)
```

##### quay.io/coreos/flannel
```
frontier:~ gyoho$ docker logs 60e1cb84ee71
I1128 21:31:50.010361 00001 main.go:275] Installing signal handlers
I1128 21:31:50.010511 00001 main.go:130] Determining IP address of default interface
I1128 21:31:50.011007 00001 main.go:188] Using 10.0.2.15 as external interface
I1128 21:31:50.011057 00001 main.go:189] Using 10.0.2.15 as external endpoint
E1128 21:32:20.011544 00001 network.go:53] Failed to retrieve network config: client: etcd cluster is unavailable or misconfigured
E1128 21:32:51.013408 00001 network.go:53] Failed to retrieve network config: client: etcd cluster is unavailable or misconfigured
E1128 21:33:22.013846 00001 network.go:53] Failed to retrieve network config: client: etcd cluster is unavailable or misconfigured
[... cut ...]
E1128 21:49:23.061293 00001 network.go:53] Failed to retrieve network config: client: etcd cluster is unavailable or misconfigured
E1128 21:49:54.061811 00001 network.go:53] Failed to retrieve network config: client: etcd cluster is unavailable or misconfigured
E1128 21:50:25.062957 00001 network.go:53] Failed to retrieve network config: client: etcd cluster is unavailable or misconfigured
E1128 21:50:56.065799 00001 network.go:53] Failed to retrieve network config: client: etcd cluster is unavailable or misconfigured
```

I guess I need to start `etcd` container on the `master` node?

luxas, 2015-11-28:
Oh, I meant that you should replace `$MASTER_IP` with the real value, which is specific to your setup. Now, when you probably copied literally the command `$MASTER_IP` expands to ''

gyoho, 2015-11-28:
I stopped the two containers and started them by the following commands:
```
frontier:~ gyoho$ docker stop $(docker ps -a -q)
frontier:~ gyoho$ docker run --net=host -d --privileged -v /dev/net:/dev/net quay.io/coreos/flannel:0.5.4 /opt/bin/flanneld --etcd-endpoints=http://192.68.10.101:4001
frontier:~ gyoho$ docker run --net=host -d --privileged gcr.io/google_containers/hyperkube:v1.1.1 /hyperkube proxy --master=http://192.68.10.101:8080 --v=2

frontier:~ gyoho$ docker ps
CONTAINER ID        IMAGE                                       COMMAND                  CREATED             STATUS              PORTS               NAMES
56d10193b2a8        gcr.io/google_containers/hyperkube:v1.1.1   "/hyperkube proxy --m"   2 minutes ago       Up 2 minutes                            tender_khorana
b533d90aa030        quay.io/coreos/flannel:0.5.4                "/opt/bin/flanneld --"   4 minutes ago       Up 4 minutes                            evil_poitras
```

But, still same error `curl: (7) Failed to connect to 10.0.0.222 port 80: Network is unreachable`...

#### Logs

###### quay.io/coreos/flannel
```
frontier:~ gyoho$ docker logs b533d90aa030
I1128 22:04:33.310015 00001 main.go:275] Installing signal handlers
I1128 22:04:33.310153 00001 main.go:130] Determining IP address of default interface
I1128 22:04:33.310733 00001 main.go:188] Using 10.0.2.15 as external interface
I1128 22:04:33.310793 00001 main.go:189] Using 10.0.2.15 as external endpoint
E1128 22:05:03.311150 00001 network.go:53] Failed to retrieve network config: client: etcd cluster is unavailable or misconfigured
E1128 22:05:34.315400 00001 network.go:53] Failed to retrieve network config: client: etcd cluster is unavailable or misconfigured
E1128 22:06:05.370035 00001 network.go:53] Failed to retrieve network config: client: etcd cluster is unavailable or misconfigured
E1128 22:06:36.372207 00001 network.go:53] Failed to retrieve network config: client: etcd cluster is unavailable or misconfigured
E1128 22:07:07.374049 00001 network.go:53] Failed to retrieve network config: client: etcd cluster is unavailable or misconfigured
E1128 22:07:38.375156 00001 network.go:53] Failed to retrieve network config: client: etcd cluster is unavailable or misconfigured
E1128 22:08:09.376510 00001 network.go:53] Failed to retrieve network config: client: etcd cluster is unavailable or misconfigured
E1128 22:08:40.378273 00001 network.go:53] Failed to retrieve network config: client: etcd cluster is unavailable or misconfigured
E1128 22:09:11.379635 00001 network.go:53] Failed to retrieve network config: client: etcd cluster is unavailable or misconfigured
```

###### gcr.io/google_containers/hyperkube
```
frontier:~ gyoho$ docker logs 56d10193b2a8
I1128 22:06:01.726416       1 server.go:202] Running in resource-only container "/kube-proxy"
I1128 22:06:01.728554       1 iptables.go:173] Could not connect to D-Bus system bus: dial unix /var/run/dbus/system_bus_socket: no such file or directory
E1128 22:06:31.730228       1 server.go:355] Not trying iptables proxy: can't get Node "default": Get http://192.68.10.101:8080/api/v1/nodes/default: dial tcp 192.68.10.101:8080: i/o timeout
I1128 22:06:31.730274       1 server.go:261] Using userspace Proxier.
I1128 22:06:31.730641       1 proxier.go:157] Setting proxy IP to 10.0.2.15 and initializing iptables
I1128 22:06:31.751162       1 server.go:274] Tearing down pure-iptables proxy rules. Errors here are acceptable.
E1128 22:06:31.752653       1 proxier.go:193] Error removing pure-iptables proxy rule: error checking rule: exit status 2: iptables v1.4.21: Couldn't load target `KUBE-SERVICES':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
E1128 22:06:31.754006       1 proxier.go:197] Error removing pure-iptables proxy rule: error checking rule: exit status 2: iptables v1.4.21: Couldn't load target `KUBE-SERVICES':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
E1128 22:07:01.761514       1 event.go:197] Unable to write event: 'Post http://192.68.10.101:8080/api/v1/namespaces/default/events: dial tcp 192.68.10.101:8080: i/o timeout' (may retry after sleeping)
E1128 22:07:40.204941       1 event.go:197] Unable to write event: 'Post http://192.68.10.101:8080/api/v1/namespaces/default/events: dial tcp 192.68.10.101:8080: i/o timeout' (may retry after sleeping)
E1128 22:08:20.206903       1 event.go:197] Unable to write event: 'Post http://192.68.10.101:8080/api/v1/namespaces/default/events: dial tcp 192.68.10.101:8080: i/o timeout' (may retry after sleeping)
E1128 22:09:00.210040       1 event.go:197] Unable to write event: 'Post http://192.68.10.101:8080/api/v1/namespaces/default/events: dial tcp 192.68.10.101:8080: i/o timeout' (may retry after sleeping)
```

Thanks for your advice!

luxas, 2015-11-28:
It just seems like the apiserver and etcd is down. Can you confirm they are running properly?

gyoho, 2015-11-28:
Are they supposed to run on the `master` node?

luxas, 2015-11-28:
```
          your pi cluster                                                   other machines
-----------------------------                                    -----------------------------
|   -----> pimaster           |  <<----------may access services |          yourmac        |
|   |----- pinode1            |    |---------------------------- |    someone w/ linux     | 
|   |----- pinode2            |                                    ---------------------------
-----------------------------
pimaster => kube-config enable-master
pinode{1,2} => kube-config enable-worker
yourmac => linux vm => hyperkube, flannel
linux => hyperkube, flannel
```

pimaster controls the nodes, right?
pimaster hosts `etcd`, `apiserver` and the other master components

the "consumer" machines runs `flannel` and `kube-proxy` in `hyperkube`. Via those processes, they may access the cluster `services`

gyoho, 2015-11-29:
Thanks for the detailed explanation! It helped a lot. I think I started getting what micro-services are :)

I started `etcd` on the master node. And, now `etcd` and `apiserve` are running properly.
```
[root@pi1 ~]# docker run -d -P kubernetesonarm/etcd
2e7b880f03d3629fdf2ae29fdba913eb42bf2e8d0314d92eb1f900068d060440

[root@pi1 ~]# docker ps
CONTAINER ID        IMAGE                                                              COMMAND                CREATED             STATUS              PORTS               NAMES
2e7b880f03d3        kubernetesonarm/etcd                                               "/usr/bin/etcd --add   13 seconds ago      Up 11 seconds                           boring_bardeen
1063cf9ab39b        9964fbe9797a489ed893e8fd2e218d7b86dd5006193adffee70e8a548c79f379   "/bin/sh -c ./inbuil   13 minutes ago      Up 12 minutes                           angry_bartik
2a5f990716a5        kubernetesonarm/hyperkube                                          "/hyperkube controll   13 hours ago        Up 13 hours                             k8s_controller-manager.7047e990_k8s-master-192.168.10.101_kube-system_447c171dfac8ae64dc585f8e9cbfa7e6_d9c1d357
c7f084fc00e3        kubernetesonarm/hyperkube                                          "/hyperkube schedule   13 hours ago        Up 13 hours                             k8s_scheduler.e5efe276_k8s-master-192.168.10.101_kube-system_447c171dfac8ae64dc585f8e9cbfa7e6_dc8989ff
1aaa5eb4f997        kubernetesonarm/hyperkube                                          "/hyperkube apiserve   13 hours ago        Up 13 hours                             k8s_apiserver.c358020f_k8s-master-192.168.10.101_kube-system_447c171dfac8ae64dc585f8e9cbfa7e6_b7cdc955
b7fa0946de2c        kubernetesonarm/pause                                              "/pause"               13 hours ago        Up 13 hours                             k8s_POD.7ad6c339_k8s-master-192.168.10.101_kube-system_447c171dfac8ae64dc585f8e9cbfa7e6_32993ca6
a8b084fbeb51        kubernetesonarm/hyperkube                                          "/hyperkube kubelet    13 hours ago        Up 13 hours                             k8s-master
cbced2743fdc        kubernetesonarm/hyperkube                                          "/hyperkube proxy --   13 hours ago        Up 13 hours                             k8s-worker-
```
However,I'm still getting the same error: `curl: (7) Failed to connect to 10.0.0.222 port 80: Network is unreachable`...
#### Updated Logs
###### quay.io/coreos/flannel
```
E1129 06:37:09.226423 00001 network.go:53] Failed to retrieve network config: client: etcd cluster is unavailable or misconfigured
```
###### gcr.io/google_containers/hyperkube
```
E1128 22:14:20.238766       1 event.go:131] Unable to write event '&api.Event{TypeMeta:unversioned.TypeMeta{Kind:"", APIVersion:""}, ObjectMeta:api.ObjectMeta{Name:"default.141afcfc8b6ce21f", GenerateName:"", Namespace:"default", SelfLink:"", UID:"", ResourceVersion:"", Generation:0, CreationTimestamp:unversioned.Time{Time:time.Time{sec:0, nsec:0, loc:(*time.Location)(nil)}}, DeletionTimestamp:(*unversioned.Time)(nil), DeletionGracePeriodSeconds:(*int64)(nil), Labels:map[string]string(nil), Annotations:map[string]string(nil)}, InvolvedObject:api.ObjectReference{Kind:"Node", Namespace:"", Name:"default", UID:"default", APIVersion:"", ResourceVersion:"", FieldPath:""}, Reason:"Starting", Message:"Starting kube-proxy.", Source:api.EventSource{Component:"kube-proxy", Host:"default"}, FirstTimestamp:unversioned.Time{Time:time.Time{sec:63584345191, nsec:754162719, loc:(*time.Location)(0x2a668c0)}}, LastTimestamp:unversioned.Time{Time:time.Time{sec:63584345191, nsec:754162719, loc:(*time.Location)(0x2a668c0)}}, Count:1}' (retry limit exceeded!)
```
Am I missing something?

gyoho, 2015-11-29:
By the way, in this case a user needs to run the two container, `hyperkube` and `flannel`, on his/her machine in order to access the iPython-notebook container on one of the worker nodes?

As being cloud-based iPython notebook service, I would like to provide the service through a web browser. I'm thinking to use node.js server to handle user requests and communicate to the k8s cluster, so that I can decouple a user from the cluster. When a user requests iPython-notebook service, I want to return an IP address of the `rpi-ipython-notebook` container to the user so that he/she can just use a browser to access the service. (No `docker` or `kubectl` required here).

However, in current case a user cannot really access the container without `hyperkube` and `flannel`... 
Do you think my idea is possible?

luxas, 2015-11-29:
The only thing you have to do on the master is `kube-config enable-master`. You should not start etcd manually, because it's already running. See `system-docker ps`. There are two `docker` daemons running on every node.

`enable-master` starts
 - `etcd` in system-docker
 - `flannel` in system-docker => fetches `etcd` data
 - `docker`, who depends on `flannel` configuration
 - `k8s-master` in normal docker, spins up additional master containers as well
   - `apiserver`, stores all data in the `etcd` container
   - `controller-manager`, talks to `apiserver` and controls the cluster
   - `scheduler`, talks to `apiserver` and schedules pods to nodes
 - `kube-proxy`, talks to `apiserver` and makes the `10.0.0.x` ips accessible

`enable-worker` starts
 - `flannel` in system-docker => fetches `etcd` data from master
 - `docker`, who depends on `flannel` configuration
 - `k8s-worker` in normal docker => talks to master `apiserver`
 - `kube-proxy`, talks to master `apiserver` and makes the `10.0.0.x` ips accessible

the client:
 - `flannel` talks to master `etcd`, joins the k8s subnet `10.1.x.y`
 - `kube-proxy` talks to master `apiserver` and makes the `10.0.0.x` ips accessible
 - => all services accessible

As I said earlier, there is an another option too:
> curl -L http://$MASTER_IP:8080/api/v1/proxy/namespaces/default/services/my-nginx

So you could just run an nginx server on your Mac, which proxies: 
`http://$MAC_IP/ipython` => 
`curl -L http://$MASTER_IP:8080/api/v1/proxy/namespaces/default/services/ipython` for example

gyoho, 2015-11-29:
Thanks for the detailed explanation! It really helped!!
I didn't know `system-docker ps`command... I restarted the master node with: `kube-config disable-node` => `kube-config enable-master`, and now the master node shows:
```
[root@pi1 ~]# system-docker ps
CONTAINER ID        IMAGE                     COMMAND                CREATED             STATUS              PORTS               NAMES
784bf5beddef        kubernetesonarm/flannel   "/flanneld --etcd-en   12 minutes ago      Up 12 minutes                           k8s-flannel
11907d3a8093        kubernetesonarm/etcd      "/usr/bin/etcd --add   12 minutes ago      Up 12 minutes                           k8s-etcd

[root@pi1 ~]# docker ps
CONTAINER ID        IMAGE                       COMMAND                CREATED             STATUS              PORTS               NAMES
982e4ffef4a7        kubernetesonarm/hyperkube   "/hyperkube schedule   11 minutes ago      Up 11 minutes                           k8s_scheduler.e5efe276_k8s-master-192.168.10.101_kube-system_447c171dfac8ae64dc585f8e9cbfa7e6_b1a2d72b
54296d7e3ec5        kubernetesonarm/hyperkube   "/hyperkube apiserve   11 minutes ago      Up 11 minutes                           k8s_apiserver.c358020f_k8s-master-192.168.10.101_kube-system_447c171dfac8ae64dc585f8e9cbfa7e6_fbfcec7c
5011c08201e8        kubernetesonarm/hyperkube   "/hyperkube controll   11 minutes ago      Up 11 minutes                           k8s_controller-manager.7047e990_k8s-master-192.168.10.101_kube-system_447c171dfac8ae64dc585f8e9cbfa7e6_ef467ef8
b807e89bc703        kubernetesonarm/pause       "/pause"               11 minutes ago      Up 11 minutes                           k8s_POD.7ad6c339_k8s-master-192.168.10.101_kube-system_447c171dfac8ae64dc585f8e9cbfa7e6_b530d82e
5d9d5ac54fc5        kubernetesonarm/hyperkube   "/hyperkube kubelet    12 minutes ago      Up 12 minutes                           k8s-master
01262a2fd928        kubernetesonarm/hyperkube   "/hyperkube proxy --   12 minutes ago      Up 12 minutes                           k8s-worker-proxy
```

### Logs
#### Master
##### System-docker Process
###### kubernetesonarm/flannel
```
[root@pi1 ~]# system-docker logs 784bf5beddef
I1129 16:26:09.544367 00001 main.go:275] Installing signal handlers
I1129 16:26:09.546275 00001 main.go:130] Determining IP address of default interface
I1129 16:26:09.552574 00001 main.go:188] Using 192.168.10.101 as external interface
I1129 16:26:09.552819 00001 main.go:189] Using 192.168.10.101 as external endpoint
I1129 16:26:09.574503 00001 etcd.go:129] Found lease (10.1.72.0/24) for current IP (192.168.10.101), reusing
I1129 16:26:09.587740 00001 etcd.go:84] Subnet lease acquired: 10.1.72.0/24
I1129 16:26:09.602801 00001 udp.go:222] Watching for new subnet leases
I1129 16:26:09.612913 00001 udp.go:247] Subnet added: 10.1.27.0/24
I1129 16:33:02.766644 00001 udp.go:247] Subnet added: 10.1.27.0/24
```
###### kubernetesonarm/etcd
```
[root@pi1 ~]# system-docker logs 11907d3a8093
2015/11/29 16:26:04 etcd: already initialized as member before, starting as etcd member...
2015/11/29 16:26:04 etcd: listening for peers on http://localhost:2380
2015/11/29 16:26:04 etcd: listening for peers on http://localhost:7001
2015/11/29 16:26:04 etcd: listening for client requests on http://0.0.0.0:4001
2015/11/29 16:26:04 etcdserver: datadir is valid for the 2.0.1 format
2015/11/29 16:26:05 etcdserver: recovered store from snapshot at index 520052
2015/11/29 16:26:05 etcdserver: name = default
2015/11/29 16:26:05 etcdserver: data dir = /var/etcd/data
2015/11/29 16:26:05 etcdserver: member dir = /var/etcd/data/member
2015/11/29 16:26:05 etcdserver: heartbeat = 100ms
2015/11/29 16:26:05 etcdserver: election = 1000ms
2015/11/29 16:26:05 etcdserver: snapshot count = 10000
2015/11/29 16:26:05 etcdserver: advertise client URLs = http://127.0.0.1:4001
2015/11/29 16:26:05 etcdserver: loaded cluster information from store: default=http://localhost:2380,default=http://localhost:7001
2015/11/29 16:26:06 etcdserver: restart member ce2a822cea30bfca in cluster 7e27652122e8b2ae at commit index 524981
2015/11/29 16:26:06 raft: ce2a822cea30bfca became follower at term 7
2015/11/29 16:26:06 raft: newRaft ce2a822cea30bfca [peers: [ce2a822cea30bfca], term: 7, commit: 524981, applied: 520052, lastindex: 524981, lastterm: 7]
2015/11/29 16:26:07 raft: ce2a822cea30bfca is starting a new election at term 7
2015/11/29 16:26:07 raft: ce2a822cea30bfca became candidate at term 8
2015/11/29 16:26:07 raft: ce2a822cea30bfca received vote from ce2a822cea30bfca at term 8
2015/11/29 16:26:07 raft: ce2a822cea30bfca became leader at term 8
2015/11/29 16:26:07 raft.node: ce2a822cea30bfca elected leader ce2a822cea30bfca at term 8
2015/11/29 16:26:08 etcdserver: published {Name:default ClientURLs:[http://127.0.0.1:4001]} to cluster 7e27652122e8b2ae
```
##### Docker Process
###### scheduler
```
[root@pi1 ~]# docker logs 982e4ffef4a7
I1129 16:27:27.433461       1 factory.go:126] creating scheduler from algorithm provider 'DefaultProvider'
I1129 16:27:27.433903       1 factory.go:161] creating scheduler with fit predicates 'map[PodFitsPorts:{} PodFitsResources:{} NoDiskConflict:{} MatchNodeSelector:{} HostName:{}]' and priority functions 'map[LeastRequestedPriority:{} BalancedResourceAllocation:{} ServiceSpreadingPriority:{}]
```
###### controller-manager
```
[... 129 lines cut ...]

However, the first option (using proxy) worked nicely!
```
[... cut ...]
NAME          LABELS                                    SELECTOR          IP(S)        PORT(S)
kubernetes    component=apiserver,provider=kubernetes   <none>            10.0.0.1     443/TCP
my-nginx      run=my-nginx                              run=my-nginx      10.0.0.222   80/TCP
my-notebook   run=my-notebook                           run=my-notebook   10.0.0.206   8888/TCP
test          run=test                                  run=test          10.0.0.135   80/TCP
```

So, I guess I don't have to solve the error?

luxas, 2015-11-29:
Now you may try to connect to the subnet via your Mac if you want (the two flannel and proxy commands above), but using apiserver proxy should work in your use case.

Yes, in e.g. `kube-proxy` logs, `apiserver` is not reachable in the beginning, because **it hasn't started yet**. The `apiserver` will start up some seconds later and everything will work as expected.

gyoho, 2015-11-29:
Actually even after the `apiserver` started, the service is still unreachable...
```
frontier:~ gyoho$ docker ps
CONTAINER ID        IMAGE                                       COMMAND                  CREATED             STATUS              PORTS               NAMES
56d10193b2a8        gcr.io/google_containers/hyperkube:v1.1.1   "/hyperkube proxy --m"   20 hours ago        Up 20 hours                             tender_khorana
b533d90aa030        quay.io/coreos/flannel:0.5.4                "/opt/bin/flanneld --"   20 hours ago        Up 20 hours                             evil_poitras

frontier:~ gyoho$ kubectl -s http://192.168.10.101:8080 get service
NAME          LABELS                                    SELECTOR          IP(S)        PORT(S)
kubernetes    component=apiserver,provider=kubernetes   <none>            10.0.0.1     443/TCP
my-nginx      run=my-nginx                              run=my-nginx      10.0.0.222   80/TCP
my-notebook   run=my-notebook                           run=my-notebook   10.0.0.206   8888/TCP
test          run=test                                  run=test          10.0.0.135   80/TCP

frontier:~ gyoho$ curl 10.0.0.222
curl: (7) Failed to connect to 10.0.0.222 port 80: Network is unreachable
```

Nevertheless, since the 1st option (`proxy` and `/api/v1/proxy/namespaces/default/services/`) is working I'm going to stick with that option :)

Another issue I'm facing is that after accessing the service via browser and when clicking on `new notebook`, it threw me `the server could not find the requested resource` error...

[image]

[image]

Originally, by clicking `new notebook` it should:
1. Creates a new notebook
2. Open new browser page showing the new notebook

###### Example: started iPython notebook on a worker node
```
[root@pi2 ~]# docker run -p 8888:8888 -t -i ontouchstart/rpi-ipython-notebook
```

[image]

[image]

Any thoughts?

luxas, 2015-11-29:
You probably have to restart the services on the Mac.
However, I would suggest that you run a `nginx` proxy on you Mac, which just turns $MAC_IP:$SOME_PORT into $MASTER_IP:8080/api/v1/proxy/namespaces/default/services/[service]

If it works, this issue may be closed

gyoho, 2015-11-29:
Yes, I'm running `proxy` with `kubectl -s http://192.168.10.101:8080 proxy --port=8080`.
It's working nicely, and I think I can mark this case closed.

The issue I have about `ipython-notebook` is out of scope of this issue, but I would really appreciate if you could give me some suggestion or point me to some resource.

The problem I'm facing is I can reach the initial page by `$MASTER_IP:8080/api/v1/proxy/namespaces/default/services/[service]`, but cannot go further (links etc)...

Thanks very much again.

luxas, 2015-11-30:
I can't say, but I think that somewhere, you are using an absolute path (`/xxx`) instead of an relative (`../xxx`)
The absolute path resolves to `$MASTER_IP/xxx`, which indeed doesn't exist.

You said that running the web service directly on a port works, and I would suggest that you do that with `nginx`
> However, I would suggest that you run a nginx proxy on you Mac, which just forwards $MAC_IP:$SOME_PORT to $MASTER_IP:8080/api/v1/proxy/namespaces/default/services/[service]

closed 2015-11-30 by luxas