luxas/kubernetes-on-arm #24: [0.6.0 - Cubietruck] /etc/systemd/network/dns.network : Unknow value 'UseDomains' in section Network

nsteinmetz, 2015-12-01:
What I have before running `kube-config install` when I want to set up manually my network (no dhcp in dmz)

```
/etc/systemd/network/dns.network : Unknow value 'UseDomains' in section Network
```

nsteinmetz, 2015-12-01:
and in addition dns.network "conflicrs" with eth0.network for eth0 settings. 

Putting everyhting in eth0.network works (with UseDomains commented so far)

luxas, 2015-12-01:
Yeah, I'm wondering which way is the best to do this.
Now, I added a `dns.network` file which overrides the other `.network` files
`dns.network` uses `eth0` and `DHCP` by default.

The purpose is to include this line in `/etc/resolv.conf`
```
search default.svc.cluster.local svc.cluster.local cluster.local
```
That's `systemd`'s way of configuring this file.
Any suggestion to make this better?

nsteinmetz, 2015-12-01:
On 2nd cubietruck, I'll try to setup dns.network instead of using eth0.network (which is more straight forward and what I expected).

I would do more in fact what I did, that is to say add what was in the dns.network into the eth0.network. Seems more in line with official doc of systemd. Otherwhise, it should be mentionned in this project documentation.

nsteinmetz, 2015-12-01:
Ok, so on 2nd cubie:
* Editing only ``dns.network`` to switch from DHCP to static adress works and override eth0.network
* It still says that UseDomains is an unknown value in section network. but it does not prevent to set-up eth0.

nsteinmetz, 2015-12-02:
... and so:

``curl my-nginx.default.svc.cluster.local`` works but not ``curl my-nginx`` which is logical.

luxas, 2015-12-26:
I've fixed the `UseDomains` issue and documented that one should edit `dns.network`.

Thanks. (PS. `v0.6.2` is soon here)

closed 2015-12-26 by luxas