luxas/kubernetes-on-arm #25: [0.6.0 - Cubietruck] kube-config enable master fails

nsteinmetz, 2015-12-01:
It failed on first init like we had previously when flannel/etcd failed to init properly

I could trap at least:

```
[root@kube1 ~]# systemctl status -l etcd
* etcd.service - Etcd Master Data Store for Kubernetes Apiserver
   Loaded: loaded (/usr/lib/systemd/system/etcd.service; enabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since Tue 2015-12-01 21:09:38 CET; 4min 58s ago
 Main PID: 734 (code=exited, status=0/SUCCESS)

Dec 01 21:09:03 kube1 docker[734]: 2015-12-01 20:09:03.824524 N | etcdserver: set the initial cluster version to 2.2
Dec 01 21:09:03 kube1 docker[734]: 2015-12-01 20:09:03.825445 I | etcdserver: published {Name:default ClientURLs:[http://127.0.0.1:4001]} to cluster 7e27652122e8b2ae
Dec 01 21:09:30 kube1 bash[735]: Error:  client: etcd cluster is unavailable or misconfigured
Dec 01 21:09:31 kube1 bash[735]: error #0: client: endpoint http://127.0.0.1:4001 exceeded header timeout
Dec 01 21:09:35 kube1 systemd[1]: etcd.service: Control process exited, code=exited status=4
Dec 01 21:09:35 kube1 docker[734]: 2015-12-01 20:09:35.929778 N | osutil: received terminated signal, shutting down...
Dec 01 21:09:38 kube1 docker[780]: k8s-etcd
Dec 01 21:09:38 kube1 systemd[1]: Failed to start Etcd Master Data Store for Kubernetes Apiserver.
Dec 01 21:09:38 kube1 systemd[1]: etcd.service: Unit entered failed state.
Dec 01 21:09:38 kube1 systemd[1]: etcd.service: Failed with result 'exit-code'.
```

Then flannel fails as etcd is not there and the rest too.

I disabled nodev via ``kube-config disable-node``, checked that all docker images are there (they are in lastest and 0.6.0 versions) and re-run ``kube-config enable-master`` which was successful this time.

Will double check on next cubie.

nsteinmetz, 2015-12-01:
Damned, it worked on 1st attempt on my 2nd cubie - so closing it unless you think something should be investigated. I'll keep kubie1 as-is if needed.

closed 2015-12-01 by nsteinmetz

luxas, 2015-12-02:
Maybe a timeout or something was too small. Don't know, but we'll keep this as a ref if it happens again