luxas/kubernetes-on-arm #26: [0.6.0 - RPI1] Worker does not survive a reboot

nsteinmetz, 2015-12-02:
So worker was corretly initialised, then I made a reboot to check if it restarts correctly and it failed.

```
[root@rpi1 ~]# docker ps
An error occurred trying to connect: Get http://%2Fvar%2Frun%2Fdocker.sock/v1.21/containers/json: read unix @->/var/run/docker.sock: read: connection reset by peer
```

```
[root@rpi1 ~]# systemctl status docker -l
* docker.service - Docker Application Container Engine
   Loaded: loaded (/usr/lib/systemd/system/docker.service; enabled; vendor preset: disabled)
  Drop-In: /usr/lib/systemd/system/docker.service.d
           `-docker-flannel.conf
   Active: activating (auto-restart) (Result: resources) since Wed 2015-12-02 22:44:17 CET; 374ms ago
     Docs: https://docs.docker.com

Dec 02 22:44:17 rpi1 systemd[1]: docker.service: Unit entered failed state.
Dec 02 22:44:17 rpi1 systemd[1]: docker.service: Failed with result 'resources'.
Warning: docker.service changed on disk. Run 'systemctl daemon-reload' to reload units.
```

```
[root@rpi1 ~]# systemctl status flannel -l
* flannel.service - Flannel Overlay Network for Kubernetes
   Loaded: loaded (/usr/lib/systemd/system/flannel.service; enabled; vendor preset: disabled)
   Active: failed (Result: timeout) since Wed 2015-12-02 22:39:40 CET; 5min ago
  Process: 229 ExecStartPre=/usr/bin/mkdir -p /var/lib/kubernetes/flannel (code=exited, status=0/SUCCESS)
  Process: 226 ExecStartPre=/usr/bin/rm -rf /var/lib/kubernetes/flannel (code=exited, status=0/SUCCESS)
  Process: 219 ExecStartPre=/usr/bin/docker -H unix:///var/run/system-docker.sock rm k8s-flannel (code=exited, status=0/SUCCESS)
  Process: 160 ExecStartPre=/usr/bin/docker -H unix:///var/run/system-docker.sock kill k8s-flannel (code=exited, status=2)

Dec 02 22:37:42 rpi1 systemd[1]: Starting Flannel Overlay Network for Kubernetes...
Dec 02 22:39:37 rpi1 systemd[1]: flannel.service: Start-pre operation timed out. Terminating.
Dec 02 22:39:40 rpi1 docker[219]: k8s-flannel
Dec 02 22:39:40 rpi1 systemd[1]: Failed to start Flannel Overlay Network for Kubernetes.
Dec 02 22:39:40 rpi1 systemd[1]: flannel.service: Unit entered failed state.
Dec 02 22:39:40 rpi1 systemd[1]: flannel.service: Failed with result 'timeout'.
```

```
[root@rpi1 ~]# systemctl status k8s-worker -l
* k8s-worker.service - The Worker Components for Kubernetes
   Loaded: loaded (/usr/lib/systemd/system/k8s-worker.service; enabled; vendor preset: disabled)
   Active: activating (start-post) since Wed 2015-12-02 22:45:45 CET; 9s ago
  Process: 1187 ExecStop=/usr/bin/docker stop k8s-worker k8s-worker-proxy (code=exited, status=1/FAILURE)
  Process: 1228 ExecStartPre=/usr/bin/mkdir -p /etc/kubernetes/static/worker (code=exited, status=0/SUCCESS)
  Process: 1217 ExecStartPre=/usr/bin/docker rm k8s-worker k8s-worker-proxy (code=exited, status=1/FAILURE)
  Process: 1207 ExecStartPre=/usr/bin/docker kill k8s-worker k8s-worker-proxy (code=exited, status=1/FAILURE)
 Main PID: 1232 (bash);         : 1233 (docker)
   CGroup: /system.slice/k8s-worker.service
           |-1232 /bin/bash -c source /etc/kubernetes/k8s.conf; exec docker run                --name=k8s-worker                        --net=host                          -v /etc/kubernetes/static/worker:/etc/kubernetes/manifests              -v /:/rootfs:ro                        -v /sys:/sys:ro                        -v /dev:/dev                         -v /var/lib/docker/:/var/lib/docker:rw                   -v /var/lib/kubelet:/var/lib/kubelet:rw                  -v /var/run:/var/run:rw                      --privileged                         --pid=host                          kubernetesonarm/hyperkube /hyperkube kubelet                  --allow-privileged=true                      --containerized                        --pod_infra_container_image=kubernetesonarm/pause               --api-servers=http://192.168.8.110:8080                  --cluster-dns=10.0.0.10                      --cluster-domain=cluster.local                     --v=2                           --address=127.0.0.1                       --enable-server                        --hostname-override=$(/usr/bin/hostname -i | /usr/bin/awk '{print $1}')          --config=/etc/kubernetes/manifests
           |-1234 /bin/bash -c source /etc/kubernetes/k8s.conf; exec docker run                --name=k8s-worker                        --net=host                          -v /etc/kubernetes/static/worker:/etc/kubernetes/manifests              -v /:/rootfs:ro                        -v /sys:/sys:ro                        -v /dev:/dev                         -v /var/lib/docker/:/var/lib/docker:rw                   -v /var/lib/kubelet:/var/lib/kubelet:rw                  -v /var/run:/var/run:rw                      --privileged                         --pid=host                          kubernetesonarm/hyperkube /hyperkube kubelet                  --allow-privileged=true                      --containerized                        --pod_infra_container_image=kubernetesonarm/pause               --api-servers=http://192.168.8.110:8080                  --cluster-dns=10.0.0.10                      --cluster-domain=cluster.local                     --v=2                           --address=127.0.0.1                       --enable-server                        --hostname-override=$(/usr/bin/hostname -i | /usr/bin/awk '{print $1}')          --config=/etc/kubernetes/manifests
           |-1235 /usr/bin/hostname -i
           |-1236 /usr/bin/awk {print $1}
           `-control
             `-1233 /usr/bin/docker run -d --name=k8s-worker-proxy --net=host --privileged kubernetesonarm/hyperkube /hyperkube proxy --master=http://192.168.8.110:8080 --v=2

Dec 02 22:45:45 rpi1 systemd[1]: Starting The Worker Components for Kubernetes...
Dec 02 22:45:47 rpi1 docker[1207]: An error occurred trying to connect: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.21/containers/k8s-worker/kill?signal=KILL: read unix @->/var/run/docker.sock: read: connection reset by peer
Dec 02 22:45:49 rpi1 docker[1207]: An error occurred trying to connect: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.21/containers/k8s-worker-proxy/kill?signal=KILL: read unix @->/var/run/docker.sock: read: connection reset by peer
Dec 02 22:45:49 rpi1 docker[1207]: Error: failed to kill containers: [k8s-worker k8s-worker-proxy]
Dec 02 22:45:52 rpi1 docker[1217]: An error occurred trying to connect: read unix @->/var/run/docker.sock: read: connection reset by peer
Dec 02 22:45:54 rpi1 docker[1217]: An error occurred trying to connect: read unix @->/var/run/docker.sock: read: connection reset by peer
Dec 02 22:45:54 rpi1 docker[1217]: Error: failed to remove containers: [k8s-worker k8s-worker-proxy]
```
Seems I need to make:

```
systemctl daemon-reload
systemctl start flannel
```
 
From this point, seems docker and k8s-worker is restarting and working well.

nsteinmetz, 2015-12-02:
On cubietruck, I don't have the ```systemctl daemon-reload``` issue but I have the ```flannel``` one.

luxas, 2015-12-09:
Is this the same timeout issue we discussed earlier?
The timeout will be 8 secs in `v0.6.2`
Could we close this?

nsteinmetz, 2015-12-09:
Let's try with 8 and we'll see.

closed 2015-12-09 by nsteinmetz