luxas/kubernetes-on-arm #27: Unable to start using Hypriot OS

GrantGochnauer, 2015-12-03:
I started with the latest 0.6.1 Hypriot OS and perhaps there is a compatibility issue but when trying to start the Kubelet with the latest compiled binaries here, the api server isn't accepting connections for some reason. I'm using Docker 1.9.1. 

Here are the startup logs if anyone has any insights into why this might be:

```
black-pearl# sudo docker run -i -t --name=kube --net=host -v /var/run/docker.sock:/var/run/docker.sock kubernetesonarm/hyperkube /bin/bash
root@black-pearl:/# /hyperkube kubelet --pod_infra_container_image="kubernetesonarm/pause" --api-servers=http://localhost:8080 --v=2 --address=0.0.0.0 --enable-server --config=/etc/kubernetes/manifests-multi
W1203 10:43:15.123409       6 server.go:585] Could not load kubeconfig file /var/lib/kubelet/kubeconfig: stat /var/lib/kubelet/kubeconfig: no such file or directory. Trying auth path instead.
W1203 10:43:15.124034       6 server.go:547] Could not load kubernetes auth path /var/lib/kubelet/kubernetes_auth: stat /var/lib/kubelet/kubernetes_auth: no such file or directory. Continuing with defaults.
I1203 10:43:15.124816       6 plugins.go:71] No cloud provider specified.
I1203 10:43:15.125006       6 server.go:468] Successfully initialized cloud provider: "" from the config file: ""
I1203 10:43:15.125866       6 manager.go:128] cAdvisor running in container: "/system.slice/docker-0978411d3717bd4d48ab4461fbadfd784c5aa9a645b47240950481886756aa6e.scope"
I1203 10:43:16.474856       6 fs.go:108] Filesystem partitions: map[/dev/root:{mountpoint:/etc/resolv.conf major:179 minor:2 fsType: blockSize:0}]
E1203 10:43:16.498711       6 machine.go:126] failed to get cache information for node 0: open /sys/devices/system/cpu/cpu0/cache: no such file or directory
E1203 10:43:16.499532       6 machine.go:86] Failed to get system UUID: open /proc/device-tree/vm,uuid: no such file or directory
I1203 10:43:16.531350       6 manager.go:163] Machine: {NumCores:4 CpuFrequency:900000 MemoryCapacity:970452992 MachineID:2d763621e59c44cbbaf0f8fd51e18e91 SystemUUID: BootID:655b09ab-2e4e-41ea-9224-d5481904e8c2 Filesystems:[{Device:/dev/root Capacity:29970837504}] DiskMap:map[179:0:{Name:mmcblk0 Major:179 Minor:0 Size:32026656768 Scheduler:deadline}] NetworkDevices:[{Name:eth0 MacAddress:b8:27:eb:40:30:33 Speed:100 Mtu:1500} {Name:wlan0 MacAddress:74:da:38:5c:6a:63 Speed:0 Mtu:1500}] Topology:[{Id:0 Memory:0 Cores:[{Id:0 Threads:[0] Caches:[]} {Id:1 Threads:[1] Caches:[]} {Id:2 Threads:[2] Caches:[]} {Id:3 Threads:[3] Caches:[]}] Caches:[]}] CloudProvider:Unknown InstanceType:Unknown}
I1203 10:43:16.537569       6 manager.go:169] Version: {KernelVersion:4.1.12-hypriotos-v7+ ContainerOsVersion:Raspbian GNU/Linux 8 (jessie) DockerVersion:1.9.1 CadvisorVersion: CadvisorRevision:}
I1203 10:43:16.544711       6 server.go:485] Using root directory: /var/lib/kubelet
W1203 10:43:16.547045       6 server.go:490] failed to set oom_score_adj to -999: write /proc/self/oom_score_adj: permission denied
I1203 10:43:16.556465       6 server.go:798] Adding manifest file: /etc/kubernetes/manifests-multi
I1203 10:43:16.557125       6 file.go:47] Watching path "/etc/kubernetes/manifests-multi"
I1203 10:43:16.561219       6 server.go:808] Watching apiserver
I1203 10:43:17.941575       6 plugins.go:56] Registering credential provider: .dockercfg
I1203 10:43:17.949930       6 plugins.go:262] Loaded volume plugin "kubernetes.io/aws-ebs"
I1203 10:43:17.950359       6 plugins.go:262] Loaded volume plugin "kubernetes.io/empty-dir"
I1203 10:43:17.950653       6 plugins.go:262] Loaded volume plugin "kubernetes.io/gce-pd"
I1203 10:43:17.950936       6 plugins.go:262] Loaded volume plugin "kubernetes.io/git-repo"
I1203 10:43:17.951372       6 plugins.go:262] Loaded volume plugin "kubernetes.io/host-path"
I1203 10:43:17.951665       6 plugins.go:262] Loaded volume plugin "kubernetes.io/nfs"
I1203 10:43:17.952300       6 plugins.go:262] Loaded volume plugin "kubernetes.io/secret"
I1203 10:43:17.952653       6 plugins.go:262] Loaded volume plugin "kubernetes.io/iscsi"
I1203 10:43:17.953033       6 plugins.go:262] Loaded volume plugin "kubernetes.io/glusterfs"
I1203 10:43:17.953378       6 plugins.go:262] Loaded volume plugin "kubernetes.io/persistent-claim"
I1203 10:43:17.954007       6 plugins.go:262] Loaded volume plugin "kubernetes.io/rbd"
I1203 10:43:17.954336       6 plugins.go:262] Loaded volume plugin "kubernetes.io/cinder"
I1203 10:43:17.954637       6 plugins.go:262] Loaded volume plugin "kubernetes.io/cephfs"
I1203 10:43:17.955051       6 plugins.go:262] Loaded volume plugin "kubernetes.io/downward-api"
I1203 10:43:17.955639       6 plugins.go:262] Loaded volume plugin "kubernetes.io/fc"
I1203 10:43:17.955963       6 plugins.go:262] Loaded volume plugin "kubernetes.io/flocker"
E1203 10:43:17.959349       6 kubelet.go:756] Image garbage collection failed: unable to find data for container /
W1203 10:43:17.971264       6 kubelet.go:775] Failed to move Kubelet to container "/kubelet": mkdir /sys/fs/cgroup/cpu/kubelet: read-only file system
I1203 10:43:17.971670       6 kubelet.go:777] Running in container "/kubelet"
E1203 10:43:17.975968       6 event.go:197] Unable to write event: 'Post http://localhost:8080/api/v1/namespaces/default/events: dial tcp 127.0.0.1:8080: connection refused' (may retry after sleeping)
I1203 10:43:17.976475       6 server.go:72] Starting to listen on 0.0.0.0:10250
I1203 10:43:17.976920       6 server.go:757] Started kubelet
I1203 10:43:17.977057       6 server.go:89] Starting to listen read-only on 0.0.0.0:10255
I1203 10:43:18.147873       6 kubelet.go:2293] Recording NodeReady event message for node black-pearl
I1203 10:43:18.148279       6 kubelet.go:2293] Recording NodeHasSufficientDisk event message for node black-pearl
I1203 10:43:18.148539       6 kubelet.go:869] Attempting to register node black-pearl
I1203 10:43:18.156045       6 kubelet.go:872] Unable to register black-pearl with the apiserver: Post http://localhost:8080/api/v1/nodes: dial tcp 127.0.0.1:8080: connection refused
I1203 10:43:18.470475       6 kubelet.go:2293] Recording NodeReady event message for node black-pearl
I1203 10:43:18.471410       6 kubelet.go:2293] Recording NodeHasSufficientDisk event message for node black-pearl
I1203 10:43:18.472090       6 kubelet.go:869] Attempting to register node black-pearl
I1203 10:43:18.478213       6 kubelet.go:872] Unable to register black-pearl with the apiserver: Post http://localhost:8080/api/v1/nodes: dial tcp 127.0.0.1:8080: connection refused
I1203 10:43:18.966961       6 kubelet.go:2293] Recording NodeReady event message for node black-pearl
I1203 10:43:18.968500       6 kubelet.go:2293] Recording NodeHasSufficientDisk event message for node black-pearl
I1203 10:43:18.968975       6 kubelet.go:869] Attempting to register node black-pearl
I1203 10:43:18.975396       6 kubelet.go:872] Unable to register black-pearl with the apiserver: Post http://localhost:8080/api/v1/nodes: dial tcp 127.0.0.1:8080: connection refused
I1203 10:43:19.857823       6 kubelet.go:2293] Recording NodeReady event message for node black-pearl
I1203 10:43:19.858466       6 kubelet.go:2293] Recording NodeHasSufficientDisk event message for node black-pearl
I1203 10:43:19.858713       6 kubelet.go:869] Attempting to register node black-pearl
I1203 10:43:19.864068       6 kubelet.go:872] Unable to register black-pearl with the apiserver: Post http://localhost:8080/api/v1/nodes: dial tcp 127.0.0.1:8080: connection refused
I1203 10:43:20.088875       6 factory.go:236] Registering Docker factory
I1203 10:43:20.099986       6 factory.go:93] Registering Raw factory
E1203 10:43:20.849985       6 event.go:197] Unable to write event: 'Post http://localhost:8080/api/v1/namespaces/default/events: dial tcp 127.0.0.1:8080: connection refused' (may retry after sleeping)
I1203 10:43:21.540537       6 kubelet.go:2293] Recording NodeReady event message for node black-pearl
I1203 10:43:21.541315       6 kubelet.go:2293] Recording NodeHasSufficientDisk event message for node black-pearl
I1203 10:43:21.541606       6 kubelet.go:869] Attempting to register node black-pearl
I1203 10:43:21.547994       6 kubelet.go:872] Unable to register black-pearl with the apiserver: Post http://localhost:8080/api/v1/nodes: dial tcp 127.0.0.1:8080: connection refused
I1203 10:43:21.986156       6 manager.go:1006] Started watching for new ooms in manager
I1203 10:43:21.993452       6 oomparser.go:183] oomparser using systemd
[... cut ...]
I1203 10:43:22.091720       6 manager.go:1707] Need to restart pod infra container for "k8s-master-black-pearl_kube-system" because it is not found
W1203 10:43:22.097917       6 manager.go:108] Failed to updated pod status: error updating status for pod "k8s-master-black-pearl_kube-system": Get http://localhost:8080/api/v1/namespaces/kube-system/pods/k8s-master-black-pearl: dial tcp 127.0.0.1:8080: connection refused
E1203 10:43:22.997428       6 manager.go:1864] Failed to create pod infra container: open /proc/8748/cgroup: no such file or directory; Skipping pod "k8s-master-black-pearl_kube-system"
```

GrantGochnauer, 2015-12-03:
I did change the master.json config to run the securityContext as root instead of as privileged because with the default settings, I get this error on startup:

```
 file.go:123] Can't process config file "/etc/kubernetes/manifests-multi/master.json": invalid pod: [spec.containers[3].securityContext.privileged: forbidden '<*>(0x56e8af08)true']
```

GrantGochnauer, 2015-12-03:
OK, this can be closed. etcd wasn't running. The only change I had to make to get this to work was to modify the original command to add:

```
--allow_privileged=true
```

luxas, 2015-12-03:
See the `systemd` service files: https://github.com/luxas/kubernetes-on-arm/tree/master/sdcard/rootfs/kube-archlinux/usr/lib/systemd/system

There you'll find the right commands to use.

There the commands for starting the containers are. If you are running on many nodes, start `etcd` first in `system-docker`, then `flannel` in `system-docker`, then main `docker` with `flannel` settings, after that `k8s-master/worker`.

If you just test it on one node, just start `etcd` and `k8s-master`, as you said.

This can be quite scary with all these service and dependencies.
However, I'm probably going to make a `.deb` package for HypriotOS, so installing the whole system is just two commands away.

Hope it helps!

GrantGochnauer, 2015-12-03:
Ah @luxas this is super helpful and makes a ton a sense.  :+1:

GrantGochnauer, 2015-12-04:
I've adapted your systemd settings but for some reason it appears that kuber isn't creating the kuber namespace properly. Did I miss another initialization step?

```
E1204 15:58:21.824083   25444 event.go:188] Server rejected event '&api.Event{TypeMeta:unversioned.TypeMeta{Kind:"", APIVersion:""}, ObjectMeta:api.ObjectMeta{Name:"k8s-master-127.0.0.1.141cc060b8cc4521", GenerateName:"", Namespace:"kube-system", SelfLink:"", UID:"", ResourceVersion:"", Generation:0, CreationTimestamp:unversioned.Time{Time:time.Time{sec:0, nsec:0, loc:(*time.Location)(nil)}}, DeletionTimestamp:(*unversioned.Time)(nil), DeletionGracePeriodSeconds:(*int64)(nil), Labels:map[string]string(nil), Annotations:map[string]string(nil)}, InvolvedObject:api.ObjectReference{Kind:"Pod", Namespace:"kube-system", Name:"k8s-master-127.0.0.1", UID:"fd6de061dd407e666e5047943d7f91fc", APIVersion:"v1", ResourceVersion:"", FieldPath:"spec.containers{controller-manager}"}, Reason:"Started", Message:"Started with docker id 734d71a45bfa", Source:api.EventSource{Component:"kubelet", Host:"127.0.0.1"}, FirstTimestamp:unversioned.Time{Time:time.Time{sec:63584841501, nsec:756245281, loc:(*time.Location)(0x1caf0e8)}}, LastTimestamp:unversioned.Time{Time:time.Time{sec:63584841501, nsec:756245281, loc:(*time.Location)(0x1caf0e8)}}, Count:1}': 'namespaces "kube-system" not found' (will not retry!)
E1204 15:58:21.852363   25444 kubelet.go:1361] Failed creating a mirror pod "k8s-master-127.0.0.1_kube-system": namespaces "kube-system" not found
E1204 15:58:31.008328   25444 kubelet.go:1361] Failed creating a mirror pod "k8s-master-127.0.0.1_kube-system": namespaces "kube-system" not found
E1204 15:58:40.810850   25444 kubelet.go:1361] Failed creating a mirror pod "k8s-master-127.0.0.1_kube-system": namespaces "kube-system" not found
E1204 15:58:50.839258   25444 kubelet.go:1361] Failed creating a mirror pod "k8s-master-127.0.0.1_kube-system": namespaces "kube-system" not found
E1204 15:59:00.813521   25444 kubelet.go:1361] Failed creating a mirror pod "k8s-master-127.0.0.1_kube-system": namespaces "kube-system" not found
```

luxas, 2015-12-04:
I have no idea why it says that `kube-system` is created automatically. 
Only on GCE, I would guess. At least, it **doesn't**, as you noticed.
As a workaround, `kube-config` in this project automatically creates this.
So, you have to create it yourself.

However, I've managed to make a `.deb` package of this project!
I will maybe upload an experimental package of `v0.6.0` and hope that I may support it in `v0.6.2`

GrantGochnauer, 2015-12-04:
Awesome. You rock! Thank you.

GrantGochnauer, 2015-12-05:
One other thing that may be related to running on Hypriot OS is that I see this in the logs every few mins:

```
Image garbage collection failed: non-existent label "docker-images"
```

luxas, 2015-12-05:
This happens on all containerized `kubelet' enviroinments, even on my `amd64` machine
It's a main kubernetes issue, and google folks are working on the best UX and behavior when using docker for spinning up k8s.

GrantGochnauer, 2015-12-05:
More debugging, I'm noticing this when I spin up containers and my health checks are failing and am not sure whether this is related:

```
hairpin.go:49] Unable to find pair interface, setting up all interfaces: exec: "ethtool"
```
I see that every time a new pod is created

luxas, 2015-12-05:
Quite funny, I made an issue about this some hours ago: https://github.com/kubernetes/kubernetes/issues/18251

It's just a line that must be changed: `apt-get install ethtool` 
I've updated this to `v0.6.2`

I had noticed this one, but if you continue digging I'm sure you'll find something odd I haven't encountered :smile:

luxas, 2015-12-26:
Merry chrismas @GrantGochnauer!

Here comes a small chrismas present: `v0.6.2` :)
I hope you like it.

I'm exited to hear what you will come up with.
Now HypriotOS is supported OOTB, so have fun

GrantGochnauer, 2015-12-27:
This is amazing news!! Thank you so much for all your hard work. So you have any suggestions on migrating from the manual install to the deb installer? Can I just install the deb version on top of my already installed kuberentes? Thank you!

luxas, 2015-12-27:
It's a little risky, but I think we can make it

I came up with an amazing scipt here for checking that: 
```
cd /
wget https://github.com/luxas/kubernetes-on-arm/releases/download/v0.6.2/kube-systemd.deb
dpkg -c kube-systemd.deb | awk '{print $6}' | xargs ls -d 2>&1 | grep -v "No such file"

# If you want to install
sudo dpkg -i kube-systemd.deb
```
That script will output all files you have in your system that also the deb has => probably conficts
If you after the install want to check if you are missing some files, remove the `-v` in the `grep` command.

Try to resolve the conflicts if any, and enjoy :smile:

GrantGochnauer, 2015-12-30:
No conflicts. Worked like a charm!

```
[root:~]# kube-config info
Architecture: armv7l
Kernel: Linux 4.1.12
CPU: 4 cores x 900 MHz

Used RAM Memory: 613 MiB
RAM Memory: 925 MiB

Used disk space: 11GB (10735504 KB)
Free disk space: 17GB (17161224 KB)

SD Card/deb package was built: 28-12-2015 18:57

kubernetes-on-arm: 
Latest commit: 247134b
Version: 0.6.2

systemd version: v215
docker version: v1.9.1
```

luxas, 2015-12-31:
I'm glad to hear that! Closing as fixed :)

closed 2015-12-31 by luxas