luxas/kubernetes-on-arm #29: kube-config delete-data - strange message

nsteinmetz, 2015-12-04:
Hi,

I tested the ``delete-data`` and even if it seems to work, as I got the doc of umount, seems somethinkg was not done correctly.

```
[root@kube1 ~]# kube-config delete-data
Do you want to delete all Kubernetes data about this cluster? m(ove) is default, which moves the directories to {,old}. y deletes them and n exits [M/n/y] y

Usage:
 umount [-hV]
 umount -a [options]
 umount [options] <source> | <directory>

Unmount filesystems.

Options:
 -a, --all               unmount all filesystems
 -A, --all-targets       unmount all mountpoints for the given device in the
                           current namespace
 -c, --no-canonicalize   don't canonicalize paths
 -d, --detach-loop       if mounted loop device, also free this loop device
     --fake              dry run; skip the umount(2) syscall
 -f, --force             force unmount (in case of an unreachable NFS system)
 -i, --internal-only     don't call the umount.<type> helpers
 -n, --no-mtab           don't write to /etc/mtab
 -l, --lazy              detach the filesystem now, clean up things later
 -O, --test-opts <list>  limit the set of filesystems (use with -a)
 -R, --recursive         recursively unmount a target with all its children
 -r, --read-only         in case unmounting fails, try to remount read-only
 -t, --types <list>      limit the set of filesystem types
 -v, --verbose           say what is being done

 -h, --help     display this help and exit
 -V, --version  output version information and exit

For more details see umount(8).
Deleted all Kubernetes data
```

luxas, 2015-12-04:
Oops, I didn't think about if no containers was running...
Source: 
https://github.com/luxas/kubernetes-on-arm/blob/master/sdcard/rootfs/kube-archlinux/usr/bin/kube-config.sh#L522

The reason is when containers are running: `kubelet` mounts a `tmpfs` in every container, `serviceAccount secrets`. So I `umount` them so `/var/lib/kubelet` may be removed.
However, when no `tmpfs` are mounted it just runs `umount` => help message.

Will fix

luxas, 2015-12-05:
Should work with latest `dev` now. Thanks for the report. (But do not test `dev` right now)
I have to close issues when I fix them, otherwise I will forget which ones I have/haven't fixed :smile:

closed 2015-12-05 by luxas