luxas/kubernetes-on-arm #30: Pi Cubed/Project31

KurtStam, 2015-12-04:
@luxas Hi Luxas,

I just created an Organization called Project31 and I'm trying to bundle the efforts on Kubernetes deployment on Raspberry Pi. I just finished my own little project (https://kurtstam.github.io/2015/12/04/Build-a-Kubernetes-Cluster-on-Raspberry-Pi.html) and I'm hungry for more. So if you are interested to becoming a contributor please let me know and we can learn from each other.

Cheers,

--Kurt

luxas, 2015-12-05:
Hello @KurtStam!
Glad you notified me.

Just some thoughts:
 - `hyperkube` should be compiled dynamically, if one wants `cAdvisor`: https://github.com/kubernetes/kubernetes/issues/8772
 - What kind of networking solution do you use? I'm using `flannel`
 - Do you plan to deploy addons? I have `dns` and `registry` for the moment.
 - How do one start and stop k8s easily?
 - Try to make your images as small as possible
 - Which OSes does it run on? `HypriotOS`?
 - What about `serviceAccounts` and `secrets`?

My project is 2-3 months old now, and I think k8s and ARM are a match made in heaven.
Just now I have quite a lot to do with this project and mainline k8s.
Some things I'm adding in `v0.6.2`
 - A `.deb` package, so you guys may use it on `HypriotOS` out-of-the-box.
 - Statically linked docker daemon for Raspberry Pi 1, removes dependencies on other package managers. Optional to use.
 - `iptables` proxying by default for `kube-proxy`
 - Banana Pro support

Just say if there is something that you think could be added.

KurtStam, 2015-12-07:
Hi @luxas, now that I have a working setup I'm going to try to get OpenShift v3 to work (which comes with a Kubernetes implementation). Out of the box our http://fabric8.io project runs on top of that, though it's meant to run on plain vanilla Kubernetes too, however we do seem to be using some of the OpenShifts Kubernetes extensions. To answer some of your questions; yes I'm running on HypriotOS and it's cool to see that you are working on a .deb package for Kubernetes :).

>iptables proxying by default for kube-proxy

Do you  mean an kube-proxy implementation based on iptables, or iptables sitting in front of kube-proxy? I'm thinking about a fabric8/apiman gateway implementation of the kube-proxy so we can add policies on services. So I'm curious to see what you are thinking..

luxas, 2015-12-07:
I mean that by default, `kube-proxy` uses `userspace` proxying (don't exactly know what it does) but since `v1.1` another proxy mode is usable and much better; pure `iptables`.
With `iptables` proxying, I'll test [this](https://github.com/kubernetes/contrib/tree/master/scale-demo)
I've already added it to `dev`

KurtStam, 2015-12-08:
Ah ok, interestingly enough on Openshift 3 we have an yet another
implementation, based on http://www.haproxy.org, and I'm thinking about
using the apiman gateway - so we can apply all sorts of policies (think
authentication, rate limiting, metering.

Iptables is faster or has more features?

luxas, 2015-12-08:
`iptables` is much faster AFAIU