luxas/kubernetes-on-arm #31: kube-config upgrade issue

Millyna, 2015-12-13:
Hello,
Great project, been following for a while!
Unsure if this is myself, I'm new to this and learning. However, I've just done a kibe-config upgrade command which resulted in the failed updates

'kube-config upgrade
Upgrading the system
:: Synchronizing package databases...
error: failed retrieving file 'core.db' from mirror.archlinuxarm.org : Resolving timed out after 10522 milliseconds
error: failed to update core (download library error)'

Everything else ran perfectly and then I ran 'docker ps', which results in the error:
Error response from daemon: client is newer than server (client API version: 1.21, server API version: 1.19)

Would this be due to the error upgrading the system? Apologies, pretty new to all of this :)

Thank you.

nsteinmetz, 2015-12-13:
Based on https://github.com/luxas/kubernetes-on-arm/issues/21#issuecomment-164116065 ; you should not upgrade docker to 1.9.x as there are some bugs around. You should stick with docker 1.7.x

I did the same error and had to rewrite my sdcard from scratch to make it work again. Maybe the instructions from @luxas can avoid this step...

Millyna, 2015-12-13:
Eeeek panic mode! I've just started from scratch already once tonight :')
Thank you @nsteinmetz , I'll see if @luxas steps will work.

nsteinmetz, 2015-12-13:
What I do so far : just install it as it is and do not upgrade. And it should work :)

luxas, 2015-12-13:
@Millyna Hello! Yes, I know you and @nsteinmetz have been following the project nearly since I uploaded it. I am very thankful for it :smile:

Well, the bug. 
In my issue https://github.com/docker/docker/issues/18125 I think we've come to the conclusion that it only affects `ARM` and docker built with `go1.5.1`.

Arch Linux ARM maintainers always offer the latest version, and there is no easy way to roll back.
Some steps to workaround:
```
pacman -R docker
curl -sSL https://s3.amazonaws.com/docker-armv7/docker-1:1.7.1-2-armv7h.pkg.tar.xz > /var/cache/pacman/pkg/docker-1:1.7.1-2-armv7h.pkg.tar.xz
pacman -U  /var/cache/pacman/pkg/docker-1:1.7.1-2-armv7h.pkg.tar.xz --noconfirm

reboot
```
That should roll back to `v1.7.1` which works.
I exactly know the panic feeling you described with the SD Cards for example
Such things happens :joy:

Thanks to this issue, I'll add a warning for Arch Linux ARMv7 users when they do `kube-config upgrade`

nsteinmetz, 2015-12-13:
Btw, what is weird for me is that my RPI1 upgraded to docker 1.9.1 and do not have the issue I had with my cubietruck.

Is there a difference on the way it's built on ARMv6/RPi vs ARMv7?

luxas, 2015-12-13:
Yes, and I have no idea why. It's just ARMv7 I think

Millyna, 2015-12-13:
Ok so I've done all of the above but.....
Now another issue :p 
```
kube-config enable-worker 
Disabling k8s if it is running
Removed symlink /etc/systemd/system/multi-user.target.wants/flannel.service.
Removed symlink /etc/systemd/system/multi-user.target.wants/k8s-worker.service.
Checks so all images are present
Transferring images to system-docker, if necessary
Created symlink from /etc/systemd/system/multi-user.target.wants/flannel.service to /usr/lib/systemd/system/flannel.service.
Starting the worker containers
Created symlink from /etc/systemd/system/multi-user.target.wants/k8s-worker.service to /usr/lib/systemd/system/k8s-worker.service.
Job for k8s-worker.service failed because the control process exited with error code. See "systemctl status k8s-worker.service" and "journalctl -xe" for details.
Worker Kubernetes services enabled
```

Also on my ```enable-master```, it shows as no containers when i ```docker ps```

luxas, 2015-12-22:
@Millyna Oops. In some way I missed your last comment. 
I'm so sorry, did you get it working?

Otherwise, try to run `systemctl status k8s-worker` when you reproduce.

luxas, 2016-01-09:
@Millyna Do you think this is solved?

Millyna, 2016-01-09:
Opps! Sorry @luxas, I forgot to reply to this. I ended up re-doing my cards. Just nothing would get it working. :)

luxas, 2016-01-09:
@Millyna Yes, (un)fortunately, one has to do that often
I make new SD Cards all the time to test things, but it's also the way I want it.
I'm really glad the Raspberry Pi Foundation didn't decide to use some kind of internal flash storage for example.

When one starts with a newly-flashed SD Card, the system is as fast as it can be, that's one good thing, but of course it can be annoying sometimes.

I'll close this then if you don't mind

closed 2016-01-09 by luxas