luxas/kubernetes-on-arm #33: [0.6.2-Cubietruck] kube-config enable-worker [master-ip] does not work

nsteinmetz, 2015-12-27:
Hi,

On master side, everything is fine so far but on worker side, I can't override the 127.0..0.1, seems the argument is not taken into account.

```
[root@kube2 ~]# kube-config enable-worker 192.168.8.110
Disabling k8s if it is running
Removed symlink /etc/systemd/system/multi-user.target.wants/flannel.service.
Using master ip: 127.0.0.1
curl: (28) Resolving timed out after 5516 milliseconds
Kubernetes Master IP is required.
Value right now: 127.0.0.1
Exiting...

Command:
kube-config enable-worker [master-ip]
Checks so all images are present
Transferring images to system-docker, if necessary
Created symlink from /etc/systemd/system/multi-user.target.wants/flannel.service to /usr/lib/systemd/system/flannel.service.
Job for docker.service failed because a configured resource limit was exceeded. See "systemctl status docker.service" and "journalctl -xe" for details.
```

If I edit manually ``/etc/kubernetes/k8s.conf`` file, then ``kube-config enable-worker`` works as expected.

What is also weird I think is that if it did not find the master node, it should just exit and should not try to launch the process which of course fails. You should have an exit statement [line 393 of kube-config.sh file](https://github.com/luxas/kubernetes-on-arm/blob/9d36023f1c03c2f971472c2d6c049c327c356b22/sdcard/rootfs/kube-systemd/usr/bin/kube-config.sh#L393)

luxas, 2015-12-28:
@nsteinmetz Yeah, you're right. The `exit` statement should be there, but somehow I forgot it...

I must investigate why it doesn't take the argument.
Thanks for this bug report.

luxas, 2015-12-28:
Fixed with 1a9a2b5867c07c26d8fc74ec14298da3705e5e15 and 88a805e06c9b7dce76a3d92ffd331c1a359fcbec

closed 2015-12-28 by luxas