luxas/kubernetes-on-arm #34: install fails on hypriot

TimCook1, 2015-12-28:
$ dpkg -i kube-systemd.deb
(Reading database ... 53240 files and directories currently installed.)
Preparing to unpack kube-systemd.deb ...
Unpacking kubernetes-on-arm (0.6.2-2) ...
dpkg: error processing archive kube-systemd.deb (--install):
 trying to overwrite '/README.md', which is also in package occi 0.6.0
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Errors were encountered while processing:
 kube-systemd.deb

HypriotOS: root@hostname in ~
$ uname -na
Linux hostname 4.1.12-hypriotos-v7+ #2 SMP PREEMPT Tue Nov 3 19:44:55 UTC 2015 armv7l GNU/Linux

HypriotOS: root@hostname in ~
$ cat /etc/*release
profile: hypriot
image_build: 20151115-132854
image_commit: ac0496d13f8034bbb13fb60af4d3988773aec860
kernel_build: 20151103-193133
kernel_commit: 4b796cf43834bf552a86cc267a97069edc8dfe08

PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

luxas, 2015-12-28:
@TimCook1 Hello! 
Could you kindly rename the README.md file with this command and try again?
```
mv /README.md /README_occi.md
dpkg -i kube-systemd.deb
```

Thanks for using this project! 
I will fix this conflict in the next release

TimCook1, 2015-12-28:
$ dpkg -i kube-systemd.deb
(Reading database ... 53240 files and directories currently installed.)
Preparing to unpack kube-systemd.deb ...
Unpacking kubernetes-on-arm (0.6.2-2) ...
dpkg: error processing archive kube-systemd.deb (--install):
 trying to overwrite '/README.md', which is also in package occi 0.6.0
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Errors were encountered while processing:
 kube-systemd.deb
HypriotOS: root@hostname in ~
$ ls /
bin  boot  debian-binary  dev  etc  home  lib  lost+found  media  mnt  opt  proc  README.md.occi  root  run  sbin  srv  sys  tmp  usr  var

luxas, 2015-12-28:
OK, I will update the `.deb` package today or tomorrow. 
Stay tuned for updates to: [the release page](https://github.com/luxas/kubernetes-on-arm/releases/tag/v0.6.2) and the `kube-systemd-3` package.

luxas, 2015-12-28:
You may try --force-all also

luxas, 2015-12-28:
This should be fixed now with: 247134b7719177cf26bd02f64cc43703ae644143

closed 2016-01-02 by luxas