luxas/kubernetes-on-arm #42: dev branch error: updatefile: command not found

kyletravis, 2016-01-04:
Happy New Year's!

I ran into an issue today when deploying the dev branch from scratch.
```
[root@kube-master ~]# kube-config enable-addon dns
namespace "kube-system" created
/etc/kubernetes/dynamic-env/os/archlinux.sh: line 59: updatefile: command not found
```

A quick edit to these two lines in /etc/kubernetes/dynamic-env/os/archlinux.sh changing `updatefile` to `updateline` did the trick

```
[root@kube-master conf]# grep update /etc/kubernetes/dynamic-env/os/archlinux.sh
	updateline /etc/systemd/network/dns.network "Domains" "Domains=default.svc.$DNS_DOMAIN svc.$DNS_DOMAIN $DNS_DOMAIN"
	updateline /etc/systemd/network/dns.network "DNS" "DNS=$DNS_IP;"

```

After the mod, all is well:

```
[root@kube-master ~]# kube-config enable-addon dns
replicationcontroller "kube-dns-v8" created
service "kube-dns" created
Started addon: dns

[root@kube-master ~]# curl my-nginx.default.svc.cluster.local
<p>WELCOME TO NGINX</p>
```

luxas, 2016-01-04:
Yep, came across this now. Just a copy-paste error.
I hadn't had time to test until now. If you don't find any other bugs (and not I either), I'm gonna release `v0.6.3` as a stability and refactoring release soon.

I don't know if you noticed, but now it's possible to change `cluster.local` to something more interesting.

kyletravis, 2016-01-05:
Cool cool, I did see the domain name tweak - good thought! Haven't had a chance to take it for a spin yet with something other than `cluster.local`.  A couple of quick items:

1. I noticed is that in a multi node setup with DNS + registry add ons enabled, the worker nodes hang when curling the DNS text entry instead of the IP on nodes where the DNS isn't deployed. This may be a Kube thing (aka my needing to understand it better) and not an ARM specific item. Use case is the worker nodes pulling a home grown docker image w/o the need to use Docker Hub.

1. My resolv.conf was also a little wonky on my master following the install. It looks like it was trying to use Google Public DNS IPv6 as well. It was this:

 ```
[root@kube-master ~]# cat /etc/resolv.conf 
# This file is managed by systemd-resolved(8). Do not edit.
#
# Third party programs must not access this file directly, but
# only through the symlink at /etc/resolv.conf. To manage
# resolv.conf(5) in a different way, replace the symlink by a
# static file or a different symlink.

nameserver 8.8.8.8
nameserver 8.8.4.4
nameserver 2001:4860:4860::8888
# Too many DNS servers configured, the following entries may be ignored.
nameserver 2001:4860:4860::8844
```

 Removing the symbolic link at /etc/resolv.conf and changing it to a flat text file with the contents below did the trick:

 ```
[root@kube-master ~]# cat /etc/resolv.conf 
# This file is managed by systemd-resolved(8). Do not edit.
#
# Third party programs must not access this file directly, but
# only through the symlink at /etc/resolv.conf. To manage
# resolv.conf(5) in a different way, replace the symlink by a
# static file or a different symlink.

nameserver 10.0.0.10
nameserver 8.8.8.8
nameserver 8.8.4.4
search default.svc.cluster.local svc.cluster.local cluster.local
```

luxas, 2016-01-08:
I ran into the same Google DNS issue. I don't know why it handled like that, but it was a semi-colon in the wrong place in the code that caused it at least. a86d46085702efac5c06fe1e398d4d300ef1b07a

luxas, 2016-01-09:
Now `v0.6.3` is released! Check it out and see if it fixes the problems. 
It should, but one can't know for sure :)

Millyna, 2016-01-09:
Has the upgrade feature been fixed so it's not upgrading docker? Saves me re-doing the cards :P

luxas, 2016-01-09:
Thanks to your issue I've made this: https://github.com/luxas/kubernetes-on-arm/blob/master/sdcard/rootfs/kube-systemd/etc/kubernetes/dynamic-env/os/archlinux.sh#L44

It doesn't update if it's a ARMv7 board, where `docker` is corrupt. I have an issue for this in docker, and we've come to the conclusion that it's `go1.5` that's causing it.

TL;DR; It should be safe, but you'll never know :smile:

luxas, 2016-01-12:
@kyletravis I'm closing this issue as solved (primary issue: the updatefile typo)
1st secondary issue: could you open up a new issue if you comes across that again (with DNS and registry)?
2nd secondary issue fixed (with `/etc/resolv.conf` and `archlinux`)

closed 2016-01-12 by luxas