luxas/kubernetes-on-arm #43: need to install bridge-utils?

saturnism, 2016-01-06:
/usr/lib/systemd/system/docker.service.d/docker-flannel.conf uses brctl, which is available in package bridge-utils.  this isn't installed by default when i ran the script. is this a known issue?

thanx

luxas, 2016-01-06:
Yep, noticed it and included it in `v0.6.3`
Now it's [here](https://github.com/luxas/kubernetes-on-arm/blob/dev/sdcard/rootfs/kube-systemd/etc/kubernetes/dynamic-env/os/systemd.sh#L40) and [here](https://github.com/luxas/kubernetes-on-arm/blob/dev/sdcard/rootfs/kube-systemd/etc/kubernetes/dynamic-env/os/hypriotos.sh#L10).

But thanks for notifying anyway.
I will release `v0.6.3` soon, after I have tested it a little bit more.

luxas, 2016-01-09:
25 December this was committed: 693b620a019839a60728166c209878ce36027d2e for `systemd`
1 January this was committed: fa141eda117d90a20560450dacc7a292ba3948e6 for `hypriot`

So this should be fixed now in `v0.6.3`
Thanks for using this project!
Closing...

closed 2016-01-09 by luxas