luxas/kubernetes-on-arm #45: Only one volume being mounted?

sokoow, 2016-01-08:
Hey, I have a following config:
---
kind: PersistentVolume
apiVersion: v1
metadata:
  name: pv0001
  labels:
    type: local
spec:
  capacity:
    storage: 1Gi
  accessModes:
    - ReadWriteOnce
  hostPath:
    path: "/volumes/first"
---
kind: PersistentVolume
apiVersion: v1
metadata:
  name: pv0002
  labels:
    type: local
spec:
  capacity:
    storage: 1Gi
  accessModes:
    - ReadWriteOnce
  hostPath:
    path: "/volumes/second"
---
Kind: Pod
...
      volumeMounts:
      - mountPath: "/etc/sudoers.d"
        name: mypv1
      volumeMounts:
      - mountPath: "/var/lib"
        name: mypv2
  volumes:
    - name: mypv1
      persistentVolumeClaim:
       claimName: myclaim1
    - name: mypv2
      persistentVolumeClaim:
       claimName: myclaim2

kubectl get pv 
NAME      LABELS       CAPACITY   ACCESSMODES   STATUS    CLAIM              REASON    AGE
pv0001    type=local   1Gi        RWO           Bound     default/myclaim2             6s
pv0002    type=local   1Gi        RWO           Bound     default/myclaim1             6s

kubectl get pvc
NAME       LABELS    STATUS    VOLUME    CAPACITY   ACCESSMODES   AGE
myclaim1   <none>    Bound     pv0002    1Gi        RWO           55s
myclaim2   <none>    Bound     pv0001    1Gi        RWO           54s

FIrst thing that doesn't work when we have volumes attached is kube exec:

kubectl exec -t -i volume-test /bin/bash
error: unexpected response status code 500 (Internal Server Error)

But when I exec directly from docker, here's what I have:

mounts:

        {
            "Name": "f1736e7aa452778344c10acb0df3ae08ab5283136a7b59d6d26d39a58993bd73",
            "Source": "/media/docker/volumes/f1736e7aa452778344c10acb0df3ae08ab5283136a7b59d6d26d39a58993bd73/_data",
            "Destination": "/etc/sudoers.d",
            "Driver": "local",
            "Mode": "",
            "RW": true
        },
        {
            "Source": "/volumes/first",
            "Destination": "/var/lib",
            "Mode": "",
            "RW": true
        }

So, first volume is being mounted to a docker directory, second volume is actually a mount, not a volume, but seems to be ok. Could somebody test same please? Just create two volumes, two claims, and attach to docker.

sokoow, 2016-01-08:
And it's exactly same with nfs mounts...

luxas, 2016-01-08:
First off, could you add three of these ` characters before and after the code snippets. 
It will increase readability a lot. I may also do it for you.
Example: 
```
# here's some code
```

Second, I could investigate it. 
It seems like a k8s issue though. 

Will test it on my `amd64` machine first, and I could then make an issue if you want me to check it out

sokoow, 2016-01-09:
Ok, I think I found it, so instead:

```
      volumeMounts:
      - mountPath: "/etc/sudoers.d"
        name: mypv1
      volumeMounts:
      - mountPath: "/var/lib"
        name: mypv2
```

there should be:

```
      volumeMounts:
        - name: mypv1
          mountPath: "/etc/tor"
        - name: mypv2
          mountPath: "/var/lib/tor"
```

definitely a k8s issue, it parses only once when you specify multiple volumeMounts statements

luxas, 2016-01-09:
Yes, feel free to create an issue if you want
Closing this as working then, reopen if it doesn't work

closed 2016-01-09 by luxas