luxas/kubernetes-on-arm #47: 0.6.3

TimCook1, 2016-01-09:
Did you not change the package version in the .deb file?  After downloading the 0.6.3 and installing it, dpkg -l still shows version 0.6.2-2

ii  kubernetes-on-arm                                                         0.6.2-2                                   armhf                                     Kubernetes for ARM devices

luxas, 2016-01-09:
Have you downloaded the latest `.deb` package?
I did this now:
```
$ wget https://github.com/luxas/kubernetes-on-arm/releases/download/v0.6.3/kube-                             systemd.deb
$ dpkg --info kube-systemd.deb
 new debian package, version 2.0.
 size 188806 bytes: control archive=5030 bytes.
     365 bytes,    11 lines      control
   15419 bytes,   167 lines      md5sums
 Package: kubernetes-on-arm
 Version: 0.6.3-1
 Architecture: armhf
 Maintainer: Lucas Kaldstrom <lucas.kaldstrom@hotmail.co.uk>
 Installed-Size: 1180
 Recommends: ca-certificates, cgroupfs-mount, git, systemd, iptables | cgroup-lite, xz-utils
 Section: admin
 Priority: optional
 Homepage: https://github.com/luxas/kubernetes-on-arm
 Description: Kubernetes for ARM devices
```

But had you uninstalled the previous version `v0.6.2` before installing `v0.6.3`?

TimCook1, 2016-01-09:
PEBKAC, shouldn't install things this early in the morning apparently, it's fine.

closed 2016-01-09 by TimCook1