luxas/kubernetes-on-arm #48: verify image pull policy

saturnism, 2016-01-11:
I noticed that the image pull policy is not set in the static pod yamls. The images pulled also don't have specific tags.  IIRC, this means that Kubernetes may be applying the default pull always policy.

Would it be possible to do either one of the following?
* specify the image pull policy
* tag the docker images and specify the tags in the pod yamls?

luxas, 2016-01-12:
The images are tagged with e.g. `v0.6.2`. It's just not tagged in the `json` file.

If I get you right, you mean that this could be a problem when newer images (e.g. `v0.6.5`) appear and k8s has `imagePullPolicy: always`, so `v0.6.2` images will be replaced with `v0.6.5` on older clusters.

Is that what you meant?
Seems resonable I could specify an explicit `imagePullPolicy: IfNotPresent` and then the user may pull newer images if he/she want to.

saturnism, 2016-01-16:
yup! sgtm.

luxas, 2016-01-19:
Fixed with: 5449eae4c565c4075ae826efbf0598cdf534df86

closed 2016-01-19 by luxas