luxas/kubernetes-on-arm #49: kube-ui not in DockerHub

saturnism, 2016-01-20:
When trying to use the kube-ui addon, I'm getting the follow error when the pod is trying to start:
```
Failed	Failed to pull image "kubernetesonarm/kube-ui": Error: image kubernetesonarm/kube-ui:latest not found
```

luxas, 2016-01-20:
In Readme: 
> kube-ui was removed, because the Kubernetes team shifted focus to dashboard.

It was a bit outdated and didn't fully work, so I removed it. Just kept the `.yaml` file for reference. Maybe I should write a README also in that folder about it.

But I have good news, dashboard is released soon and it's working on ARM also

luxas, 2016-01-21:
@saturnism Now I've uploaded a snapshot of today's `dashboard`. So here's the Web UI
```
mkdir -p /etc/kubernetes/source/addons/dashboard
curl -sSL https://raw.githubusercontent.com/luxas/kubernetes-on-arm/dev/addons/dashboard/dashboard-rc.yaml > /etc/kubernetes/source/addons/dashboard/dashboard-rc.yaml
curl -sSL https://raw.githubusercontent.com/luxas/kubernetes-on-arm/dev/addons/dashboard/dashboard-svc.yaml > /etc/kubernetes/source/addons/dashboard/dashboard-svc.yaml
kube-config enable-addon dashboard
```

w0mbat, 2016-01-27:
I tried it on a Pi 2 with latest hypriot and got the following error:
[image]

luxas, 2016-01-28:
Strange, does it persist? 
Could maybe try to build the image again...

w0mbat, 2016-01-29:
well, yes - now I have another error while starting the pod:

```
kubectl logs kubernetes-dashboard-qei1f -p --namespace=kube-system
2016/01/29 12:39:49 Starting HTTP server on port 9090
2016/01/29 12:39:49 Creating API client for 
2016/01/29 12:39:49 open /var/run/secrets/kubernetes.io/serviceaccount/token: no such file or directory
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xb code=0x1 addr=0x0 pc=0x19304]

goroutine 1 [running]:
main.CreateHeapsterRESTClient(0x0, 0x0, 0x0, 0x113c47c0, 0x0, 0x0)
	/dashboard/.tmp/backend/src/github.com/kubernetes/dashboard/heapsterclient.go:36 +0xdc
main.main()
	/dashboard/.tmp/backend/src/github.com/kubernetes/dashboard/dashboard.go:49 +0x244

goroutine 17 [chan receive]:
github.com/golang/glog.(*loggingT).flushDaemon(0xfaab50)
	/dashboard/Godeps/_workspace/src/github.com/golang/glog/glog.go:882 +0x60
created by github.com/golang/glog.init.1
	/dashboard/Godeps/_workspace/src/github.com/golang/glog/glog.go:410 +0x2cc
```

From the message I guess, it tries to load a token and panics - will investigate further. If you guys have a quick fix, just tell me :)

luxas, 2016-01-29:
@w0mbat Are you running this on a Pi 1? I've tested it on a Pi 2 only

w0mbat, 2016-01-29:
no its pi 2 only. are you running hypriot?
where should this secret token come from?

luxas, 2016-01-29:
Are you running a `v0.6.3` cluster?

w0mbat, 2016-01-29:
@luxas I used the hypriot 0.6.1 image and installed all available updates

luxas, 2016-02-10:
You can try with the latest manifest that uses `gcr.io/google_containers/kubernetes-dashboard-arm:v0.1.0`. We built an official ARM binary with the dashboard team in the same way as I built `kubernetesonarm/dashboard`. Maybe it's working with that one @w0mbat?

luxas, 2016-02-14:
Closing this as fixed. Removed `kube-ui` in favor for `dashboard`

@w0mbat open up a new issue if it panics with this version too

closed 2016-02-14 by luxas