luxas/kubernetes-on-arm #50: kubernetesonarm/loadbalancer in in DockerHub

saturnism, 2016-01-20:
```
Pulling	Pulling image "kubernetesonarm/loadbalancer"
```

luxas, 2016-01-20:
Have you tested to build `loadbalancer` by hand? If not, try.
I haven't got it working yet, so therefore it isn't pushed.

larmog, 2016-02-05:
luxas: 

I'm using the loadbalancer. I built the binary from k8s contrib and changed the template.cfg (I found some errors in the https config from the master branch) and used your Dockerfile to build a new image.
I't works for me, haven't had any problems with serviceaccount.
/Regards

luxas, 2016-02-07:
@larmog Awesome! Could you please share the details how you changed `template.cfg`? It was probably because of that I didn't get it working. Thanks for getting in touch

larmog, 2016-02-07:
I've issued a pull request, take a look and see what you think. :)

luxas, 2016-02-14:
Fixed with `v0.6.5`

closed 2016-02-14 by luxas