luxas/kubernetes-on-arm #51: support for Odroid C1

lavvy, 2016-01-21:
Hello has odroid c1 been supported.  I thought it could be quite easier. 
It's like the hypriotos guys have added it to their supported platforms. 
Meanwhile thumbs up luxas,  and when are we seeing the release with the new dashboard and heapster.

joshuacox, 2016-01-21:
I'm 2nding the anticipation of new dashboard and heapster

luxas, 2016-01-21:
Well even though I haven't promised support for `heapster` I did cross-compile it yesterday. So that's fortunate. 

Feel free to install HypriotOS on your Odroid C1 and install the .deb package on it. @zimme is also working on Arch support: #46

luxas, 2016-01-21:
So just to clarify: I have both `heapster` and `dashboard` working on my Pi, but it will take some time before I release something, 'cause I'm preparing PRs to the main projects first

joshuacox, 2016-01-21:
is there a way to indicate which docker images are ARM?  methinks this should maybe something that should be incorporated into docker itself.  i.e.  `docker pull debian` should detect that you are on ARM and pull the appropriate images.  I know this is 'pie in the sky' kind of stuff, but it seems that maybe with the 'cavium' stuff coming mainstream soon this should be possible.

lavvy, 2016-01-21:
I was thinking it would be better to release the version for some guys to be testing and reporting before it meets k8 mainstream.

luxas, 2016-01-21:
Here's an alpha version of `dashboard`:
```
mkdir -p /etc/kubernetes/source/addons/dashboard
curl -sSL https://raw.githubusercontent.com/luxas/kubernetes-on-arm/dev/addons/dashboard/dashboard-rc.yaml > /etc/kubernetes/source/addons/dashboard/dashboard-rc.yaml
curl -sSL https://raw.githubusercontent.com/luxas/kubernetes-on-arm/dev/addons/dashboard/dashboard-svc.yaml > /etc/kubernetes/source/addons/dashboard/dashboard-svc.yaml
kube-config enable-addon dashboard
```