luxas/kubernetes-on-arm #53: source of kube-config

w0mbat, 2016-01-30:
I just switched from k8s-on-rpi to this project but I am curious where the kube-config command is coming from to install master and worker nodes?

TimCook1, 2016-01-30:
It's just a script.  You can read/modify it with any text editor.

w0mbat, 2016-01-30:
yeah but where is it? cant find it once I booted the hypriot version

w0mbat, 2016-01-30:
I just talked to @saturnism and we noticed, that I folled the README from top to bottom and the sd card setup seems to be missing the kube-config.
@saturnism suggested to flash the pure hypriot image and install via the debian package - I will try it

luxas, 2016-01-30:
`kube-config` is located in `/usr/bin` and is a normal bash script. The `kube-systemd` [README](https://github.com/luxas/kubernetes-on-arm/blob/master/sdcard/rootfs/kube-systemd/etc/kubernetes/README.md)

luxas, 2016-01-31:
Link to [kube-config](https://github.com/luxas/kubernetes-on-arm/blob/master/sdcard/rootfs/kube-systemd/usr/bin/kube-config.sh)

w0mbat, 2016-01-31:
@luxas Hmm, so from my recent experience it was not in the rpi2 / hypriot image, when I used sdcard/write.sh script. I used find after starting from sdcard but I couldn´t find it.

The workaround was to flash a plain hypriot image and installing the deb package.

luxas, 2016-02-01:
@w0mbat did you forget to include `kube-systemd` rootfs at the end? As the README says:
> sudo sdcard/write.sh /dev/sdX [board] [os] [rootfs]
> Example:
> sudo sdcard/write.sh /dev/sdX rpi-2 hypriotos kube-systemd

I made it this way, because it should be to flash the plain hypriot or arch image too from `sdcard/write`
But in the next release it could print a warning if one omits it.

w0mbat, 2016-02-01:
@luxas nope - used exactly that command and just replaced the device to match my sdcard.

luxas, 2016-02-01:
Strange. It works for me... 
Do you have more details?

luxas, 2016-03-28:
@w0mbat Can you reproduce on `v0.7.0` or should I close it?

luxas, 2016-04-10:
Closing as probably fixed in `v0.7.0`.
Reopen if not

closed 2016-04-10 by luxas