luxas/kubernetes-on-arm #54: Hostname for master

lavvy, 2016-02-07:
Hey can the master be connected to by its Hostname.  

For instance :
# To enable the worker service, run
kube-config enable-worker [master-ip or hostname]

luxas, 2016-02-07:
@lavvy I can test if it works on my cluster. Maybe... 
If you can ping your master node by hostname from a worker, it should be possible to use hostname as arg to `kube-config`

lavvy, 2016-02-07:
I don't have a working cluster yet . I believe It should work.
Can you confirm

lavvy, 2016-02-09:
Guys,  any confirmed test so far on this ?

luxas, 2016-02-09:
No, I can't ping my master from a worker. Can't even the master from itself by hostname.
Is this very important to you? I guess some more services/daemons are required for this or otherwise it's just my router...

lavvy, 2016-02-09:
Yeah,  i think this should be important cos,  it will ease the setup for
hybrid clusters

luxas, 2016-02-09:
Feel free to try to install resolving software once you have your cluster up and running. 
But IP addresses aren't hard to remember either...

luxas, 2016-03-21:
This hasn't anything to do with Kubernetes itself, it depends on your router/network setup locally.
If you can ping the node by name, you also may use the node name in `kube-config enable-worker [node-name]`, otherwise not.

closed 2016-03-21 by luxas