luxas/kubernetes-on-arm #56: Script downloads new image for every install.

GoldenBadger, 2016-02-11:
`sdcard/write.sh` downloads a new image every time you run it. If you are installing the image on a large number of SD cards in one go, `/etc/tmp` fills up quite rapidly with redundant copies of the image. It also adds a large amount of time to the installation process. Having it only download an image if one does not already exist would be a nice change.

luxas, 2016-02-11:
Hello!
Sure, I can fix it. I came across the same thing while writing some SD cards last week, just haven't had time to do it, but now I can patch that.
Thanks for notifying me :)

luxas, 2016-02-12:
Haven't tested it but should work: 4aa7cccc3ac5899c1ae65aac7b9995299ea4c241
Just remember that the `dev` branch is unstable, so test things on your own risk

luxas, 2016-02-14:
I think this is fixed with `v0.6.5`. Reopen if not

closed 2016-02-14 by luxas