luxas/kubernetes-on-arm #57: kube-config install runs docker with invalid flag.

GoldenBadger, 2016-02-11:
When running `kube-config install` it fails when starting the systemctl docker service. Running `systemctl status docker` shows the following:

```
Process: 15988 ExecStart=/usr/bin/docker -d -H unix:///var/run/docker.sock -s overlay --exec-opt native.cgroupdriver=cgroupfs (code=exited, status=125)
[...]
docker[15988]: flag provided but not defined: -d
```

Looking in `/usr/lib/systemd/system/docker.service.d/docker-overlay.conf`, it tries to run `/usr/bin/docker` with the `-d` flag, which does not exist.

luxas, 2016-02-11:
That's because `docker 1.10` removed the `-d` flag in favor for `daemon`. I haven't switched to it earlier, because right now docker 1.7.1 is used in the project (and `1.10` was released some days ago)
I'm going to go all-in for `docker 1.10` in the next release. Until that, just replace `-d` with `daemon`
Just so I know, which os/board do you run on?

GoldenBadger, 2016-02-11:
Awesome, thanks. We're running Hypriot on Pi 1 Model B's.

luxas, 2016-02-11:
Yeah, guessed that. The Hypriot guys adapts new docker releases fast :)

luxas, 2016-02-12:
I have updated the config files now (b3d8e2224f4320bda238d7c265e0bd8ff46ff538), and will update the docker daemon later on

luxas, 2016-02-14:
I also think this is fixed with `v0.6.5`. Hope it helps and thanks for notifying :smile:

closed 2016-02-14 by luxas