luxas/kubernetes-on-arm #58: kube-config enable-master cannot connect to docker daemon.

GoldenBadger, 2016-02-11:
When running `kube-config enable-master` I get:

```
[...]
Transferring images to system-docker, if necessary
Cannot connect to the Docker daemon. Is the docker daemon running on this host?
Copying kubernetesonarm/etcd to system-docker
Cannot connect to the Docker daemon. Is the docker daemon running on this host?
write /dev/stdout: broken pipe
```

Running `systemctl status docker` shows that docker is reporting that it is running correctly.

luxas, 2016-02-12:
Seems like `system-docker` isn't running. You have to update that manifest too (`-d` => `daemon` in `/usr/lib/systemd/system/system-docker.service`)

GoldenBadger, 2016-02-12:
Hi, that worked great, thanks. Now we're getting a problem where `kube-config enable-master` hangs after saying it's transferring images to system-docker. We poked around in the source for kube-config and it happens at the line `systemctl start etcd flannel`.

luxas, 2016-02-12:
I think something is wrong with `system-docker` then. Can you start containers manually on `system-docker`. I think it isn't running or something...

It was unfortunate that this would cause problems for you now, when `docker-1.10` came in and broke the project. It has been working earlier (believe it or not :smile:)

GoldenBadger, 2016-02-12:
Hmmm, we got that working again by fixing docker daemon in `/etc/kubernetes/dropins/docker-flannel.conf` but then it just hung again at the next step :stuck_out_tongue:. I don't suppose there's a way to just use an older stable version of everything?

luxas, 2016-02-12:
I don't really know if one may downgrade to `docker-1.9` on Hypriot. I think it should be possible. Something like `apt-get update && apt-get remove docker-hypriot && apt-get install docker-hypriot=1.9.1`? I don't know if the name is `docker-hypriot`, you'll have to check it up...

But overall, if you get both `system-docker` and `docker` working manually by editing files and rebooting, it may work...

luxas, 2016-02-14:
With `v0.6.5` only `docker 1.10` is supported, and it fits well for your setup (I hope you haven't downgraded everything :sweat_smile:)
Reopen if it's not working. Thanks

closed 2016-02-14 by luxas

GoldenBadger, 2016-02-14:
Awesome, we'll give that a go. Thanks very much!