luxas/kubernetes-on-arm #60: Dashboard addon timing out on pod

GoldenBadger, 2016-02-16:
Hi, not sure if this problem is specific to kubernetes-on-arm, but after running `kube-config enable-addon dashboard` on my master (192.168.1.101), I am unable to connect to it. `curl -L http://192.168.1.101:8080/api/v1/proxy/namespaces/kube-system/services/kubernetes-dashboard/` returns:
```
Error: 'dial tcp 10.1.63.2:9090: i/o timeout'
Trying to reach: 'http://10.1.63.2:9090/'
```
When I try to curl the pod address (10.1.63.2:9090) directly it just waits indefinitely for a connection. Would you be able to provide any insight on this? Thanks.

luxas, 2016-02-16:
Is it maybe this problem that's causing it? Sorry for the typo, but easy to fix: #59

GoldenBadger, 2016-02-24:
Both this and #61 were caused by a dodgy install. I've explained this in more detail on that thread.

closed 2016-02-24 by GoldenBadger