luxas/kubernetes-on-arm #61: Connection reset when connecting to a pod or service.

GoldenBadger, 2016-02-16:
When running the nginx test described in the root README.md, I cannot connect to any endpoints on either the pods themselves or a service IP. They all simply return `curl: (56) Recv failure: Connection reset by peer`. This is on HypriotOS on the latest version of the master branch. I'm sorry I can't be more descriptive, but I have no idea what is going wrong.

luxas, 2016-02-16:
Are the containers running? What does `docker ps`, `kubectl get po,svc`, `kube-config info` output?
Are you running on one node (master only) or do you have workers too?

GoldenBadger, 2016-02-16:
No, I noticed that and changed it myself. This seems to be a separate problem. I have 1 master and 4 workers currently running.

GoldenBadger, 2016-02-24:
Sorry about the delay responding. It turns out that both this and #60 were caused by a dodgy install on one of our Pis. We're still not sure what caused it but we switched the master to another Pi and reimaged the problem one as a worker, and then both issues were resolved.

closed 2016-02-24 by GoldenBadger

luxas, 2016-02-25:
Glad you got it fully working!