luxas/kubernetes-on-arm #62: Using host-gw as backend for flannel instead of udp.

larmog, 2016-02-25:
According to this blog article: http://machinezone.github.io/research/networking-solutions-for-kubernetes/ maybe a better configuration for flannel would be to use `host-gw`. I think it's safe to assume that most users will setup theirs k8s nodes on the same layer 2 network. I've changed my flannel config and verified that flannel is using `host-gw`.

```
$ etcdctl set /coreos.com/network/config   '{ "Network": "10.1.0.0/16", "Backend": { "Type": "host-gw" } }'
```
and logs for flannel:
```
I0225 08:06:01.217331 00001 main.go:275] Installing signal handlers
I0225 08:06:01.219378 00001 main.go:130] Determining IP address of default interface
I0225 08:06:01.225660 00001 main.go:188] Using 192.168.1.28 as external interface
I0225 08:06:01.226183 00001 main.go:189] Using 192.168.1.28 as external endpoint
I0225 08:06:01.252644 00001 etcd.go:129] Found lease (10.1.70.0/24) for current IP (192.168.1.28), reusing
I0225 08:06:01.262135 00001 etcd.go:84] Subnet lease acquired: 10.1.70.0/24
I0225 08:06:01.264613 00001 hostgw.go:100] Watching for new subnet leases
I0225 08:06:01.279136 00001 hostgw.go:140] Subnet added: 10.1.64.0/24 via 192.168.1.23
I0225 08:06:01.280315 00001 hostgw.go:140] Subnet added: 10.1.82.0/24 via 192.168.1.26
I0225 08:06:01.281465 00001 hostgw.go:140] Subnet added: 10.1.32.0/24 via 192.168.1.27
I0225 08:06:01.283433 00001 hostgw.go:140] Subnet added: 10.1.49.0/24 via 192.168.1.24
I0225 08:06:01.292481 00001 hostgw.go:140] Subnet added: 10.1.54.0/24 via 192.168.1.25
```

luxas, 2016-02-25:
I added this to `dev`

luxas, 2016-03-21:
Thanks @larmog, fixed in `v0.7.0`

closed 2016-03-21 by luxas