luxas/kubernetes-on-arm #63: DNS problems

kaspernissen, 2016-02-25:
After a clean install and running for a day with no problems, the DNS seems to not be working anymore. I'm running on HypriotOS. `kube-config delete-data` does not resolve the problem. Tried the following with a simple busybox image:

`$ curl http://busybox.default:90  
curl: (6) Could not resolve host: busybox.default  
HypriotOS: root@node21 in ~
$ curl http://busybox.default.svc.cluster.local:90
curl: (6) Could not resolve host: busybox.default.svc.cluster.local`

any ideas?

`$ kubectl cluster-info
Kubernetes master is running at http://localhost:8080
KubeDNS is running at http://localhost:8080/api/v1/proxy/namespaces/kube-system/services/kube-dns
KubeRegistry is running at http://localhost:8080/api/v1/proxy/namespaces/kube-system/services/registry`

`$ run-test master
Architecture: armv7l
Kernel: Linux 4.1.12
CPU: 4 cores x 900 MHz

Used RAM Memory: 743 MiB
RAM Memory: 925 MiB

Used disk space: 3.4GB (3489268 KB)
Free disk space: 9.8GB (10190984 KB)

SD Card/deb package was built: 14-02-2016 10:21

kubernetes-on-arm:
Latest commit: febacd6
Version: 0.6.5

systemd version: v215
docker version: v1.10.1
kubernetes server version: v1.1.3

CPU Time (minutes):
kubelet: 4:11
kubelet has been up for: 11 minutes
apiserver: 12:47
controller-manager: 0:33
scheduler: 0:05
proxy: 0:10
Disabling k8s if it is running
Removed symlink /etc/systemd/system/multi-user.target.wants/k8s-master.service.
Removed symlink /etc/systemd/system/multi-user.target.wants/flannel.service.
Removed symlink /etc/systemd/system/multi-user.target.wants/etcd.service.
Checks so all images are present
Transferring images to system-docker, if necessary
Created symlink from /etc/systemd/system/multi-user.target.wants/etcd.service to /usr/lib/systemd/system/etcd.service.
Created symlink from /etc/systemd/system/multi-user.target.wants/flannel.service to /usr/lib/systemd/system/flannel.service.
Starting master components in docker containers
Created symlink from /etc/systemd/system/multi-user.target.wants/k8s-master.service to /usr/lib/systemd/system/k8s-master.service.
Kubernetes master services enabled

real	0m57.154s
user	0m3.810s
sys	0m0.790s
curl: (7) Failed to connect to 10.0.0.1 port 443: Connection refused
curl: (7) Failed to connect to 10.0.0.1 port 443: Connection refused
curl: (7) Failed to connect to 10.0.0.1 port 443: Connection refused
curl: (7) Failed to connect to 10.0.0.1 port 443: Connection refused
curl: (7) Failed to connect to 10.0.0.1 port 443: Connection refused
curl: (7) Failed to connect to 10.0.0.1 port 443: Connection refused
curl: (7) Failed to connect to 10.0.0.1 port 443: Connection refused
curl: (7) Failed to connect to 10.0.0.1 port 443: Connection refused
curl: (7) Failed to connect to 10.0.0.1 port 443: Connection refused
curl: (7) Failed to connect to 10.0.0.1 port 443: Connection refused
curl: (7) Failed to connect to 10.0.0.1 port 443: Connection refused
curl: (7) Failed to connect to 10.0.0.1 port 443: Connection refused
[... 351 lines cut ...]
ebcb43837302        kubernetesonarm/pause         "/pause"                 4 minutes ago        Up 4 minutes                                k8s_POD.b10bdbaa_my-nginx-3d3vt_default_64c8dc88-dbc6-11e5-b62b-b827ebc5b105_37717e44
cd575ccf6892        kubernetesonarm/pause         "/pause"                 4 minutes ago        Up 4 minutes                                k8s_POD.b10bdbaa_my-nginx-gkahh_default_64c753ca-dbc6-11e5-b62b-b827ebc5b105_21f52f9f
0858fb615aca        kubernetesonarm/pause         "/pause"                 4 minutes ago        Up 4 minutes                                k8s_POD.b10bdbaa_my-nginx-ki2xh_default_64c6cd20-dbc6-11e5-b62b-b827ebc5b105_e653d983
0f67de150066        kubernetesonarm/hyperkube     "/hyperkube controlle"   4 minutes ago        Up 4 minutes                                k8s_controller-manager.857e6441_k8s-master-192.168.1.21_default_e17265de010338ec79ec0b59ed6bd2e0_6cb9918e
da06775ce2fa        kubernetesonarm/hyperkube     "/hyperkube scheduler"   5 minutes ago        Up 5 minutes                                k8s_scheduler.9849f626_k8s-master-192.168.1.21_default_e17265de010338ec79ec0b59ed6bd2e0_66b079b6
fc9ea83c9dd3        kubernetesonarm/hyperkube     "/hyperkube apiserver"   5 minutes ago        Up 5 minutes                                k8s_apiserver.1631abf2_k8s-master-192.168.1.21_default_e17265de010338ec79ec0b59ed6bd2e0_2065639d
749db8a8ef0c        kubernetesonarm/hyperkube     "/hyperkube proxy --m"   5 minutes ago        Up 5 minutes                                k8s_kube-proxy.31eb3d50_k8s-master-192.168.1.21_default_e17265de010338ec79ec0b59ed6bd2e0_bee3099a
7fc06243f036        kubernetesonarm/pause         "/pause"                 5 minutes ago        Up 5 minutes                                k8s_POD.b10bdbaa_k8s-master-192.168.1.21_default_e17265de010338ec79ec0b59ed6bd2e0_782fc536
90fa061b0f0a        kubernetesonarm/hyperkube     "/hyperkube kubelet -"   6 minutes ago        Up 6 minutes                                k8s-master
Summary:

Seconds before apiserver came up: 83
Seconds before nginx came up: 54
Seconds before dns came up: 10
Seconds before registry came up: 5

Services in Kubernetes are working
DNS on host isn't working
DNS on host with shorthand search commands isn't working
The apiserver proxy is working
The registry is up and running

real	7m24.998s
user	0m49.830s
sys	0m12.930s`

luxas, 2016-02-25:
What does 
```
$ cat /etc/kubernetes/k8s.conf
$ docker logs 08f3cb81c281
$ docker exec -it a632a342aeb4 /bin/bash
$>> nslookup kubernetes.default
$>> nslookup kubernetes.default.svc.kubernetesonarm.com
```
output?
Note that the `master` test sets `DNS_DOMAIN` to `cluster.local` and `DNS_IP` to `10.0.0.110`. That's the purpose of the test.

Does it get any better? The test isn't 100% reliable
> The test might fail, although the thing it's testing is in fact working

I would recommend the "normal" way `kube-config enable-master`.
(P.S. I couldn't resist adding proper code formatting to your issue above. Now it became a lot easier to read and review)

kaspernissen, 2016-02-26:
Thank you for the quick answer and sorry for the lack of proper formatting and thanks for cleaning it up.

Output:

```bash
$ cat /etc/kubernetes/k8s.conf
K8S_MASTER_IP=127.0.0.1
FLANNEL_SUBNET=10.1.0.0/16
DNS_DOMAIN=kubernetesonarm.com
DNS_IP=10.0.0.110
```
I've tried poking around and found the following:
```bash
$ kubectl logs kube-dns-v8-wy03h --namespace=kube-system -c kube2sky
I0226 07:53:06.757471       1 kube2sky.go:433] Etcd server found: http://127.0.0.1:4001
I0226 07:53:07.771385       1 kube2sky.go:500] Using https://10.0.0.1:443 for kubernetes master
I0226 07:53:07.776137       1 kube2sky.go:501] Using kubernetes API
```
```bash
$ kubectl logs kube-dns-v8-wy03h --namespace=kube-system -c skydns
2016/02/26 07:53:06 skydns: falling back to default configuration, could not read from etcd: 100: Key not found (/skydns/config) [449]
2016/02/26 07:53:06 skydns: ready for queries on kubernetesonarm.com. for tcp://0.0.0.0:53 [rcache 0]
216/02/26 07:53:06 skydns: ready for queries on kubernetesonarm.com. for udp://0.0.0.0:53 [rcache 0]
```

It seems that skydns cannot find the key in etcd. Any ideas?

luxas, 2016-02-29:
It seems quite normal to me, I think the `skydns` thing is just a fallback, but anyway you may kill the `skydns` container for testing (it will come up again, I promise :smile:) and see if it gets better.

However, as I said. Please use the normal `cluster.local` and `10.0.0.10` and try again. The master test isn't stable and is meant just for testing purposes.
Also check
```
$ cat /etc/resolv.conf
$ docker exec -it <alpine_contianer> /bin/sh
$>> nslookup kubernetes.default
$>> nslookup kubernetes.default.svc.kubernetesonarm.com
```

sokoow, 2016-02-29:
I'd also suggest to browse through etcd directories and see whether skydns entries are there. I had just had a case when kube2sky wasn't writing to etcd properly.

kaspernissen, 2016-02-29:
Should `10.0.0.10` also be registed in `/etc/resolv.conf`? 
At the moment we only have Google DNS and our local router IP. 

How to find the alpine container? - I cannot find any containers when checking `docker ps` or looking through pods in Kubernetes. I'm not that familiar with what happens under the hood in Kubernetes and Docker. Can you point me in the right direction ? 

@sokoow thank you for the suggestion. I'll try to poke around etcd.

kaspernissen, 2016-03-01:
@sokoow What was the problem that you had exactly? and how did you fix it? 

It sounds very similar to what we are experiencing. We cannot find any skydns entries i etcd and theres no `:4001/v2/keys/skydns`and returns key not found. 

When looking at `:4001/v2/keys/registry/services/endpoints/default` with no services running, no sight of Skydns.

```json
{
"action": "get",
"node": {
"key": "/registry/services/endpoints/default",
"dir": true,
"nodes": [
{
"key": "/registry/services/endpoints/default/kubernetes",
"value": "{\"kind\":\"Endpoints\",\"apiVersion\":\"v1\",\"metadata\":{\"name\":\"kubernetes\",\"namespace\":\"default\",\"uid\":\"4ed2bf3d-dbc6-11e5-b62b-b827ebc5b105\",\"creationTimestamp\":\"2016-02-25T13:47:27Z\"},\"subsets\":[{\"addresses\":[{\"ip\":\"192.168.1.21\"}],\"ports\":[{\"port\":6443,\"protocol\":\"TCP\"}]}]}\n",
"modifiedIndex": 10,
"createdIndex": 10
}
],
"modifiedIndex": 10,
"createdIndex": 10
}
}
```

sokoow, 2016-03-01:
Well, I checked for /skydns dir on etcd and it was there, containing all needed data, but somehow skydns didn't pick it up. I poked around for half a day, then got irritated and ... pulled kube2sky out of a container, pointed it on to system etcd, then changed -machines in skydns to point out to other etcd also. Now it works all the time. I don't know why my install can't just work with etcd that starts during dns package setup.

I'd suggest running skydns with -verbose and then checking logs of skydns and structure of etcd directories. If all is good you should see reverse dns lookups in skydns verbose mode, and kube2sky shouldn't complain about etcd. Debugging this is a bit hard, but as you see it's doable.

luxas, 2016-03-01:
Please note though that this is a upstream k8s issue, it isn't directly related to the ARM binaries.
Upstream DNS is also a bit broken, but it should be better since my patch: kubernetes/kubernetes#21504.
One thing you may test, is uncommenting:
```
#        livenessProbe:
#          httpGet:
#            path: /healthz
#            port: 8080
#            scheme: HTTP
#          initialDelaySeconds: 30
#          timeoutSeconds: 5
```
and see if it gets better. But again, please do not use `run-test master` for a setup you won't erase directly after (rewrite the SD Card).

kaspernissen, 2016-03-09:
I just tried a clean install of a master and a slave, and unfortunately i'm still running into the same problems with DNS after following the instructions in the README.

```
$ cat /etc/kubernetes/k8s.conf
K8S_MASTER_IP=127.0.0.1
FLANNEL_SUBNET=10.1.0.0/16
DNS_DOMAIN=cluster.local
DNS_IP=10.0.0.10
```

When trying to curl the my-nginx i get the following
```
$ curl my-nginx.default.svc.cluster.local
curl: (6) Could not resolve host: my-nginx.default.svc.cluster.local
```

Nothing is written to `/etc/resolv.conf` and running `kubectl cluster-info` states that KubeDNS is running af `http://localhost:8080/api/v1/proxy/namespaces/kube-system/services/kube-dns`, however when accessing the URL i get the following:

```
$ curl http://localhost:8080/api/v1/proxy/namespaces/kube-system/services/kube-dns
{
  "kind": "Status",
  "apiVersion": "v1",
  "metadata": {},
  "status": "Failure",
  "message": "no endpoints available for service \"kube-dns\"",
  "reason": "ServiceUnavailable",
  "code": 503
}
```

Anybody else experiencing similar issues?

luxas, 2016-03-15:
Sounds like the dns containers aren't running properly.

> Nothing is written to /etc/resolv.conf

On host or in containers?

> curl http://localhost:8080/api/v1/proxy/namespaces/kube-system/services/kube-dns

This is expected, though a bit counter-intuitive: https://github.com/kubernetes/kubernetes/issues/20070

Is it flannel that isn't working again? One may try to examine the endpoints.

luxas, 2016-03-21:
https://github.com/luxas/kubernetes-on-arm/issues/77#issuecomment-199498499

luxas, 2016-03-28:
@kaspernissen Does this happen with `v0.7.0`?

kaspernissen, 2016-03-31:
We are experimenting with Arch Linux at the moment - Unfortunately haven't had the time to verify that v.0.7.0 on HypriotOS has the same symptoms.  Will report back when we get the time to test it out.

luxas, 2016-05-13:
Closing as probably fixed in `v0.7.0`

closed 2016-05-13 by luxas