luxas/kubernetes-on-arm #64: Can't access the dashboard / UI

e-nouri, 2016-03-01:
First of all, than you for this helper, what Debian did with Jessie is horible lol :D, you project helped me set up kubernetes real quick on my pies.

After the kube-ui have being removed for the favor of the new dashboard, I added the addon dashboard.

I fixed this issue before adding:

The metadata.labels name was invalid:
metadata.labels: invalid value 'Kubernetes Dashboard', Details: must have at most 63 characters, matching regex (([A-Za-z0-9][-A-Za-z0-9_.]*)?[A-Za-z0-9])?: e.g. "MyValue" o
/Regards

Made sure that i had a service and the dashboard running, I cannot access the dashboard from the master ip. Are you able to run the dashboard ? do you have any tips ?

luxas, 2016-03-01:
I don't really get what is your problem, but you've fixed the whitespace in `/etc/kubernetes/addons/dashboard.yaml`, then you should `kube-config disable-addon dashboard` and `kube-config enable-addon dashboard` and it should be up again.
Assuming `v0.6.5`

e-nouri, 2016-03-01:
how do you access the dashboard after adding it ?

luxas, 2016-03-01:
See https://github.com/luxas/kubernetes-on-arm#access-your-cluster.
The easiest way is `http://[master-ip]:8080/api/v1/proxy/namespaces/kube-system/services/kubernetes-dashboard`
You'll also get these urls from `kubectl cluster-info`

In `v1.2` (which I'm working on) the `http://[master-ip]:8080/ui` will redirect to `http://[master-ip]:8080/api/v1/proxy/namespaces/kube-system/services/kubernetes-dashboard`. So it's gonna be easier

e-nouri, 2016-03-01:
Yes I tried that, I got this routing error:

Error: 'dial tcp 10.1.2.2:9090: i/o timeout'
Trying to reach: 'http://10.1.2.2:9090/'

luxas, 2016-03-01:
Maybe the same as @GoldenBadger? https://github.com/luxas/kubernetes-on-arm/issues/60#issuecomment-188431458 then? It's hard to say for me, cause I can't reproduce

luxas, 2016-03-28:
Can you reproduce with `v0.7.0`?

luxas, 2016-04-23:
Closing as probably fixed w/ `v0.7.0`.
Reopen if not

closed 2016-04-23 by luxas