luxas/kubernetes-on-arm #65: kubectl: command not found

nooproblem, 2016-03-02:
Hi,

Firstly, thanks to luxas for k8s and rancheros on ARM.

Running: a single RPI-2/HypriotOS/.deb package deployment

When _kube-config_ _enable-addon dns_ is run as root the following messages are written to stdout:
```
/usr/bin/kube-config: line 473: kubectl: command not found
/usr/bin/kube-config: line 477: kubectl: command not found
```
However, kubectl will run fine as a non-root user, but kube-config needs to be run as root (it seems).

I just wanted someones thoughts as the best way to get around this catch 22 situation?

luxas, 2016-03-02:
Yeah, thanks! I have put much work into it, so it's fun to get positive feedback :smile:

To the issue, sometimes `sudo` modifies `$PATH` in a stange way. 
The workaround is documented [here](https://github.com/luxas/kubernetes-on-arm#setup-your-board-from-an-sd-card) (I think)
> sudo env PATH=$PATH ...

I hope that's the problem, 'cause it's easy to solve. Or you may `ssh` into the `root` user directly (assuming you are a non-root user right now)

nooproblem, 2016-03-02:
Awesome, that workaround was successful. Keep up the amazing work @luxas and thanks once again!

closed 2016-03-02 by nooproblem

ldco2016, 2016-11-29:
Greetings, 

I am have successfully installed docker and kubernetes with control-master and three minions and have successfully created a pod, but when I go to delete the pod, even if I run the  whole path such as /usr/bin/kubectl delete pod busybox, I get back sh: /usr/bin/kubectl: not found.

Can anybody point me towards some troubleshooting tips?