luxas/kubernetes-on-arm #77: Services stops working (probably a flannes issue)

mrjensen, 2016-03-14:
I have a situation where everything works fine at first, but after some time, rolling-updates, and a reboot the services stop working. It looks like the proxy is failing.

I am running a service on port 9000 but the proxy logs the following when I make a request from a browser.
```
E0314 09:40:50.725351       1 proxysocket.go:138] Failed to connect to balancer: failed to connect to an endpoint.
E0314 09:46:03.462857       1 proxysocket.go:104] Dial failed: dial tcp 10.1.86.2:9000: i/o timeout
E0314 09:46:03.964366       1 proxysocket.go:104] Dial failed: dial tcp 10.1.86.2:9000: i/o timeout
E0314 09:46:04.965488       1 proxysocket.go:104] Dial failed: dial tcp 10.1.86.2:9000: i/o timeout
E0314 09:46:06.966921       1 proxysocket.go:104] Dial failed: dial tcp 10.1.86.2:9000: i/o timeout
E0314 09:46:06.967550       1 proxysocket.go:138] Failed to connect to balancer: failed to connect to an endpoint.
E0314 09:46:07.225211       1 proxysocket.go:104] Dial failed: dial tcp 10.1.86.2:9000: i/o timeout
E0314 09:46:07.727560       1 proxysocket.go:104] Dial failed: dial tcp 10.1.86.2:9000: i/o timeout
E0314 09:46:08.729850       1 proxysocket.go:104] Dial failed: dial tcp 10.1.86.2:9000: i/o timeout
E0314 09:46:10.731650       1 proxysocket.go:104] Dial failed: dial tcp 10.1.86.2:9000: i/o timeout
E0314 09:46:10.731911       1 proxysocket.go:138] Failed to connect to balancer: failed to connect to an endpoint.
```

Furthermore when I look at the flannel log it seems like the wrong subnet is removed?

```
$ journalctl -xeu flannel
~
~
-- Logs begin at Fri 2016-03-11 15:45:02 EET, end at Mon 2016-03-14 11:55:06 EET. --
Mar 11 15:45:04 node11 systemd[1]: Starting Flannel Overlay Network for Kubernetes...
-- Subject: Unit flannel.service has begun with start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- Unit flannel.service has begun starting up.
Mar 11 15:45:07 node11 docker[282]: Failed to kill container (k8s-flannel): Error response from daemon: Cannot kill container k8s-flannel: Container 88b80fb9d5025420c12ac04fe376b6648a9fe19e89e0333701c124c7bebec93c is not running
Mar 11 15:45:08 node11 docker[353]: k8s-flannel
Mar 11 15:45:08 node11 systemd[1]: Started Flannel Overlay Network for Kubernetes.
-- Subject: Unit flannel.service has finished start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- Unit flannel.service has finished starting up.
--
-- The start-up result is done.
Mar 11 15:45:12 node11 docker[369]: I0311 13:45:12.326375 00001 main.go:275] Installing signal handlers
Mar 11 15:45:12 node11 docker[369]: I0311 13:45:12.329131 00001 main.go:130] Determining IP address of default interface
Mar 11 15:45:12 node11 docker[369]: I0311 13:45:12.332833 00001 main.go:188] Using 192.168.1.11 as external interface
Mar 11 15:45:12 node11 docker[369]: I0311 13:45:12.333117 00001 main.go:189] Using 192.168.1.11 as external endpoint
Mar 11 15:45:12 node11 docker[369]: E0311 13:45:12.338227 00001 network.go:53] Failed to retrieve network config: client: etcd cluster is unavailable or misconfigured
Mar 11 15:45:17 node11 docker[369]: I0311 13:45:17.069655 00001 etcd.go:129] Found lease (10.1.17.0/24) for current IP (192.168.1.11), reusing
Mar 11 15:45:18 node11 docker[369]: I0311 13:45:18.046167 00001 etcd.go:84] Subnet lease acquired: 10.1.17.0/24
Mar 11 15:45:18 node11 docker[369]: I0311 13:45:18.120775 00001 udp.go:222] Watching for new subnet leases
Mar 11 15:45:18 node11 docker[369]: I0311 13:45:18.136222 00001 udp.go:247] Subnet added: 10.1.87.0/24
Mar 11 15:45:18 node11 docker[369]: I0311 13:45:18.136763 00001 udp.go:247] Subnet added: 10.1.50.0/24
Mar 11 15:45:18 node11 docker[369]: I0311 13:45:18.136982 00001 udp.go:247] Subnet added: 10.1.86.0/24
Mar 14 09:23:50 node11 docker[369]: I0314 07:23:50.158176 00001 udp.go:252] Subnet removed: 10.1.86.0/24
Mar 14 09:23:50 node11 docker[369]: I0314 07:23:50.216053 00001 udp.go:252] Subnet removed: 10.1.87.0/24
Mar 14 09:23:50 node11 docker[369]: I0314 07:23:50.254460 00001 udp.go:252] Subnet removed: 10.1.50.0/24
```

A quick fix has been to run the commands below on the master node and restart the worker nodes. But it doesn't seem like a good solution to delete everything.
```
kube-config disable-node
kube-config delete-data
kube-config enable-master
```

It's running on a Raspberry Pi cluster running HypriotOS with kubernetes-on-arm installed from the .deb package.

luxas, 2016-03-15:
> after some time, rolling-updates, and a reboot the services stop working

Have you rebooted the master while the workers are running? Just so I get you right.

From the logs, obviously `10.1.86.2` isn't reachable if `10.1.86.0/24` is removed from `flannel`. The question is why it does that. Did you do something special at 14th March 7:23? As we may see, `flannel` removed the subnets. All flannel data is stored in `etcd` so you may look there if something is suspicious. If you did reboot the master node, then flannel noticed that and removed the connections because of no etcd. And it seems like it didn't add the subnets back again.
What do you think?

mrjensen, 2016-03-15:
In one of the cases all the nodes had been turned of and booted again, in the other case the nodes ran without any reboots (freshly installed HypriotOS with the .deb) and it still happened.

Yes, it is exactly the subnet removal that seems suspicious. `etcd` looks alrightfrom on port 4001. The timeout and missing flannel subnet happened for three of four services, which is even more odd.

I am currently trying with a clean installation archlinux from this repository instead of the .deb to see if it also happens in that configuration.

luxas, 2016-03-21:
Maybe you could do a pilot test with two nodes running `v0.7.0` + `Arch Linux` and give some feedback if it works better.

luxas, 2016-03-28:
@mrjensen Does this happen with v0.7.0? Are you able to reproduce or should we close this?

mrjensen, 2016-03-28:
@luxas I think we should close it.
v0.6.5 on Arch Linux worked like a charm, and I am currently installing v0.7.0 on Arch Linux.
The only place I had issues was running "HypriotOS Version 0.6.1 Hector" (not installed from your script) with the .deb package of v0.6.5.

closed 2016-03-28 by mrjensen