luxas/kubernetes-on-arm #81: kube-proxy + iptables

DorianGray, 2016-03-20:
What exactly is the problem with --proxy-mode=iptables right now?

luxas, 2016-03-20:
In `v1.1*` (`v0.6.x` of Kubernetes on ARM), it didn't work. I don't know why.
I actually tested it yesterday, and now with `v1.2`, it works! So I have enabled it in `v0.7.0`. 
Please test it if you can

DorianGray, 2016-03-20:
I was wondering because of a change I made for getting glusterfs working. I have it working with 1.1 on my local. It works if you move ip-masq config to flannel and explicitly disable it in docker.

It needs to be configured that way for 1.2 as well, so I'll put it in a PR. Applications like gluster discover their IP as the host's IP instead of the container IP if docker is configured to do the masqing.

DorianGray, 2016-03-20:
https://github.com/luxas/kubernetes-on-arm/pull/82

luxas, 2016-04-10:
Thanks, I think it's fixed with #82

closed 2016-04-10 by luxas