luxas/kubernetes-on-arm #83: What to do about hyperkube dependencies?

DorianGray, 2016-03-21:
For instance, for kubernetes glusterfs support the glusterfs binaries need to be installed inside the hyperkube container(so mount works)...but the fuse kernel module and a handful of other things also need to be available in the host OS.

luxas, 2016-03-21:
Haven't tested `glusterfs` yet, but I made an image for it: https://github.com/luxas/kubernetes-on-arm/blob/v0.7.0/images/kubernetesonarm/glusterfs/Dockerfile

I will probably look into this for the next release. PRs are welcome

DorianGray, 2016-03-25:
We're going to need to figure out a way to deal with this...maybe just require the fuse package on all os images?...then include some scripts in the hyperkube image that know how to install missing dependencies?

luxas, 2016-03-25:
I'm a bit concerned about including too much extra things into the `kubernetesonarm/hyperkube` image as `v0.8` (= k8s `v1.3`) probably will use the official `hyperkube` image.

I think this is an upstream issue (could you open it there too?), but I'm glad to discuss it here too.
WYDT?