luxas/kubernetes-on-arm #84: Kubelet in v0.7.0 can't mount glusterfs

larmog, 2016-03-24:
The new `native` kubelet mode can't mount `glusterfs` volumes. This is probably due to that kubelet tries to exec mount in the container rather on the host:
```
E0323 15:02:28.120954   10623 glusterfs.go:246] Glusterfs: mount failed: exit status 32
.
.
.
Output: mount: unknown filesystem type 'glusterfs'
```
When running in "real" native mode (outside docker) mount works.

DorianGray, 2016-03-25:
https://github.com/luxas/kubernetes-on-arm/issues/83
https://github.com/luxas/kubernetes-on-arm/issues/40

I commented with information on how to work around this issue in the above places.

The fuse kernel module needs to be probed on the host, and the glusterfs fuse driver needs to be installed inside the kubelet container.

DorianGray, 2016-03-25:
As an aside, it is impossible to probe kernel modules in containers...so this is a weird issue to solve

larmog, 2016-03-25:
Seeing before believing. ;)

@DorianGray I read your comments in #40 but somehow missed the implications, and totally missed #83. Thanks for you pointing that out.

This is a duplicate of #83, I'm closing it.

closed 2016-03-25 by larmog