luxas/kubernetes-on-arm #85: docker-on-arm.md missing, broken link in readme.md

v1k0d3n, 2016-03-29:
hey @luxas, it looks like the https://github.com/luxas/kubernetes-on-arm/blob/master/docs/docker-on-arm.md is missing. not that there isn't other information out there, but i just wanted to let you know in case there was any specific information that could help out new users using your repo. keep up the great work, man!

luxas, 2016-03-29:
Hi @v1k0d3n! I'm glad you're still following my project so you're able to create issues :)'
I deleted the file because if I would keep it, it would be misleading because so much things have happened since I wrote it.
I hadn't time to rewrite the doc when I made the release, but maybe I'll write up something for the next release :)

madmas, 2016-04-27:
Hi,
I started a new document :-) hope you find this useful

luxas, 2016-05-13:
Fixed at `dev` with #97

closed 2016-05-13 by luxas