luxas/kubernetes-on-arm #88: Dashboard/ui defaults to kube-system namespace

stevemcquaid, 2016-04-01:
This could be an error on my end, but it seems like the dashboard is unable to switch between namespaces. New services run in the default namespace, and so there is no way to monitor them via the ui since it is on the kube-system namespace.

larmog, 2016-04-08:
@stevemcquaid is your pods listed when running
`kubectl --all-namespaces=true get pods -o wide`?

stevemcquaid, 2016-04-08:
Yes. That command does list all pods. The dashboard however only shows pods
in the same namespace: kube-system

larmog, 2016-04-08:
Strange, as you can see it works for me.
[image]

Can you take a look in the log?
/Regards

stevemcquaid, 2016-04-08:
when you view the dashboard pod itself - what namespace is it run as on
yours?

Regards,
Steve McQuaid
484.727.8383

larmog, 2016-04-08:
The `kubernetes-dashboard` runs in `kube-system` namespace.

stevemcquaid, 2016-04-08:
hmm same thing on my system. I'm curious how you are able to see default
namespace from dashboard.. LIke you said I guess I will have to take a look
at my logs. Thanks for looking at it! I really appreciate it. Will follow
up if I discover the cause

luxas, 2016-04-10:
@stevemcquaid Sorry for not having time to answer, and thanks @larmog for helping.

The dashboard displays _all replication controllers_ in _all namespaces_ at version `v1.0`.
This is by design. It does not display replicasets (rs), daemonsets, jobs or some other resources.
It just queries `kubectl get rc --all-namespaces`. If only pods in the `kube-system` namespace is found, it displays the `zerostate` (the welcome page), and there is a link to show rcs in `kube-system` (itself and some other addons).

I think you're only running replicasets or some other thing, Note that `kubectl run` creates deployments, which creates replicasets, which creates pods in `v1.2`, in `v1.1` it did create rcs.

Hope it helps

luxas, 2016-05-13:
@stevemcquaid Was the problem that you created `deployments` instead of `replicationcontrollers`?
If so, I think this issue may be closed.

stevemcquaid, 2016-05-13:
Hey sorry for the delay here. I have been rolling out k8s for my company and not had much time to fiddle on rpi. Will revisit-retest this when I get a chance. But for now we can close it :)

closed 2016-05-13 by luxas