luxas/kubernetes-on-arm #89: Service type NodePort works only on the Node where the pod are deployed

jbclaramonte, 2016-04-09:
Hello

First of all, thanks for your work and for sharing it !
I've created a 5 Nodes rpi cluster and this is really great to learn how to usz kubernetes !

Now the issue ;-)

I'm currently facing an issue on my cluster, here are the simple commands I run:
$ kubectl run my-nginx --image=luxas/nginx-test

$ kubectl expose deployment my-nginx --port=80 --container-port=80 --type=NodePort

$ kubectl describe svc my-nginx
Name:			my-nginx-4186726867
Namespace:		default
Labels:			pod-template-hash=4186726867,run=my-nginx
Selector:		pod-template-hash=4186726867,run=my-nginx
Type:			NodePort
IP:			10.0.0.171
Port:			<unset>	80/TCP
NodePort:		<unset>	31031/TCP
Endpoints:		10.1.51.2:80
Session Affinity:	None

$ kubectl get nodes
NAME          STATUS    AGE
192.168.0.2   Ready     3d
192.168.0.3   Ready     36m
192.168.0.5   Ready     3d
192.168.0.6   Ready     2d
192.168.0.8   Ready     4h

$ kubectl get pods -o wide
NAME                        READY     STATUS    RESTARTS   AGE       NODE
k8s-master-192.168.0.2      4/4       Running   2          3d        192.168.0.2
my-nginx-4186726867-nwogd   1/1       Running   0          34m       192.168.0.6

$ curl http://192.168.0.6:31031
<p>WELCOME TO NGINX</p>

It works, cool, but if I do:
$ curl http://192.168.0.5:31031
... no response

So if I ssh to 192.168.0.5
and run:
$ netstat -nlp | grep 31031
tcp6       0      0 :::31031                :::*                    LISTEN      662/hyperkube

How could I figure out what is missing ?

Thanks for your help

luxas, 2016-04-10:
Hello @jbclaramonte and thanks for using this project!

Hmm, can you `curl 10.0.0.171` or `curl 10.1.51.2:80` from node `192.168.0.5`?
That way you'll see if the network works as expected.

jbclaramonte, 2016-04-10:
nothing is happening when I curl those IP from any other Node than 192.168.0.6 (where the pod is)
What do you think I should try to fix this ?

luxas, 2016-04-10:
Something is wrong with the proxying then.

Verify if `kube-proxy` is running fine, it's probably there the issue is.
You can try to restart `kube-proxy` on every node first by killing the `kube-proxy` container and starting like specified here: https://github.com/luxas/kubernetes-on-arm/blob/master/sdcard/rootfs/kube-systemd/usr/lib/systemd/system/k8s-worker.service#L34

If that doesn't work, you can disable the cluster node-by-node and apply these changes: https://github.com/luxas/kubernetes-on-arm/pull/82/files

This will be better with the next release when `kube-proxy` is a `DaemonSet`

jbclaramonte, 2016-04-10:
the second option worked (applying the change and then disabling cluster node by node) !
I'm so happy :-)
Thanks !

luxas, 2016-04-10:
I'm glad you got it fixed! 
Now you may continue with some tinkering :)

closed 2016-04-10 by luxas

chrissound, 2018-07-04:
I seem to have run into the same issue. What is meant by disabling a node? How would I do this?