luxas/kubernetes-on-arm #96: master does not have anything running on 8080

richtong, 2016-04-25:
OK, Luxas what's the best way to debug this? I have gone back to the original RPI1. It is running hypriot 201603. I have done a kube-config install and then a `kube-config enable-master` and I see one container called kubernetesonarm/hyperkube start up. Yeah!

tjskrish, 2016-05-09:
I am getting the same error!

oiavorskyi, 2016-05-10:
Same problem. Using archlinux on RPi2. There are no visible signs of errors.

luxas, 2016-05-13:
Hi all,
Haven't come across this earlier.

Seems like some cgroup mount is wrong and that leads to that kubelet isn't running correctly.
Please output `docker info` and `mount` as a first step

johan-wallermark, 2016-05-15:
I ran into the same problem. I'm not really sure where to being troubleshooting this so any suggestions would be appreciated. Here's some information.

I did a clean install using the latest in dev.
```
sdcard/write.sh /dev/sdf rpi-3 archlinux kube-systemd
Config wlan
kube-config install
kube-config enable-master
```
(Don't mind the hostname)

```
[root@rpi-slave01 ~]# kube-config info
Architecture: armv7l
Kernel: Linux 4.4.10
CPU: 4 cores x 1200 MHz

Used RAM Memory: 73 MiB
RAM Memory: 922 MiB

Used disk space: 2.7GB (2733040 KB)
Free disk space: 12GB (11624336 KB)

SD Card/deb package was built: 15-05-2016 06:56

kubernetes-on-arm:
Latest commit: 290cef1
Version: 0.7.0

systemd version: v229
docker version: v1.10.0
kubernetes client version: v1.2.0

[root@rpi-slave01 ~]# mount
/dev/mmcblk0p2 on / type ext4 (rw,relatime,data=ordered)
devtmpfs on /dev type devtmpfs (rw,relatime,size=467888k,nr_inodes=116972,mode=755)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/usr/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/net_cls,net_prio type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls,net_prio)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/perf_event type cgroup (rw,nosuid,nodev,noexec,relatime,perf_event)
cgroup on /sys/fs/cgroup/pids type cgroup (rw,nosuid,nodev,noexec,relatime,pids)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,nosuid,nodev,noexec,relatime,memory)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=24,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
mqueue on /dev/mqueue type mqueue (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw)
configfs on /sys/kernel/config type configfs (rw,relatime)
/dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
overlay on /var/lib/docker/overlay/92abe79255bb7ad059f4bbb01c0de7419b6a1bf69f2414348c43dc3967a92e95/merged type overlay (rw,relatime,lowerdir=/var/lib/docker/overlay/d8c7bdb962528d534a8a61af449b87eba8a630397a5553db510506d6d9eb496d/root,upperdir=/var/lib/docker/overlay/92abe79255bb7ad059f4bbb01c0de7419b6a1bf69f2414348c43dc3967a92e95/upper,workdir=/var/lib/docker/overlay/92abe79255bb7ad059f4bbb01c0de7419b6a1bf69f2414348c43dc3967a92e95/work)
nsfs on /run/docker/netns/default type nsfs (rw)
shm on /var/lib/docker/containers/8f6668a380e8a13415ec38ded7eb663687cac3d137472c4675e87731f05a1d4e/shm type tmpfs (rw,nosuid,nodev,noexec,relatime,size=65536k)
mqueue on /var/lib/docker/containers/8f6668a380e8a13415ec38ded7eb663687cac3d137472c4675e87731f05a1d4e/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
tmpfs on /run/user/0 type tmpfs (rw,nosuid,nodev,relatime,size=94444k,mode=700)

[root@rpi-slave01 ~]# docker --version
Docker version 1.10.0, build 590d5108-unsupported

[root@rpi-slave01 ~]# docker info
Containers: 1
 Running: 1
[... 87 lines cut ...]
I0515 14:22:11.984806    1242 plugins.go:291] Loaded volume plugin "kubernetes.io/flocker"
I0515 14:22:11.985001    1242 plugins.go:291] Loaded volume plugin "kubernetes.io/azure-file"
I0515 14:22:11.985180    1242 plugins.go:291] Loaded volume plugin "kubernetes.io/configmap"
I0515 14:22:11.985580    1242 server.go:632] Started kubelet
E0515 14:22:11.986377    1242 kubelet.go:956] Image garbage collection failed: unable to find data for container /
I0515 14:22:11.986625    1242 server.go:109] Starting to listen on 0.0.0.0:10250
I0515 14:22:11.992270    1242 server.go:126] Starting to listen read-only on 0.0.0.0:10255
E0515 14:22:12.009745    1242 event.go:202] Unable to write event: 'Post http://localhost:8080/api/v1/namespaces/default/events: dial tcp 127.0.0.1:8080: connection refused' (may retry after sleeping)
E0515 14:22:12.013200    1242 kubelet.go:1016] Failed to start ContainerManager system validation failed - Following Cgroup subsystem not mounted: [cpuset]
I0515 14:22:12.013374    1242 manager.go:123] Starting to sync pod status with apiserver
I0515 14:22:12.013487    1242 kubelet.go:2356] Starting kubelet main sync loop.
I0515 14:22:12.013573    1242 kubelet.go:2365] skipping pod synchronization - [Failed to start ContainerManager system validation failed - Following Cgroup subsystem not mounted: [cpuset] container runtime is down]
I0515 14:22:12.219074    1242 factory.go:230] Registering Docker factory
I0515 14:22:12.230926    1242 factory.go:97] Registering Raw factory
I0515 14:22:12.280502    1242 manager.go:1003] Started watching for new ooms in manager
I0515 14:22:12.288396    1242 oomparser.go:182] oomparser using systemd
[... cut ...]
I0515 14:22:27.014808    1242 kubelet.go:2365] skipping pod synchronization - [Failed to start ContainerManager system validation failed - Following Cgroup subsystem not mounted: [cpuset]]
E0515 14:22:29.355421    1242 event.go:202] Unable to write event: 'Post http://localhost:8080/api/v1/namespaces/default/events: dial tcp 127.0.0.1:8080: connection refused' (may retry after sleeping)
I0515 14:22:32.015174    1242 kubelet.go:2365] skipping pod synchronization - [Failed to start ContainerManager system validation failed - Following Cgroup subsystem not mounted: [cpuset]]
```

richtong, 2016-05-17:
Ok sorry Lukas, I didn't see the notification will do.

luxas, 2016-05-22:
Maybe this solves the problem?
```
mkdir /sys/fs/cgroup/cpuset
mount -t cgroup -ocpuset cpuset /sys/fs/cgroup/cpuset
```

The problem is obviously that `cpuset` isn't mounted, and that's a feature Kubernetes require.
The problem isn't Kubernetes on ARM. I don't know why `cpuset` isn't mounted automatically...
Which OS are you using?

brujoand, 2016-05-28:
Tried this but failed, on latest raspbian: 
```
root@splinter:/home/pi# mount -o remount,rw /sys/fs/cgroup
root@splinter:/home/pi# mkdir /sys/fs/cgroup/cpuset
root@splinter:/home/pi# mount -t cgroup -ocpuset cpuset /sys/fs/cgroup/cpuset
mount: special device cpuset does not exist
```

Which I guess is obvious when looking at /proc/cgroups:
``` 
root@splinter:/home/pi# cat /proc/cgroups
#subsys_name	hierarchy	num_cgroups	enabled
cpuset	0	1	0
cpu	2	5	1
cpuacct	2	5	1
blkio	3	5	1
memory	4	6	1
devices	5	5	1
freezer	6	5	1
net_cls	7	5	1
```

eueung, 2016-05-29:
Thanks @brujoand , just got my setup working with your hint (I'm using the latest hypriot os 0.8)

@luxas thanks for your great work :-)

btw, I'm new to this k8s stuff and still stuck with the dns and dashboard addon:

```
$ kubectl cluster-info
Kubernetes master is running at http://localhost:8080
KubeDNS is running at http://localhost:8080/api/v1/proxy/namespaces/kube-system/services/kube-dns
Dashboard is running at http://localhost:8080/api/v1/proxy/namespaces/kube-system/services/kubernetes-dashboard
```

response from apiserver

```
# http://node1.local:8080/api/v1/proxy/namespaces/kube-system/services/kube-dns
{
  "kind": "Status",
  "apiVersion": "v1",
  "metadata": {},
  "status": "Failure",
  "message": "no endpoints available for service \"kube-dns\"",
  "reason": "ServiceUnavailable",
  "code": 503
}
```

```
#http://node1.local:8080/api/v1/proxy/namespaces/kube-system/services/kubernetes-dashboard
{
  "kind": "Status",
  "apiVersion": "v1",
  "metadata": {},
  "status": "Failure",
  "message": "no endpoints available for service \"kubernetes-dashboard\"",
  "reason": "ServiceUnavailable",
  "code": 503
}
```

however, after reboot, dashboard endpoint shows a frontend view, but still: 

```
Internal Server Error (500)

Get https://10.0.0.1:443/api/v1/replicationcontrollers: x509: certificate signed by unknown authority
```

Appreciate any hints on these issues, thanks ..

EM

luxas, 2016-05-30:
Great, thanks @brujoand!

I'll add that command to `dev`, unless you want to contribute it :)

@eueung The `kube-dns` doesn't have any http endpoints, so you can't see them in a web browser, but if you want to use the endpoints elsewhere, you have to append the named port to the URL: `http://node1.local:8080/api/v1/proxy/namespaces/kube-system/services/kube-dns:{udp,tcp}`. apiserver fails with "no endpoints available" although there are endpoints when there are more than one endpoint and no named port is specified e.g. `:tcp`.

Dashboard should show up in the browser though, but it seems it isn't running. Double check that?

And when restarting a master, there is a known Kubernetes race condition in the certificate generation process, so master reboots aren't supported (at least not in this version). It will be fixed in the next release though.

johan-wallermark, 2016-05-30:
@luxas, @brujoand The cgroup_enable=cpuset seemed to work on arch linux as well.

eueung, 2016-05-30:
thanks @luxas , never thought that cluster boot sequence would introduce the issue :-D

currently I started only master node, and check the dashboard ... voila, it worked! from the rc view, it seems kube-dns is also running ... 

btw, if anyone interested, previously, I recorded my journey here :

Setup: http://eueung.github.io/docker-stuff/kubernetes/#22
Install + Basic Tests: starting from http://eueung.github.io/docker-stuff/kubernetes/#23 

It was with hypriot os 0.7 (with 0.8 is pretty much the same, except with the cpuset stuff).

EM

luxas, 2016-09-10:
I've added `cgroup_enable=cpuset` by default in v0.8.0 so closing this

closed 2016-09-10 by luxas