luxas/kubernetes-on-arm #98: Kubelet need to run w/ --containerized flag and mount / to /rootfs?

saturnism, 2016-04-30:
Hiya,

@awassink and I had issue earlier with the new distribution - it was unable to mount glusterfs (or use any of the fs options). Upon investigation, the kubelet container was unable to find the mount.glusterfs command. Arjen's kubelet start is a bit different from this one: https://github.com/awassink/k8s-on-rpi/blob/master/rootfs/lib/systemd/system/k8s-worker.service

Which is also similar to how to run a single node k8s: http://kubernetes.io/docs/getting-started-guides/docker/

awassink, 2016-04-30:
True. I tested it already. I had to add `-v /:/rootfs:ro \` and `--containerized \` to the kubenetes startups in order to be able to use NFS and GlusterFS volumes in Kubernetes. This aplies to both the k8s-master.service as well as the k8s-worker.service.

To be complete, NFS and GlusterFS client packages are installed on host system of each node. In order for the kubelet processes running in docker to be able to mount NFS and GlusterFS volumes the root filesystem must be mounted to the docker container and the containerized option must be provided to kubelet.

awassink, 2016-04-30:
I see that this issue relates to the change made in this commit: #[https://github.com/luxas/kubernetes-on-arm/commit/aefa2bca0a743edaf60c30f946ce2d231b4247bc](url)

luxas, 2016-05-02:
So earlier the `--containerized` solution relied on `mount.glusterfs` on host, but now it only searches inside the container and finds nothing. Right?

I think this is quite the same as #83 then.
This should be filed as a bug in the Kubernetes main repo, as I'm going to switch to that one in the near future. I've made huge progress with "official" ARM (both 32 and 64-bit) support for Kubernetes.
See my proposal which contains all links to relevant PRs: https://github.com/kubernetes/kubernetes/issues/17981

TL;DR; Kubernetes runs on arm, arm64 and ppc64le without any extra configuration since `v1.3.0-alpha.3` thanks to the PRs I've made.

So, I think you should propose to add `nfs-utils` and `glusterfs-server/client` to https://github.com/kubernetes/kubernetes/blob/master/cluster/images/hyperkube/Dockerfile.
Right now, the containerized kubelet has no support for special volumes (It's a hack to use `--containerized` and have them available on host)

saturnism, 2016-05-05:
> kubernetes/kubernetes#17981

To be clear - right now kubelet is started without --containerized
parameter hence it isn't finding the commands on host. It should be as easy
as adding the parameter back.

Great stuff on the official support!

luxas, 2016-09-10:
Now it's possible to append `export USE_CONTAINERIZED=true` in `/etc/kubernetes/k8s.conf` to get this behaviour

closed 2016-09-10 by luxas